platformio/platformio-core #47: Support Digistump's Digispark boards

ncolomer, 2015-01-09:
[Digistump](http://digistump.com) created a Attiny85 based microcontroller called Digispark that uses the micronucleus tiny85 bootloader (details and links [here](http://digistump.com/wiki/digispark/tutorials/connecting)). As it is quite nice piece of hardware, it would be cool to support such board in platformio?

ivankravets, 2015-02-11:
Tasks:
- [ ] Create new development platform named `digistump` with supporting for Atmel AVR & SAM MCUs (Assigned to @valeros )
- [ ] Add information about `digistump` platform to main documentation (Assigned to: @ivankravets)
- [ ] Add information about `digistump` platform to [PlatformIO Web Site](http://platformio.org) (Assigned to: @ivankravets)

ivankravets, 2015-02-11:
@ncolomer Do you have any boards from Digispark?

ncolomer, 2015-02-12:
I have two [Attiny85](http://www.aliexpress.com/item/J35-Free-Shipping-Development-Digispark-Kickstarter-USB-Board-for-Arduino/32219495175.html) coming. I'll be able to do tests for you as soon as they arrive.

ivankravets, 2015-02-19:
Hey @ncolomer,

I'm glad to inform you that all boards from "Digistump" are implemented in the latest build of PlatformIO. See supported boards here:
https://github.com/ivankravets/platformio/blob/develop/platformio/boards/digistump.json

I will add them to main documentation and site later. Could you test it?

ncolomer, 2015-02-20:
That's great!
I'll test this board as soon as I receive some
Thanks a lot

closed 2015-02-25 by ivankravets

dromer, 2017-03-11:
Has anyone worked on this yet?
I still have a couple digispark around and it would be great if I can start playing with them again using pio :)