platformio/platformio-core #48: Problem with #include <SPI.h> (from RFM69 library)

laurentS, 2015-01-15:
I'm a bit puzzled, I just installed platformIO, and used it to build some test code which includes RFM69 library (which I installed via platformio lib install 92) for arduino uno.
Everything works fine but I got an issue with the `#include <SPI.h>` line in the RFM69 file:
`.pioenvs/autogen_uno/RFM69_ID92/RFM69.cpp:33:17: fatal error: SPI.h: No such file or directory`

The strange thing is that if I `#include <SPI.h>` in main.ino, it compiles fine. I installed the lib through `platformio lib install` so I suspect there's something wrong.
I'm running linux mint if that's of any help, and followed the get started page to get things running.

Is it a bug, or did I miss something?

ivankravets, 2015-01-16:
@laurentS 

Yes, this is a bug. Thanks a lot! I will fix it.

ivankravets, 2015-01-29:
@laurentS  can you retest it with the latest `develop`?

```shell
$ pip uninstall platformio
$ pip install https://github.com/ivankravets/platformio/archive/develop.zip
```

Thanks a lot! If you have any problems, please reopen this issue.

closed 2015-01-29 by ivankravets

laurentS, 2015-01-31:
@ivankravets just tested the latest `develop`, it works fine, thanks for fixing it!

commit 1921164b (2015-01-29): Refactored *Library Dependency Finder* // Resolve #48 #50 #55