platformio/platformio-core #49: Build fails with libraries that are git repositories

aphelps, 2015-01-15:
I'm making strides to converting my code base to build on platformio, but it seems to not be able to use a library that's within a git repo (ie which I've cloned from elsewhere).  I took a look in the platformio line that generated this error, but its not clear to me what the root cause is:

$ platformio run
Processing arduino_nano environment:

scons: *** '/Users/amp/Dropbox/Arduino/HMTL.platformio/HMTL_Module/.pioenvs/arduino_nano/FastLED/refs' already has a source directory: '/Users/amp/Dropbox/Arduino/libraries/FastLED/.git/refs'.
File "/Library/Python/2.7/site-packages/platformio/builder/tools/platformio.py", line 142, in VariantDirRecursive

ivankravets, 2015-01-16:
Give me please a structure of your project. Go to project root dir and type ``tree -L 3``.

ivankravets, 2015-01-16:
Please test it with the latest `develop` version:
```shell
$ pip uninstall platformio
$ pip install https://github.com/ivankravets/platformio/archive/develop.zip
```
Does it work for you?

closed 2015-01-16 by ivankravets

aphelps, 2015-01-17:
Yeah, that fixed the issue.  Thanks.

gretel, 2015-02-13:
@ivankravets, your suggestion worked for me, too. i'd be glad to have this fix mainstream. regards

ivankravets, 2015-02-13:
@gretel Thanks a lot for your feedback. We are hardly working on the first stable release 1.0.0. PlatformIO has not been released as `stable` yet. You can monitor by active issues for 1.0.0 milestone [here](https://github.com/ivankravets/platformio/issues?q=is%3Aopen+is%3Aissue+milestone%3A1.0.0) or follow us via Twitter [@PlatformIO_Org](https://twitter.com/PlatformIO_Org).

commit 2820e5a2 (2015-01-16): Fix an issue with the libraries that are git repositories  // Resolve #49