platformio/platformio-core #51: compilation issue with arduino source

pelikhan, 2015-01-20:
platformio fails to build the following sketch (tested in Arduino IDE)

* src.ino
```
void parse_string(String word);

void setup()
{
  Serial.begin(9600);
  pinMode(13, OUTPUT);
}

void loop()
{
  digitalWrite(13, HIGH);
  delay(200);
  digitalWrite(13, LOW);
  delay(200);
  parse_string("hello world");

  delay(10);
}
void parse_string(String word) {
  for (int i = 0; i < word.length(); i++) {
    Serial.print(": "); Serial.println(i);
  }
}
```
* platformio.ini
````
[env:myenv]
platform = atmelavr
framework = arduino
board = uno
````
* build output
````
C:\cb\td\win8\shell\work\25664f81-b958-42a6-07e2-d2a7bb60824f>platformio run
Processing myenv environment:
avr-g++ -o .pioenvs\myenv\firmware.elf -Os -Wl,--gc-sections -mmcu=atmega328p .pioenvs\myenv\src\src.o -L.pioenvs\myenv .pioenvs\myenv\libFr
ameworkArduino.a -lm
c:/users/jhalleux/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.3.3/../../../../avr/lib/avr5\libc.a(cmpsf2.o): In function `_
_lesf2':
(.text.avr-libc.fplib+0x0): multiple definition of `__ltsf2'
c:/users/jhalleux/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.3.3/avr5\libgcc.a(_lt_sf.o):c:\avrdev\gcc\build-avr\avr\avr5\
libgcc/../../.././gcc/fp-bit.c:1287: first defined here
c:/users/jhalleux/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.3.3/../../../../avr/lib/avr5\libc.a(cmpsf2.o): In function `_
_lesf2':
(.text.avr-libc.fplib+0x0): multiple definition of `__lesf2'
c:/users/jhalleux/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.3.3/avr5\libgcc.a(_le_sf.o):c:\avrdev\gcc\build-avr\avr\avr5\
libgcc/../../.././gcc/fp-bit.c:1308: first defined here
scons: *** [.pioenvs\myenv\firmware.elf] Error 1
````

ivankravets, 2015-01-21:
Thanks! This issues is caused by obsolete `atmelavr` toolchain for Windows. PlatformIO is used the version from `winavr` (the latest version is dated by 2010year) :(

Now, the new toolchain is available for Windows OS (official from Atmel AVR). Just update your PlatformIO packages:
```shell
$ platformio update
```
Does this help you?

ivankravets, 2015-01-28:
Please reopen issue, if you still have problem.

closed 2015-01-28 by ivankravets