platformio/platformio-core #52: Support Pinoccio board

eiginn, 2015-01-21:
Uses a chip not used in other arduinos currently defined atmega256rfr2
https://github.com/Pinoccio/core-pinoccio/blob/master/avr/boards.txt
https://github.com/Pinoccio/core-pinoccio/blob/master/avr/platform.txt

Tried my hand at adding it to boards but don't think i fully grasp what needs to be done
```json
    "pinoccio": {
        "build": {
            "board": "PINOCCIO",
            "core": "arduino",
            "f_cpu": "16000000L",
            "mcu": "atmega256rfr2",
            "pid": "0x6051",
            "usb_product": "Pinoccio",
            "variant": "pinoccio",
            "vid": "0x1d50"
        },
        "name": "Pinoccio Scout",
        "platform": "atmelavr",
        "upload": {
            "disable_flushing": true,
            "maximum_ram_size": 8192,
            "maximum_size": 253952,
            "protocol": "wiring",
            "speed": 115200,
            "wait_for_upload_port": true
        }
    }
```

ivankravets, 2015-01-21:
Hey @eiginn ,

Thanks a lot for this proposition :+1:  Do you have this board?

P.S: You can move this code to `pinoccio.json` and place it to `~/.platformio/boards` directory. If `boards` dir doesn't exists, just create it.

eiginn, 2015-01-22:
Yeah I do have some of these boards. Thanks for the tip on keeping the board definition separate.

ivankravets, 2015-02-09:
@valeros Please test with Pinoccio library https://github.com/Pinoccio/library-pinoccio

ivankravets, 2015-02-11:
Tasks:
- [ ] Add support for @Pinoccio boards to `atmelavr` development platform (Assigned to @valeros)
- [ ] Add "arduino-pinoccio" example and add Git-submodule from https://github.com/Pinoccio/library-pinoccio to `examples/atmelavr-and-arduino/arduino-pinoccio/lib`  (Assigned to @valeros)
- [ ] Add information about @Pinoccio boards to main documentation (Assigned to: @ivankravets)
- [ ] Add information about @Pinoccio boards to [PlatformIO Web Site](http://platformio.org) (Assigned to: @ivankravets)
- [ ] Create `library.json` for @Pinoccio and pull request to https://github.com/Pinoccio/library-pinoccio (Assigned to: @ivankravets)

robin7331, 2016-07-17:
I can't manage to get the pinoccio working with PlatformIO. Any progress with this?

ivankravets, 2016-07-17:
@robin7331 is this project alive? I see that they don't support it.

robin7331, 2016-07-17:
I don't think the Pinoccio project is still active in terms of further development.
But you can still buy these boards at [CroudSupply](https://www.crowdsupply.com/pinoccio/mesh-sensor-network)

I personally have a couple of boards that I use for quick prototyping and so on. Therefore it would be great to code them via platformio!

ivankravets, 2016-07-17:
@robin7331 how do you program these boards now?

robin7331, 2016-07-17:
Arduino IDE

ivankravets, 2016-07-17:
Do you use the latest Arduino IDE with board_manifest.json or previous 1.5.x?

robin7331, 2016-07-17:
I am running the latest Version of the Arduino IDE (1.6.9).
Sorry, but what do you mean by board_manifest.json ?

ivankravets, 2016-07-17:
This is URL that you paste to Arduino IDE to add support for Pinoccio. Could you share it here?

robin7331, 2016-07-17:
Ah I got it!
Yes, here is the URL `https://robertgallup.github.io/pinoccio/package_pinoccio_index.json`

ivankravets, 2016-07-17:
@valeros please add support for this board

valeros, 2016-07-18:
Hi @robin7331 !
Please install [development branch](http://docs.platformio.org/en/latest/installation.html#development-version) and update packages via `platformio update`. Does it work?
Thanks!

commit 60b3fef3 (2016-07-18): Add support for Pinoccio Scout board // Resolve #52
closed 2016-07-18 by ivankravets