platformio/platformio-core #58: platformio not handling assembly correctly?

aphelps, 2015-02-02:
Sorry to continue with the "this works under the Arduino IDE but not platformio" topics.

In this one, I'm using an FFT library from https://github.com/adafruit/piccolo which is written in assembly. It looks like platformio does compile it into a library, but may skip a step:

```
amp@AMP-MBP-home:~/Dropbox/Arduino/ObjectLights/platformio [master]$ platformio run
Processing arduino_nano environment:
avr-g++ -o .pioenvs/arduino_nano/firmware.elf -Os -mmcu=atmega328p -Wl,--gc-sections -Wl,--start-group .pioenvs/arduino_nano/src/Sound.o .pioenvs/arduino_nano/src/SoundCLI.o .pioenvs/arduino_nano/src/SoundUnit.o -L.pioenvs/arduino_nano .pioenvs/arduino_nano/libDebug.a .pioenvs/arduino_nano/libffft.a .pioenvs/arduino_nano/libSerialCLI.a .pioenvs/arduino_nano/libRS485_non_blocking.a .pioenvs/arduino_nano/libSoftwareSerial.a .pioenvs/arduino_nano/libSPI.a .pioenvs/arduino_nano/libFastLED.a .pioenvs/arduino_nano/libHMTLTypes.a .pioenvs/arduino_nano/libPixelUtil.a .pioenvs/arduino_nano/libRS485Utils.a .pioenvs/arduino_nano/libSocket.a .pioenvs/arduino_nano/libMPR121.a .pioenvs/arduino_nano/libWire.a .pioenvs/arduino_nano/libGeneralUtils.a .pioenvs/arduino_nano/libEEPromUtils.a .pioenvs/arduino_nano/libEEPROM.a .pioenvs/arduino_nano/libHMTLMessaging.a .pioenvs/arduino_nano/libHMTLprotocol.a .pioenvs/arduino_nano/libFrameworkArduino.a -lm
.pioenvs/arduino_nano/libffft.a: could not read symbols: Archive has no index; run ranlib to add one
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/arduino_nano/firmware.elf] Error 1
```

ivankravets, 2015-02-02:
Thanks! I've not ever test PlatformIO with assembly files. Sorry :disappointed: 

Please upgrade your PlatformIO with new `develop`:
$ pip install https://github.com/ivankravets/platformio/archive/develop.zip

P.S: I've just added new lib [Adafruit-LED-Backpack](https://github.com/adafruit/Adafruit-LED-Backpack-Library) and you can compile https://github.com/adafruit/piccolo/blob/master/Piccolo/Piccolo.pde with the several commands:
```shell
# init empty project and copy https://github.com/adafruit/piccolo/blob/master/Piccolo/Piccolo.pde to "src" dir
$ platformio lib install 128
$ platformio run
```
You don't need to install Adafruit-GFX, this library has pre-defined dependency https://github.com/ivankravets/Adafruit-LED-Backpack-Library/blob/patch-1/library.json

closed 2015-02-02 by ivankravets

aphelps, 2015-02-03:
After updating it looks good.  Thanks for all the work you're putting into this, I'm pretty excited to be making the switch.

Also thanks for installing that library, although I'm not actually using it in this particular project, I just grabbed their code for the sound sensing ;)

commit cf823c47 (2015-02-02): Resolve #58 with handling assembly files