platformio/platformio-core #59: Output compiled size and static memory usage with 'platformio run'

aphelps, 2015-02-03:
(This just an enhancement request)

One minor feature I find useful from the Arduino IDE is that it outputs the size of the program being updated and (in the 1.5 branch) the total size of static variables.  These are both useful both to be careful about approaching the limits, but also for optimizing code during development.

'platformio run -t upload' already outputs the size, but since I'm not always uploading it would be useful if that was output for each environment during a build.

For the static variable usage, I have a recollection that theres a tool in the avr tool chain that produces this info, but I'm not immediately seeing it.  Regardless this sort of info is hugely valuable when developing for a chip that may only have 1k of memory.

aphelps, 2015-02-03:
Actually, I found the command for the memory size: avr-size -C -x .pioenvs/arduino_nano/firmware.elf

If this is a fairly easy change to make, if you point me to the right areas of the code I'd be happy to make the change myself.

ivankravets, 2015-02-09:
@aphelps Thanks a lot for idea! You can try it in the latest `develop` branch :+1:

closed 2015-02-09 by ivankravets

aphelps, 2015-02-09:
That works great, thanks!

commit b289d0ba (2015-02-09): Merge pull request #64 from valeros/develop

Output compiled size and static memory usage // Resolve #59