platformio/platformio-core #67: Relocation truncated to fit when using math.h with serial library on Arduino

duci9y, 2015-02-10:
This is the output:

    Processing autogen_uno environment:
    Auto-detected UPLOAD_PORT: /dev/ttyACM0
    avr-g++ -o .pioenvs/autogen_uno/firmware.elf -Os -Wl,--gc-sections -mmcu=atmega328p .pioenvs/autogen_uno/src/test.o -L.pioenvs/autogen_uno .pioenvs/autogen_uno/libFrameworkArduino.a -lm
    /home/pi/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.5.1/../../../../avr/lib/avr5/libm.a(fp_powsodd.o):../../../../../source/avr-libc-1.8.0/libm/fplib/fp_powsodd.S:59: relocation truncated to fit: R_AVR_13_PCREL against symbol `__mulsf3' defined in .text section in /home/pi/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.5.1/avr5/libgcc.a(_mul_sf.o)
    /home/pi/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.5.1/../../../../avr/lib/avr5/libm.a(fp_powsodd.o):../../../../../source/avr-libc-1.8.0/libm/fplib/fp_powsodd.S:69: relocation truncated to fit: R_AVR_13_PCREL against symbol `__mulsf3' defined in .text section in /home/pi/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.5.1/avr5/libgcc.a(_mul_sf.o)
    collect2: ld returned 1 exit status
    scons: *** [.pioenvs/autogen_uno/firmware.elf] Error 1

Apparently, gcc is invoked with the following command: 

    avr-g++ -o .pioenvs/autogen_uno/firmware.elf -Os -Wl,--gc-sections -mmcu=atmega328p .pioenvs/autogen_uno/src/test.o -L.pioenvs/autogen_uno .pioenvs/autogen_uno/libFrameworkArduino.a -lm

Sample code:

    void setup() {
        float t = sin(3.141);
        Serial.begin(9600);
    }
    void loop() {
    }

ivankravets, 2015-02-11:
@duci9y Thanks a lot for reporting! This problem was caused by obsolete `ARMv6l` toolchain. Can you update PlatformIO's packages and re-test it again?
```shell
$ platformio update
$ platformio run
```

ivankravets, 2015-02-12:
Please reopen if you still have problems

closed 2015-02-12 by ivankravets