platformio/platformio-core #68: Migrate to Arduino Framework 1.6.0

ivankravets, 2015-02-10:
* update dependent libraries 
* verify `build_flags`
* re-pack `framework-arduinoavr` package

aphelps, 2015-02-10:
It might be a good idea to retain the previous framework as an option for a bit.  I know when they initially moved from the 1.0 branch to the 1.5 one that there were some incompatibilities with existing libraries, I haven't been paying much attention to the 1.6 release process so don't know how likely this release is to break existing code.

ivankravets, 2015-02-11:
@aphelps 1.5.x has not been released in "stable" version. It was like "beta". The 1.6 is "stable" version after 1.5.x. In other words, 1.6 is fully compatible with 1.5.x. See [Arduino Release Notes](http://www.arduino.cc/en/Main/ReleaseNotes)

aphelps, 2015-02-11:
Cool, should be fine then.  I'll definitely try it out when you make the switch

ivankravets, 2015-02-11:
@aphelps Which types of boards do you have? :blush:

ivankravets, 2015-02-13:
@aphelps Please update your PlatformIO packages:
```shell
> platformio update
```

closed 2015-02-13 by ivankravets