platformio/platformio-core #74: Improvements to "platformio run" command

ivankravets, 2015-02-12:
- [ ] Cleanup automatically `.pioenvs` when `platformio.ini` has been changed
- [ ] Add "start", "end" and "took" time info per environment

ivankravets, 2015-02-15:
Dear @aphelps, @franciscoruiz, @pelikhan, @valeros, @laurentS, @yarekt, @bettabacon, @eiginn,  @mac7, @MrCustomizer, @elyulka

* Current [HISTORY of 1.0.0](http://docs.platformio.org/en/latest/history.html).
* [Active issues for 1.0.0 milestone](https://github.com/ivankravets/platformio/issues?q=is%3Aopen+is%3Aissue+milestone%3A1.0.0)

Could you help me with testing this issue? The `platformio run` command was seriously refactored. I want to know that it works good with your projects. If we wont have new issues, then I will release `1.0.0` this week.

1. [Install the latest development version of PlatformIO](http://docs.platformio.org/en/latest/installation.html#development-version)
2. Update packages: `$ platformio update`
3. Test your projects :blush: 

Thanks a lot!

franciscoruiz, 2015-02-15:
Sure thing!

I'll give a go next week or so and I'll let you know how it goes.

Cheers

Francisco Ruiz

aphelps, 2015-02-16:
I should be able to give this a try later this week

eiginn, 2015-02-16:
Same, will give it a try later in the week

MrCustomizer, 2015-02-16:
Success! The current version of the development branch is still working with my current project.

MrCustomizer, 2015-02-16:
I nearly forgot to test it on Windows. My first test was on a Linux machine. Windows works, too.

ivankravets, 2015-02-16:
@MrCustomizer Which type of board do you have? Did you test upload operation?

MrCustomizer, 2015-02-16:
I have an Arduino Nano, but I haven't tested uploading as I'm issuing a custom command for uploading via bluetooth and I haven't had the time yet to figure out how to do that with platformio. Perhaps someone else could test uploading?

ivankravets, 2015-02-16:
@MrCustomizer Yes, thanks for it! I hope it works without any problems :beer: :smile:

aphelps, 2015-02-17:
I've updated and done clean compiles of my projects, including uploading to an Arduino Nano and a Mini Pro.  All looks fine so far.

ivankravets, 2015-02-19:
Thanks! I'm going to close this issue. If you have any problems with it, don't hesitate to reopen

closed 2015-02-19 by ivankravets

franciscoruiz, 2015-03-01:
I've used it pretty thoroughly now, both with TI Launchpad and Arduino UNO; and it works like charm! :+1:

ivankravets, 2015-03-01:
@franciscoruiz Thank you for the feedback :+1: :blush:

commit cf925b03 (2015-02-14): Improve "run" command: env details, timing // Resolve #74