platformio/platformio-core #77: Header file removed while building, resulting compilation error

ncolomer, 2015-02-15:
Hello,

I observe strange behavior while building a project relying on the platformio lib id 124 (related to ivankravets/platformio-libmirror/issues/1). It seems a file is removed by the build process, resulting into a compilation error stating a header file is missing.

In my case, the file `RHReliableDatagram.h` is present in the `lib/RadioHead_ID124` but not in `.pioenvs/arduino_nano/RadioHead_ID124`. Even if I copy/paste the file in the latter, after a build, the file is deleted and the build fail as follow:

```bash
$ platformio run -t upload
Processing arduino_nano environment:
avr-g++ -o .pioenvs/arduino_nano/src/main.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO=160 -DARDUINO_AVR_NANO -I.pioenvs/arduino_nano/FrameworkArduino -I.pioenvs/arduino_nano/FrameworkArduinoVariant -I.pioenvs/arduino_nano/EEPROM -I.pioenvs/arduino_nano/RadioHead_ID124 -I.pioenvs/arduino_nano/SPI .pioenvs/arduino_nano/src/main.cpp
.pioenvs/arduino_nano/src/main.cpp:4:32: fatal error: RHReliableDatagram.h: No such file or directory
 #include <RHReliableDatagram.h>
                                ^
compilation terminated.
scons: *** [.pioenvs/arduino_nano/src/main.o] Error 1
```

Any idea?

ncolomer, 2015-02-16:
After having upgraded platformio version to `0.11.0-dev`, compilation still fail but the issue is different.

I tried form zero and created a project from start. Here is a [gist](https://gist.github.com/ncolomer/388f49e10e825d80ce65) containing files `platformio.ini` and a bare`src/main.cpp` in addition to console output.

It seems the compilation issue is due to missing files related to a RadioHead dependency `STM32ArduinoCompat`. Path/file are not be resolved correctly (or stuff are in the wrong place).

PS: there is a lot of warnings on the console output, I don't know what causes them.

ncolomer, 2015-02-16:
When I add the path to the `STM32ArduinoCompat` directory as follow in `platformio.ini`:
```
build_flags = -I/Users/ncolomer/.platformio/lib/RadioHead_ID124/STM32ArduinoCompat
```
The issue change and is complaining about a missing `stm32f4xx.h` header file:
```
scons: warning: Two different environments were specified for target /Users/ncolomer/Developer/Workspace/perso/arduino/test/.pioenvs/autogen_uno/libRadioHead_ID124.a,
	but they appear to have the same action: $AR $ARFLAGS $TARGET $SOURCES
$RANLIB $RANLIBFLAGS $TARGET
File "/Library/Python/2.7/site-packages/platformio/builder/tools/platformio.py", line 107, in BuildLibrary
In file included from .pioenvs/autogen_uno/RadioHead_ID124/RH_Serial.cpp:7:0:
/Users/ncolomer/.platformio/lib/RadioHead_ID124/STM32ArduinoCompat/HardwareSerial.h:9:23: fatal error: stm32f4xx.h: No such file or directory
 #include <stm32f4xx.h>
                       ^
compilation terminated.
scons: *** [.pioenvs/autogen_uno/RadioHead_ID124/RH_Serial.o] Error 1
```

ivankravets, 2015-02-16:
I've just updated [config for this library](https://github.com/ivankravets/platformio-libmirror/blob/master/configs/radiohead.json) and disabled "STM32" things. Just update your libs:
```shell
$ platformio lib update
```

P.S: I see the warnings from SCons. It's a PlatformIO Library Dependency Finder bug. I will work on it and report here. These warnings don't have any influence on the final firmware. 

Thanks a lot for your report! :+1:

ivankravets, 2015-02-16:
Thanks! I've just fixed it. Please update your development version of PlatformIO:
http://docs.platformio.org/en/latest/installation.html#development-version

closed 2015-02-16 by ivankravets

ncolomer, 2015-02-17:
All is running smoothly now, thanks!

commit 69d952ea (2015-02-16): Fixed SCons warning: Two different environments were specified for target // Resolve #77