platformio/platformio-core #83: platformio init: change source directory

makefu, 2015-02-17:
Maybe i missed something but is it correct that there is currently no way to change the source directory path? 
I am currently trying to migrate my arduino projects to platformio and for me the best case would be to simply throw a platformio.ini file in the base directory of my projects and compile my projects with different source directories.
i also hate the arduino ide requirement that every source file of a project must be named like the project, what a buill ... 

Being able to choose my folder and file naming  means a lot to me :)

ivankravets, 2015-02-17:
> simply throw a platformio.ini file in the base directory of my projects and compile my projects with different source directories.

I want to break main concept of PlatformIO - "to build the single source code with the multiple development platforms". `platformio.ini` file should be configured per unique source directory `src`.

See  [example](https://github.com/ivankravets/platformio/tree/develop/examples/atmelavr-and-arduino/arduino-internal-libs) how to build single "blink" code for the multiple Arduino boards.

P.S: You should estimate all benefits of [platformio.ini](http://docs.platformio.org/en/latest/projectconf.html) and start use it per project.

makefu, 2015-02-17:
Thanks for the heads-up, i think i already understood the main concept i think.

What i am looking for is a way to provide a source-folder other than `./src` either in the platformio.ini or when initializing the project.

Regards

ivankravets, 2015-02-17:
The `lib`, `src`, `.pioenvs` and `platformio.ini` are hard-coded names. You can't change them. The one thing which you can change is `lib` directory, you can setup alternative location for your project's libs.

Could I ask you: "Why do you need `other than ./src` folder? :blush: 

As I understand, you just need to initialize project in the another folder and use it. For example,
```shell
> cd /my/project/dir
> platformio run

# or

> cd /my/project/dir && platformio run

```

aphelps, 2015-02-18:
The reason is likely that the Arduino IDE forces certain naming and structural constraints, namely that a project named 'TestProject' must be within a folder named 'TestProject' and include a main file named 'TestProject/TestProject.ino'.  While I do think full compatibility is a good idea, I've been working around this by using directory links.  ie:
  $BASE_DIR/TestProject/TestProject.io
  $BASE_DIR/platformion.ini
  $BASE_DIR/src -> ./TestProject

MrCustomizer, 2015-02-22:
I came here to open an issue with exactly the same feature request. Now I see, someone else has already done that ;). I will try to clarify the problem a little bit:

I'm using platformio to build my Arduino projects while at the same time maintaining compatibility with the Arduino IDE.

As makefu described, one really annoying limitation of the Arduino IDE is the fact, that the ino-source file has to have the exact same name as the parent directory, the source file resides in. Don't ask me, why they did that…

One way to solve this, is using links like aphelp proposed, which probably will not be suitable if you want to work on different operating systems in the same project. Also, this solution will likely cause trouble when using a version control system.

My current solution is to name my ino-file "src.ino". This way,  the Arduino IDE is happy, because it can open a file with the name "src", which is located in a folder named "src" and platformio is happy, because the source file resides in a directory with the name "src". Working on several source files from different Arduino projects in the same IDE (which I'm doing a lot, lately) gets pretty confusing with this hack, though, as a lot of source files are named "src.ino".

The problem could be solved, if the name of the src-folder was configurable as makefu proposed.

closed 2015-02-22 by ivankravets

ivankravets, 2015-02-22:
Hey @makefu, @aphelps, @franciscoruiz, @valeros 

You convinced me :+1: :blush: . See `src_dir` option in [documentation](http://docs.platformio.org/en/latest/projectconf.html#platformio)

MrCustomizer, 2015-02-22:
Works like a charm, thank you!

makefu, 2015-02-22:
that is great to hear, thanks for the improvement!

aphelps, 2015-02-22:
Thanks for the change, I'll try that out as an alternative to my linking.

As a note on a comment above regarding links and version control, at least with git and Unix based platforms you can commit a symbolic link and it behaves properly when cloned.  I haven't tried this with Windows, but MacOS and Ubuntu are fine.

gsavin, 2015-03-11:
Does ``src_dir`` support multiple paths ? To share commun code between several projects for example.

ivankravets, 2015-03-11:
> Does src_dir support multiple paths ? To share commun code between several projects for example.

Sure, please read detailed information in documentation: http://docs.platformio.org/en/latest/projectconf.html#src-dir

P.S: But... I'm not sure that I've got you. The main idea of PlatformIO and ``platformio.ini`` is to keep single code  within single project and define multiple environments for cross-platform compilation.

gsavin, 2015-03-11:
Actually, I've already checked the docs, but I understand that only one path is possible :)
I have two projects which are working together, so they share some code, like a local library.  So, I have two projects with their own ``src`` directory, but I would like to add an other and common directory. Something like ``src_dir = src/ ; ../src/``. But maybe I am following a complete wrong way for doing this !

ivankravets, 2015-03-11:
Dear @gsavin,

> so they share some code, like a local library

You've just answered on this question :smile: 

Just move "common code for library" to folder, which will be located in "Project #1", then use the ``lib`` folder from "Project #1" in "Project #2". For example,

Project #1 ``platformio.ini``
```ini
[platformio]
lib_dir = /path/to/own/libdir

[env:my_project_1]
# some data
```

Project #2  ``platformio.ini``
```ini
[platformio]
lib_dir = /path/to/own/libdir

[env:my_project_2]
# some data
```

Then, create new folder (use any names, like ``common_code`` or etc) in ``/path/to/own/libdir`` and put common source code here.

P.S: You can keep common code in Project 1 or 2, and then use ``lib_dir`` from one the projects. You don't need to override ``lib_dir`` in both projects.

gsavin, 2015-03-11:
Thanks :) Sounds good.
I guess that my installed libraries will not be usable anymore, but I can copy them into the new lib folder !
Great tool btw.

ivankravets, 2015-03-11:
> Great tool btw.

Thanks! :+1: Let's wait for the new release... :blush:  + 3 new development platforms (NXP, Freescale & Nordic), +30 boards, + full support for mbed platforms and mbed framework http://developer.mbed.org/platforms/

Please follow us on Twitter [@PlatformIO_Org](https://twitter.com/platformio_org)

ThomasDevoogdt, 2017-11-26:
I do something like this:

```c++
//
// Created by thomas on 25/11/17.
//

//#define sensor_log
#define sensor_bmp

// sensor_log example
#ifdef sensor_log

#include "../../examples/sensor_log/sensor_log.ino"

#endif

// sensor_bmp example
#ifdef sensor_bmp

#include "../../examples/sensor_bmp/BMPholder.h"
#include "../../examples/sensor_bmp/BMPholder.cpp"
#include "../../examples/sensor_bmp/sensor_bmp.ino"

#endif
```

Just include the .ino files of the custom location in your main.ino in src

commit 46ae4c1a (2015-02-22): Add `src_dir` option to `[platformio]` section of `platformio.ini` which allows to redefine location to project’s source directory // Resolve #83