platformio/platformio-core #91: Library config using YAML

ricardojlrufino, 2015-02-22:
its a more easy format
http://docs.platformio.org/en/latest/librarymanager/config.html

ivankravets, 2015-02-22:
@ricardojlrufino I'm agree with you that `YAML` is more readable. However, this format doesn't have native parsers within the most popular languages. I mean, that developer should install additional libraries/packages to work with this format. To my mind, `JSON` is more standardised cross-format for configuration aims.

ricardojlrufino, 2015-02-22:
It's a bit boring to write in JSON, is usually thing the computer does.
May be an additional format

ivankravets, 2015-02-22:
You can write in `YAML` and [convert it to JSON](http://yamltojson.com).

ricardojlrufino, 2015-02-23:
His tool could do that ... just a tip

ivankravets, 2015-02-25:
@ricardojlrufino thanks a lot for tip. If you have any problems with converting `YAML` to `JSON` - please re-open this issue.

closed 2015-02-25 by ivankravets