platformio/platformio-core #92: Upload failing on current development branch

aphelps, 2015-02-22:
One of the changes to the develop branch over the last few days appears to have broken uploading.  I haven't narrowed it down to an exact commit, but I reverted back to commit 86260a48ba5a154649800146aefe7d503e6cc059 and the problem went away.

This is on MacOS, attempting to upload to arduino nanos.

```
amp@AMP-MBP-home:~/Dropbox/Arduino/ObjectLights/platformio/CubeLights [master]$ platformio run -e light_cube -t upload --upload-port /dev/cu.usbserial-AM01SK3R
[Sun Feb 22 14:37:13 2015] Processing light_cube (build_flags: -DPLATFORMIO -Wall -DCUBE_LIGHT_BASIC_CONTROL, platform: atmelavr, board: nanoatmega328, framework: arduino, global_buildflags: -DPLATFORMIO -Wall, upload_port: /dev/cu.usbserial-AM01SK3R)
-----------------------------------------------------------------------------------------------------------------------------------------------
before_upload(["upload"], [".pioenvs/light_cube/firmware.hex"])
"/Users/amp/.platformio/packages/tool-avrdude/avrdude" -q -D -p atmega328p -C "/Users/amp/.platformio/packages/tool-avrdude/avrdude.conf" -c arduino -U flash:w:.pioenvs/light_cube/firmware.hex:i
avrdude: ser_open(): can't open device "unknown": No such file or directory
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.

scons: *** [upload] Error 1
```

ivankravets, 2015-02-22:
@aphelps sorry :( It was huge refactoring with build scripts. I've just fixed it. I will re-test uploading for all platforms (which I have) tomorrow.

closed 2015-02-22 by ivankravets

aphelps, 2015-02-23:
No problem, these things happen.  I love all the tests I've been watching going in ;)

ivankravets, 2015-02-23:
Which tests do you mean? From `avrdude` ? :smile:

commit d1ca17c4 (2015-02-22): Fix uploading for "atmelavr" platform // Resolve #92