platformio/platformio-core #102: Handle PLATFORMIO_* Environment variables

ivankravets, 2015-03-03:
- [ ] Add new section "Environment variables" to documentation
- [ ] Add variable `PLATFORMIO_HOME_DIR` - see http://docs.platformio.org/en/latest/projectconf.html#home-dir
- [ ] Add variable `PLATFORMIO_LIB_DIR` - see http://docs.platformio.org/en/latest/projectconf.html#lib-dir
- [ ] Add variable `PLATFORMIO_SRC_DIR` - see http://docs.platformio.org/en/latest/projectconf.html#src-dir
- [ ] Add variable `PLATFORMIO_ENVS_DIR` - see https://github.com/ivankravets/platformio/blob/develop/platformio/util.py#L112
- [ ] Add variable `PLATFORMIO_PROMPTS` - can be `true` or `false`. By default is equal to `enable_prompts` setting.  See http://docs.platformio.org/en/latest/userguide/cmd_settings.html#platformio-settings-get
- [ ] Add variable `PLATFORMIO_SRCBUILD_FLAGS` - see https://github.com/ivankravets/platformio/blob/develop/platformio/builder/tools/platformio.py#L67

pelikhan, 2015-03-04:
+1 for these.

ivankravets, 2015-03-04:
Implemented in https://github.com/ivankravets/platformio/commit/ae92cdedeaadeae51b34bd3e22ac177102aa5dcb

closed 2015-03-04 by ivankravets

ivankravets, 2015-03-04:
Documentation: http://docs.platformio.org/en/latest/envvars.html