platformio/platformio-core #105: Automatically append -DPLATFORMIO to builder

ivankravets, 2015-03-03:
Add `PLATFORMIO` define to `gcc` builder which should be equal to `platformio.__version__` in the INT format (4bytes):
1byte = major version
2byte = minor version
3byte = bug fixes number
4byte = dev/pre/post/alpha/beta and etc

Pack this information in `MSB` order.

closed 2015-03-13 by ivankravets

aphelps, 2015-03-13:
Hah, thanks for adding that.  I had already put the same tag into all my own platformio.ini files ;)

ivankravets, 2015-03-13:
Why you haven't requested it earlier? :smile: Do you have any ideas/propositions as for 1.1.0 release?

aphelps, 2015-03-13:
Didn't think about it, its been a couple of weeks since I've had time to work on the software as I've needed to update the boards I've been running on it.  I'll contemplate new ideas, I should be back to software development this week.

ivankravets, 2015-03-14:
> I'll contemplate new ideas, I should be back to software development this week.

Thanks! We will wait for them (ideas) :+1:

commit 04281661 (2015-03-13): Automatically append -DPLATFORMIO to builder // Resolve #105