platformio/platformio-core #106: digispark-tiny upload failing

jonlorusso, 2015-03-03:
digispark-tiny upload is failing:

> platformio run --target upload 

Fails with:

> [Tue Mar  3 20:01:16 2015] Processing digispark-tiny (platform: atmelavr, board: digispark-tiny, framework: arduino)
>BeforeUpload(["upload"], [".pioenvs/digispark-tiny/firmware.hex"])
>scons: *** [upload] Explicit exit, status Error: Please specify `upload_port` for environment or use global `--upload-port` option.
> 
> Error: Please specify `upload_port` for environment or use global `--upload-port` option.

When uploading to the digispark (via micronucleus) I am normally prompted to insert the device.  No port name is required.

ivankravets, 2015-03-03:
Thanks a lot for report! This is a bug, I will fix it as soon as possible! Current solution is to put "something" to ``--upload-port``, like:
```shell
> platformio run -t upload --upload-port /dev/null
```

Does it work for you?

jonlorusso, 2015-03-03:
> $ platformio run --target upload --upload-port /dev/null

> [Tue Mar  3 21:38:14 2015] Processing digispark-tiny (platform: atmelavr, board: digispark-tiny, >framework: arduino)
> BeforeUpload(["upload"], [".pioenvs/digispark-tiny/firmware.hex"])
> "/home/pi/.platformio/packages/tool-micronucleus/micronucleus" -c digispark --timeout 60 -b -P /dev/null -U flash:w:.pioenvs/digispark-tiny/firmware.hex:i

At this point, output pauses, and I plug in the device (despite not being prompted to).  Once I plug in the device, I see:

> micronucleus: library/micronucleus_lib.c:66: micronucleus_connect: Assertion `res >= 4' failed.
> > Please plug in the device ...
> > Press CTRL+C to terminate the program.

> Aborted
> scons: *** [upload] Error 134
> ========================= [ ERROR ] Took 14.70 seconds =========================

ivankravets, 2015-03-03:
@jonlorusso We will fix it tomorrow. I've just opened for it separated issue https://github.com/ivankravets/platformio/issues/107

valeros, 2015-03-03:
It seems that this error is caused by usb permissions issue.
Did you try to insert platformio udev.rules file? https://github.com/ivankravets/platformio/blob/develop/scripts/99-platformio-udev.rules

jonlorusso, 2015-03-03:
Getting closer:

> [Tue Mar  3 22:09:52 2015] Processing digispark-tiny (platform: atmelavr, board: digispark-tiny, framework: arduino)
> BeforeUpload(["upload"], [".pioenvs/digispark-tiny/firmware.hex"])
"/home/pi/.platformio/packages/tool-micronucleus/micronucleus" -c digispark --timeout 60 -b -P /dev/null -U
flash:w:.pioenvs/digispark-tiny/firmware.hex:i
> Please plug in the device ...
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 40% complete
> Device has firmware version 1.2
> Available space for user applications: 6010 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 94  page size: 64
scons: *** [upload] Error 1
> Erase function sleep duration: 752ms
parsing: 40% complete
> Error opening flash:w:.pioenvs/digispark-tiny/firmware.hex:i: No such file or directory
> Error loading or parsing hex file.
========================= [ ERROR ] Took 16.18 seconds =========================

ivankravets, 2015-03-04:
Fixed in https://github.com/ivankravets/platformio/issues/107 and will be released in PlatformIO 1.0.2 today.

Please re-open if you still have this issue.

closed 2015-03-04 by ivankravets

valeros, 2015-03-04:
Could you try `develop` snapshot? http://docs.platformio.org/en/latest/installation.html#development-version

Does it help you?

Thanks in advance :+1:  :beer: !

ivankravets, 2015-03-05:
Could you try new version 1.1.0?  Please use `platformio upgrade` command.

Thanks in advance!

nebulous, 2016-03-27:
For future Googlers: a similar problem occurs in OS X with USB3 ports. Try connecting the target device via a USB2 hub.

erm3nda, 2019-11-19:
Case of mine, a workaround is to just press Upload, then power up device AFTER process is trying to upload (it will be paused for a while, no need to be a ninja).
No device in list, just install micronucleus drivers then voilá.

https://github.com/micronucleus/micronucleus
https://github.com/micronucleus/micronucleus/tree/master/windows_driver

Hope this helps.

icepaule, 2023-11-03:
> Case of mine, a workaround is to just press Upload, then power up device AFTER process is trying to upload (it will be paused for a while, no need to be a ninja). No device in list, just install micronucleus drivers then voilá.
> 
> https://github.com/micronucleus/micronucleus https://github.com/micronucleus/micronucleus/tree/master/windows_driver
> 
> In my particular case, i have atinny85 usb device, i found and used those drivers: https://github.com/digistump/DigistumpArduino/releases/download/1.6.7/Digistump.Drivers.zip
> 
> Hope this helps.

Thx a lot. It DID help....  ;-)