platformio/platformio-core #108: PlatformIO as Continuous Integration tool for embedded projects

ivankravets, 2015-03-03:
Aims
------
PlatformIO is #1 tool for "Cloud Compiling" now. The one missing feature is to allow each developer/script to use PlatformIO in "Continuous Integration" manner. PlatformIO can build embedded projects for the different development platforms (Atmel AVR, Atmel SAM, TI MSP430, TI TIVA, Teensy, (soon Freescale FRDM, NXP LPC, Nordic RF51) and etc).

Requirements
------------------
The main aim is to promote PlatformIO as tool which is the best for "Continuous Integration". It should allow to specify `src` files, `lib` directories and `board`s which PlatformIO should use for running.

Implementation
--------------------

It should be implemented via the new command ``platformio ci`` with the next options:
* ``-s, --src`` the list with source files/dirs.They should be copied to ``src`` directory in new PlatformIO based project.
* ``-l, --lib`` the list with source code of libraries which should be copied to ``lib`` directory of PlatformIO project. 
* ``-b, --board`` the list with boards. All tests should be passed over them.

**NOTE**
``--src`` and ``--lib`` can be MULTIPLE. [GLOB](http://en.wikipedia.org/wiki/Glob_(programming)) syntax should by allowed. 

Using
-------

For example, let's use this project https://github.com/felis/USB_Host_Shield_2.0. We want to test a few examples with the different boards/platforms:
```bash
# clone repo and change directory to it
platformio ci --src="examples/Bluetooth/PS3SPP/PS3SPP.ino" --lib="." --board=uno --board=teensy31 --board=due
```

Travis CI
-----------

Yep :) PlatformIO could be the best friend of @travis-ci. Example of ``.travis.yml``:
```yml
language: python
python:
    - "2.7"

install:
    - python -c "$(curl -fsSL https://raw.githubusercontent.com/ivankravets/platformio/master/scripts/get-platformio.py)"

script:
    - platformio ci --src="examples/Bluetooth/PS3SPP/PS3SPP.ino" --lib="." --board=uno --board=teensy31 --board=due
    - platformio ci --src="examples/pl2303/pl2303_gps/pl2303_gps.ino" --lib="." --board=megaatmega2560
# ... and etc
```

----------

We have already implemented a lot of these things... See how does testing work with [PlatformIO Examples](https://github.com/ivankravets/platformio/tree/develop/examples) via Travis CI: 
https://travis-ci.org/ivankravets/platformio/jobs/52805475

----------

Friends @valeros, @aphelps, @robertinant, @PaulStoffregen, @aethaniel, @xxxajk, @Lauszus, @pelikhan, @ncolomer, @bettabacon, @eiginn, @Informatic, @mac7, @manashmndl, @MrCustomizer, @franciscoruiz, @elyulka

What do you think about it?

pelikhan, 2015-03-04:
- What about those prompts? They are complete show stopper right now.
- Travis support sounds great however it should limited to it. There are many other build systems.
- Travis supports a feature to cache certain folders, like npm-modules. You should update your example: installing a complete platformio drop on every build might quickly turn out costly for you.

https://travis-ci.org/ivankravets/platformio/jobs/52805475

%20thub.com/manashmndl>, @MrCustomizer<https://github.com/MrCustomizer>, @franciscoruiz<https://github.com/franciscoruiz>, @elyulka<https://github.com/elyulka>

aphelps, 2015-03-04:
What's the advantage of this new command to just using 'platformio run' with appropriate options?

ivankravets, 2015-03-05:
Dear @pelikhan ,

> What about those prompts? They are complete show stopper right now.

 There are a few options to disable prompts:
* using environment variable [PLATFORMIO_SETTING_ENABLE_PROMPTS=No](http://docs.platformio.org/en/latest/envvars.html#envvar-platformio-setting-enable-prompts)
* disable global setting ``enable_prompts`` via [platformio settings command](http://docs.platformio.org/en/latest/userguide/cmd_settings.html#cmd-settings)
* masking under Continuous Integration system via environment variable [CI=true](http://docs.platformio.org/en/latest/envvars.html#envvar-ci).

> Travis support sounds great however it should limited to it. There are many other build systems.

Which others systems you could propose?

> Travis supports a feature to cache certain folders, like npm-modules. You should update your example: installing a complete platformio drop on every build might quickly turn out costly for you.

@travis-ci doesn't allow to cache "something" for free account. This feature is available only for the premium accounts.

ivankravets, 2015-03-05:
Dear @aphelps,

> What's the advantage of this new command to just using 'platformio run' with appropriate options?

``platformio ci`` command will consist from ``init + source/libs GLOB finder + run`` commands. This command is useful for the projects which want to be tested under different embedded platforms, but don't want to keep project structure in PlatformIO style. The simple example is Arduino library. The most libs like to keep in root of repository "library source code" and test code in "examples" folder. 

Therefore, developers of libraries can add the only single `.travis.yml` or `circle.yml` file to repo and PlatformIO will test all examples. Another way is to dynamically perform "bash" commands with `platformio init + cp files + platformio run`.

If you like to keep your project in PlatformIO compatible structure, the you can use "platformio run" command for Continuous Integration systems.

pelikhan, 2015-03-05:
Jenkins is another one.

aphelps, 2015-03-06:
Yeah, Jenkins is what I typically use for more serious projects.

Also, thanks for the clarification on what all 'platformio ci' will be doing, that makes sense.

commit 510c4dec (2015-05-07): PlatformIO as Continuous Integration tool for embedded projects // Resolve #108
closed 2015-05-15 by ivankravets