platformio/platformio-core #111: YCM/Vim integration

ajford, 2015-03-06:
In the IDE Integration section of the documentation, you mention pairing Vim with YouCompleteMe and Syntastic. Can you expand on this? I have tried setting up the `.ycm_extra_conf.py` with 

```python
flags = [ 
  '-Wall'
  ,'-c'
  ,'-g'
  ,'-MMD'
  ,'-I/home/aford/.platformio/packages/framework-arduinoavr/cores/arduino'
  ,'-I/home/aford/.platformio/packages/framework-arduinoavr/variants/standard'
  ,'-mmcu=atmega328p'
  ,'-DF_CPU=16000000L'
  ,'-DARDUINO_ARCH_AVR'
  ,'-DARDUINO_AVR_DUEMILANOVE'
  ,'-DARDUINO=106000'
]  
```

and I get no results for completion, even when forcing completion.

ajford, 2015-03-06:
Ok, I've figured it out. it's mostly a goof on my part. I'm cleaning up all the comments and such I made in my `.ycm_extra_conf.py` and will post a gist with it for those who follow.

ivankravets, 2015-03-06:
Could I ask you to share your final config? I hope, it will be useful for other users. Thanks! :+1:

ajford, 2015-03-09:
Hey @ivankravets, just finished hammering out everything (and now satisfied that it looks good). 

Available as a gist: https://gist.github.com/ajford/f551b2b6fd4d6b6e1ef2

Comments should hopefully be self explanatory, but I can try to flesh out a setup guide if needed. Also, I have absolutely no clue how compatible this config is with Windows. I use Debian/Linux only, and YCM doesn't officially support Windows.

closed 2015-03-09 by ajford

ivankravets, 2015-03-09:
Dear @ajford,

This is excellent example!!! :+1:  I've just added your Gist to main documentation: http://docs.platformio.org/en/latest/ide.html#vim

Please don't remove it :blush:

ajford, 2015-03-09:
No problem. Glad you find it useful. I develop heavily in Python and was sorely missing YCM. 

I'm thinking about finding a way to hook into the info available in the `platformio.ini` file (or within `platformio` itself. Right now, the file is very Arduino-centric, and requires a bit of manual rewriting if you need to switch platforms. I plan on doing some work with the TI Launchpad MSP430 soon, so I might put together a repo with a few preconfigured  YCM configs for various platforms. I'll ping you with an update if I do.

ivankravets, 2015-03-09:
> I'm thinking about finding a way to hook into the info available in the platformio.ini file (or within platformio itself. Right now, the file is very Arduino-centric, and requires a bit of manual rewriting if you need to switch platforms.

Can you explain me what do you mean? :blush: 

> I plan on doing some work with the TI Launchpad MSP430 soon, so I might put together a repo with a few preconfigured YCM configs for various platforms. I'll ping you with an update if I do.

I would ask you to put this example to https://github.com/ivankravets/platformio/tree/develop/examples/ide/vim/

This information will be interested for PlatformIO users.

ajford, 2015-03-09:
>> I'm thinking about finding a way to hook into the info available in the platformio.ini file (or within  platformio itself. Right now, the file is very Arduino-centric, and requires a bit of manual rewriting if you need to switch platforms.

>Can you explain me what do you mean? ￼:blush:

Internally, `platformio` knows what libraries/toolchains/frameworks correspond to each platform. If I switch to the `timsp430` platform, then the framework location changes (i.e. from `~/.platformio/packages/framework-arduinoavr/libraries` to `~/.platformio/packages/framework-energiamsp430`), the "core" and "std lib" locations change, etc. 

You could have a platform specific `.ycm_extra_conf.py` files, or you could dynamically generate them based on the platform in use. I don't know witch is better in the long run. 

> I would ask you to put this example to https://github.com/ivankravets/platformio/tree/develop/examples/ide/vim/

No problem. I'll fork and create a pull request as I get the chance to document everything.