platformio/platformio-core #117: Add support for panStamp AVR

gandy92, 2015-03-11:
Is it possible to add support for the panStamp AVR and panStamp NRG boards?
* Vendor: http://www.panstamp.com/
* panStamp AVR: https://github.com/panStamp/panstamp/wiki/panStamp%20AVR.-Technical%20details
* panStamp NRG: https://github.com/panStamp/panstamp/wiki/panStamp%20NRG.-Technical%20details
* Libraries for panStamp specific hardware: https://github.com/panStamp/panstamp

closed 2015-03-16 by valeros

gandy92, 2015-03-22:
Thank you for adding support for panstamp so promptly :+1:

I managed to successfully build the panstamp-blink.ino. However, I have a few issues compiling a sketch that uses the Wire lib, similar to what is mentioned here:  http://www.panstamp.org/forum/archive/index.php/thread-3978.html  Apparently, using the panstamp libraries requires a few patches to the libraries shipped with arduino.

Should I open a new issue covering this?

gandy92, 2015-03-22:
I looked into this more closely, apparently they just rename twi.c from libraries/Wire/utility to twi.cpp to make it work. Would this be a viable solution?

ivankravets, 2015-03-22:
@gandy92 thanks for report. I've just opened new issue #133 . It should resolve this problem.

P.S: You can currently avoid this problem via adding specific libs to your project ``libs`` directory. These libs have more higher priority then framework's libs. Just copy PanStamp's ``Wire`` lib to projects ``libs/Wire``. Did it help you?

gandy92, 2015-03-22:
Thanks for the workaround. For a preliminary fix I manually copied the libraries to the platformio directories. Your solution is much more elegant, though, I will try this with a clean installation, tomorrow.

ivankravets, 2015-03-22:
Ok, I will wait for your results.

P.S: Please restore default PlatformIO packages. Just uninstall/install specified platforms or remove totally ``~/.platformio`` dir.

gandy92, 2015-03-22:
Removed .platformio dir and copied panstamp-avr specific libraries to lib/

Worked fine for both of my projects. :+1:

PR #124 (2015-03-16): Add support for "panStamp" boards // Resolve #117
* Rename "opencm3" framework to "libopencm3" // Resolve #120
commit 744c6f12 (2015-03-16): Merge pull request #124 from valeros/develop

Add support for "panStamp" boards // Resolve #117
commit ccc411fb (2015-03-16): Add support for "panStamp" boards // Resolve #117