platformio/platformio-core #119: Support for ESP8266

makefu, 2015-03-12:
i would love to see support for the ESP8266 `wifi` controller which i am really excited about because it really packs a punch for $2.5.
The toolchain can be obtained via https://github.com/pfalcon/esp-open-sdk

Thanks for this great project

ivankravets, 2015-03-12:
Dear @makefu,

Thank a lot for your request. We have already received the requests from the users about ESP8266. Now I understand, that we should add this development platform to PlatformIO list.

A few questions:

1. Do you any board based on this chip?
2. Which operation system do you use?
3. Are you interested in help us with building toolchain for multiple host targets? (Mac OS X, Windows, Linux 32/64, Linux ARM). It should look quite easy according your link above (``make bla bla bla``).
4. How does look "uploading" process for this chip `SoC`?

makefu, 2015-03-12:
Hi Ivan,

Regarding you questions
1. Yes, i do have a dozen or so of these chips laying around. Due to the fact that they can be sourced extremely cheaply from china (aliexpress: http://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20150312025300&SearchText=esp8266 ebay:http://www.ebay.com/sch/i.html?_nkw=esp8266 ) it should not be a problem to obtain one. If you wish i can donate a couple for the controllers and send them to a address of your choice :)
2. I am currenly developing on linux and am willing to support this project with my knowledge if i can
3. Unfortunately linux is my only weapon of choice
4. The uploading is done by booting the uC in firmware upload mode (pull a pin to GND) and using https://github.com/themadinventor/esptool to upload the firmware

ivankravets, 2015-03-12:
1. Thanks. If we add the support for this chip, then I will ask you to test it :smile: 
2. How do you develop this chip on Linux now? Which type of toolchain do you use? Where is located SDK? Can you share here/or in separate Gist some "Hello World" project?
3. Which type of Linux distributive?
4. Thanks for information. Looks good.

makefu, 2015-03-12:
1. yay!
2. under linux/cygwin you will need https://github.com/pfalcon/esp-open-sdk which is a consolidation of everything you need (Xtensa lx106 architecture toolchain built via crosstool-ng and ESP8266 IoT SDK from Espressif Systems)
   - the detailed description of what you need to build the toolchain: https://github.com/esp8266/esp8266-wiki/wiki/Toolchain
   - a fast-forward method is to use https://github.com/pfalcon/esp-open-sdk which does all the heavy lifting
after that you can use the lx106 compiler with the SDK like this:

    $ xtensa-lx106-elf-gcc -I$(THISDIR)/sdk/include -L$(THISDIR)/sdk/lib

- this is a hello world with a makefile: https://github.com/leon-anavi/esp-hello-world
- this is a transparent serial bridge for the esp to connect to other uCs https://github.com/beckdac/ESP8266-transparent-bridge
  both are using the sdk and the xtensa compiler
3. i am running archlinux but i have access to ubuntu and debian unstable for testing stuff
4. !

makefu, 2015-03-19:
just fyi, there hackaday published an article about the esp8266 which is very detailed and gives some background about the ESP8266: http://hackaday.com/2015/03/18/how-to-directly-program-an-inexpensive-esp8266-wifi-module/

ivankravets, 2015-03-23:
Additional article: http://hackaday.com/2015/03/18/how-to-directly-program-an-inexpensive-esp8266-wifi-module/

chipolux, 2015-04-01:
Just wanted to chime in that I've been working with the ESP8266 using [esptool.py](https://github.com/themadinventor/esptool) and a [docker buildbox](https://github.com/chipolux/esp8266-buildbox) and it would be really nice to have support for this instead of swapping back and forth as I'm currently doing.

If there is any way I could help I'd be glad to know! Thanks!

ivankravets, 2015-04-01:
@chipolux Sure! Please repost/retweet our vote https://twitter.com/platformio_org/status/583353126494822402

ivankravets, 2015-04-03:
Dear @makefu, @chipolux,

You can try the latest ``develop`` version of PlatformIO, where have been released the initial support of ``espressif`` platform. See here [how to install development version of PlatformIO](http://docs.platformio.org/en/latest/installation.html#development-version).

We know about a few issues with invalid firmware and we are working on it. Do you have any problems with your ESP8266 based projects?

See examples https://github.com/platformio/platformio/tree/develop/examples/espressif

P.S: Thanks a lot @valeros for the great job on this issue! :+1: :beers:

chipolux, 2015-04-04:
Awesome @ivankravets!

I'll take a look at it later today with a few projects on OSX and Linux!

chipolux, 2015-04-05:
Alright, so it pulls down and starts up okay but there's a problem with linking:

```
xtensa-lx106-elf-g++ -o .pioenvs/esp8266/firmware.elf -nostdlib -Wl,--no-check-section -u call_user_start -Wl,-static -T /Users/nakyle/.platformio/packages/ldscripts/esp8266.ld .pioenvs/esp8266/src/piomain.o -L/Users/nakyle/.platformio/packages/framework-arduinoespressif/sdk/lib -L.pioenvs/esp8266 -Wl,--start-group -lpp -lmain -lwpa -llwip -lnet80211 -lphy -lhal .pioenvs/esp8266/libFrameworkArduino.a .pioenvs/esp8266/libFrameworkArduinoVariant.a -lgcc -lc -Wl,--end-group
/Users/nakyle/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file /Users/nakyle/.platformio/packages/ldscripts/esp8266.ld: No such file or directory
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/esp8266/firmware.elf] Error 1
```
I've completely wiped out my ~/.platformio directory and reinstalled platformio and scons but still no go.

chipolux, 2015-04-05:
Oh, I suppose you guys might want the platformio.ini I'm working with:

```
[env:esp8266]
platform = espressif
framework = arduino
board = esp01
```

chipolux, 2015-04-05:
Otherwise though, using the native libraries it seems to compile all my little projects and push them to the chip just fine.

Is there a reason you guys went with [esptool](https://github.com/tommie/esptool-ck) over [esptool.py](https://github.com/themadinventor/esptool)? I've not ran into any issues with esptool.py so far and it has friendlier output when flashing the chip.

ivankravets, 2015-04-05:
> Otherwise though, using the native libraries it seems to compile all my little projects and push them to the chip just fine.

Great! :+1:  Could I ask you to try this example: https://github.com/platformio/platformio/tree/develop/examples/espressif/esp8266-webserver ?

> Is there a reason you guys went with esptool over esptool.py? I've not ran into any issues with esptool.py so far and it has friendlier output when flashing the chip.

I agree with you.

@valeros Could I ask you to switch to https://github.com/themadinventor/esptool ? This is Python based utility and should work under all host OS.

chipolux, 2015-04-05:
@ivankravets tried with that example and still got the same error, esp266.ld doesn't exist.

ivankravets, 2015-04-05:
@chipolux give me please an output of these commands:
```shell
$ platformio --version
$ platformio update
```

Thanks!

chipolux, 2015-04-05:
Here ya go!

```
nakyle at lnakyle in ~/Projects/embedded/esp8266/platformio/webserver (embedded)
$ platformio --version
PlatformIO, version 1.4.0.dev0

nakyle at lnakyle in ~/Projects/embedded/esp8266/platformio/webserver (embedded)
$ platformio update

Platform espressif
--------
Updating sdk-esp8266 package:
Versions: Current=1, Latest=1    [Up-to-date]
Updating tool-esptool package:
Versions: Current=1, Latest=1    [Up-to-date]
Updating framework-arduinoespressif package:
Versions: Current=1, Latest=1    [Up-to-date]
Updating toolchain-xtensa package:
Versions: Current=1, Latest=1    [Up-to-date]
```

ivankravets, 2015-04-06:
@chipolux Thanks, I got it and I've just fixed. Please [re-install](http://docs.platformio.org/en/latest/installation.html#development-version) ``develop`` version of PlatformIO. Do you have any problems?

makefu, 2015-04-06:
Hi, i tried the example code at https://github.com/platformio/platformio/tree/develop/examples/espressif/esp8266-webserver and it seems to building a nice little firmware:
```
saved binimage file, total size is 42304 bytes, checksum byte is 0x5E
using ELF file ".pioenvs/esp01_8266/firmware.elf"
saved section ".irom0.text" to file ".pioenvs/esp01_8266/firmware_40000.bin"
"xtensa-lx106-elf-size" -B -d .pioenvs/esp01_8266/firmware.elf
text	   data	    bss	    dec	    hex	filename
192653	   4684	  36336	 233673	  390c9	.pioenvs/esp01_8266/firmware.elf
====================================================================== [SUCCESS] Took 3.62 seconds ======================================================================
```

Thanks for your efforts, really!

ivankravets, 2015-04-06:
@makefu Could I ask to upload firmware to device? We have problem with invalid firmware. Like, "doesn't work properly". We are working on it.

chipolux, 2015-04-06:
Awesome, looks like the new version works! I'll don't have a chip handy right now but in a few hours I'll make sure it can push this and another project or two!

using ELF file ".pioenvs/esp01_8266/firmware.elf"
saved section ".irom0.text" to file ".pioenvs/esp01_8266/firmware_40000.bin"
"xtensa-lx106-elf-size" -B -d .pioenvs/esp01_8266/firmware.elf
text       data     bss     dec     hex filename
192653     4716   36336  233705   390e9 .pioenvs/esp01_8266/firmware.elf
====================== [SUCCESS] Took 106.00 seconds =======================

chipolux, 2015-04-06:
Webserver flashed perfectly. Trying my own couple of arduino framework projects now.

```
nakyle at lnakyle in ~/Projects/embedded/esp8266/platformio/webserver (embedded)
$ platformio run --target upload
[Mon Apr  6 17:03:48 2015] Processing esp01_8266 (platform: espressif, board: esp01, framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/esp01_8266/firmware_00000.bin", ".pioenvs/esp01_8266/firmware_40000.bin"])
Auto-detected UPLOAD_PORT: /dev/cu.SLAB_USBtoUART
/Users/nakyle/.platformio/packages/tool-esptool/esptool -vv -cd none -cb 115200 -cp /dev/cu.SLAB_USBtoUART -ca 0x00000 -cf .pioenvs/esp01_8266/firmware_00000.bin -ca 0x40000 -cf .pioenvs/esp01_8266/firmware_40000.bin
esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to none
setting baudrate from 115200 to 115200
setting port from /dev/tty.usbserial to /dev/cu.SLAB_USBtoUART
setting address from 0x00000000 to 0x00000000
espcomm_upload_file
stat .pioenvs/esp01_8266/firmware_00000.bin success
opening port /dev/cu.SLAB_USBtoUART at 115200
tcgetattr
tcsetattr
serial open
opening bootloader
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_cmd: sending command header
espcomm_cmd: sending command payload
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_cmd: sending command header
espcomm_cmd: sending command payload
espcomm cmd: receiving 2 bytes of data
setting character timeout 0
done
setting character timeout 1
done
espcomm_open
Uploading 42336 bytes from .pioenvs/esp01_8266/firmware_00000.bin to flash at 0x00000000
erasing flash
espcomm_cmd: sending command header
espcomm_cmd: sending command payload
setting timeout 5000
setting character timeout 50
done
setting timeout 1
setting character timeout 1
done
espcomm cmd: receiving 2 bytes of data
writing flash
..........................................
setting address from 0x00000000 to 0x00040000
espcomm_upload_file
stat .pioenvs/esp01_8266/firmware_40000.bin success
espcomm_open
Uploading 155080 bytes from .pioenvs/esp01_8266/firmware_40000.bin to flash at 0x00040000
erasing flash
espcomm_cmd: sending command header
espcomm_cmd: sending command payload
setting timeout 5000
setting character timeout 50
done
setting timeout 1
setting character timeout 1
done
espcomm cmd: receiving 2 bytes of data
writing flash
........................................................................................................................................................
starting app without reboot
espcomm_cmd: sending command header
espcomm_cmd: sending command payload
espcomm cmd: receiving 2 bytes of data
closing bootloader
=========================================================================== [SUCCESS] Took 21.76 seconds ===========================================================================
```

ivankravets, 2015-04-06:
@chipolux, Does device response on GET request? Could I ask you to check it using browser?

chipolux, 2015-04-06:
Sure, I've flashed the webserver example back to the chip but I'm just getting junk from the serial port. I've tried adjusting the baud rate with `platformio serialports monitor` as well as adjusting it in the example but I've not got anything to work there.

Any ideas?

chipolux, 2015-04-06:
Weird, just testing with this bit of code works, but `serialports monitor` crashes with a device not configured error after the first line comes through. I can still `cat /dev/cu.SLAB_USBtoUART` and watch the messages flow though.

```
const int led = 2;
 
void setup(void)
{
  Serial.begin(9600);
  pinMode(led, OUTPUT);
  digitalWrite(led, LOW);
}
 
void loop(void)
{
    Serial.println("Just a test.");
    digitalWrite(led, HIGH);
    delay(500);
    digitalWrite(led, LOW);
    delay(500);
} 
```

chipolux, 2015-04-07:
Well I isolated an issue with my usb-to-uart adapter that was causing some issues.

But I also figured out that something about ESP8266WebServer causes a fatal exception on the chip.

```
Fatal exception (0):
epc1=0xion (0):
epc1=0x40108ca4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0):
epc1=0xion (0):
epc1=0x40108ca4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```

chipolux, 2015-04-07:
Looks like it must be a problem with the firmware the toolchain is generating as I've tried with the [arduino ide](https://github.com/esp8266/Arduino) and it's working from there.

ivankravets, 2015-04-08:
Friends, please [upgrade PlatformIO](http://docs.platformio.org/en/latest/installation.html#development-version) to the latest ``develop`` and update all packages using ``> platformio update``.

Please check your version:
```shell
> platformio --version
```

Should be equal to ``1.4.0.dev2``.

Looks like we fixed the bug with "Fatal exception". 

P.S: @gandy92 please re-test with your board. Thanks!

chipolux, 2015-04-08:
@ivankravets Everything seems to be working! I've tested out the wifi scan and the wifi server examples and both worked flawlessly!

ivankravets, 2015-04-08:
@chipolux Thanks for report! We will replace "esptool" with "esptool.py" and I'm going to release ``1.4.0`` this tomorrow.

chipolux, 2015-04-08:
@ivankravets @valeros Awesome! You guys rock! This will make my workflow a million times easier!

ivankravets, 2015-04-08:
@chipolux Thanks! :blush: 

Are you interested in native ESP8266 programming using SDK (without Arduino framework)?

gandy92, 2015-04-08:
@ivankravets Sorry, didn't yet get my programming gear for the ESP8266 to work properly, not a problem with PlatformIO, though..

ivankravets, 2015-04-08:
@gandy92  try connect/disconnect. We also have problems with this chip time by time.

chipolux, 2015-04-08:
@ivankravets I've played with the native SDK a bit when I first picked up the chips but I haven't had a need to use it instead of the Arduino libs yet..

Most people doing stuff with the chip are either using NodeMCU or the Arduino libs as far as I can tell so I doubt many people will really need the native support but I imagine it'd be good to have just in case.

closed 2015-04-11 by valeros

electronicsguy, 2015-09-21:
@ivankravets are you working on integrating platform.io support within the Arduino IDE environment? That would be very useful.

ivankravets, 2015-09-22:
@electronicsguy see http://docs.platformio.org/en/latest/ide/arduino.html

kireevco, 2015-12-19:
I'm working on bulding a truly cross-platform toolchain, since esp-open-sdk is not building well on windows: github.com/kireev.co/esp-alt-sdk.

ivankravets, 2015-12-22:
@kireevco does https://github.com/esp8266/Arduino uses your toolchain?

kireevco, 2015-12-22:
@ivankravets no, not yet, I'm working with them on that matter.
https://github.com/sminghub/sming does work well already. By the way, it would be very nice to add sming to platform.io.

ivankravets, 2015-12-22:
@kireevco please open a separate issue for switching to https://github.com/kireevco/esp-alt-sdk when you will agreed it with https://github.com/esp8266/Arduino. We currently switched to their toolchain for compatibility.

As for Sming - https://github.com/platformio/platformio/issues/398. Thanks!

commit 0486348f (2015-04-11): Merge pull request #169 from valeros/develop

Add support for ESP8266 // Resolve #119
commit 181959f1 (2015-04-11): Add support for ESP8266 // Resolve #119