platformio/platformio-core #122: Serial doesn't work for mbed framework

ivankravets, 2015-03-13:
Example
```c
#include "mbed.h"

Serial pc(USBTX, USBRX); // tx, rx

int main() {
    pc.printf("Hello World!\n\r");
    while(1) {
        pc.putc(pc.getc() + 1); // echo input back to terminal
    }
}
```

```shell
(develop)➜  mbed-blink git:(develop) ✗ platformio run -e nucleo_f401re -t upload
[Fri Mar 13 23:04:44 2015] Processing nucleo_f401re (platform: ststm32, board: nucleo_f401re, framework: mbed)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/nucleo_f401re/src/main.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -mfpu=fpv4-sp-d16 -fomit-frame-pointer -Wall -mfloat-abi=softfp -MMD -mcpu=cortex-m4 -DTARGET_NUCLEO_F401RE -DTARGET_M4 -DTARGET_CORTEX_M -DTARGET_STM -DTARGET_STM32F4 -DTARGET_STM32F401RE -DTOOLCHAIN_GCC_ARM -DTOOLCHAIN_GCC -D__CORTEX_M4 -DARM_MATH_CM4 -D__FPU_PRESENT=1 -DMBED_BUILD_TIMESTAMP=1426262453.27 -D__MBED__=1 -DTARGET_FF_ARDUINO -DTARGET_FF_MORPHO -DPLATFORMIO=010200 -I.pioenvs/nucleo_f401re/FrameworkMbedInc248832578 -I.pioenvs/nucleo_f401re/FrameworkMbedInc-213564679 -I.pioenvs/nucleo_f401re/FrameworkMbedInc-250686161 -I.pioenvs/nucleo_f401re/FrameworkMbedInc1266984661 -I.pioenvs/nucleo_f401re/FrameworkMbedInc2097404638 -I.pioenvs/nucleo_f401re/FrameworkMbedInc-1116590620 -I.pioenvs/nucleo_f401re/FrameworkMbedInc170856846 -I.pioenvs/nucleo_f401re/FrameworkMbedInc175333816 .pioenvs/nucleo_f401re/src/main.cpp
arm-none-eabi-ar rcs .pioenvs/nucleo_f401re/libFrameworkMbed.a @/var/folders/nj/nt38kxwd2xxfl5l42np_w7qw0000gn/T/pioarargs-f8d149b23e80c5a2dfe687bd465da31f
arm-none-eabi-ranlib .pioenvs/nucleo_f401re/libFrameworkMbed.a
arm-none-eabi-g++ -o .pioenvs/nucleo_f401re/firmware.elf -Wl,--gc-sections -Wl,--wrap,main -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp --specs=nano.specs -T /Users/ikravets/.platformio/packages/framework-mbed/variant/NUCLEO_F401RE/mbed/TARGET_NUCLEO_F401RE/TOOLCHAIN_GCC_ARM/STM32F401XE.ld .pioenvs/nucleo_f401re/src/main.o -L/Users/ikravets/.platformio/packages/framework-mbed/variant/NUCLEO_F401RE/mbed/TARGET_NUCLEO_F401RE/TOOLCHAIN_GCC_ARM -L.pioenvs/nucleo_f401re -Wl,--start-group -lc -lgcc -lm -lmbed -lstdc++ -lsupc++ -lnosys .pioenvs/nucleo_f401re/libFrameworkMbed.a -Wl,--end-group
.pioenvs/nucleo_f401re/libFrameworkMbed.a(retarget.o): In function `__gnu_cxx::__verbose_terminate_handler()':
retarget.cpp:(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x0): multiple definition of `__gnu_cxx::__verbose_terminate_handler()'
/Users/ikravets/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/softfp/libstdc++_s.a(vterminate.o):vterminate.cc:(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x0): first defined here
/Users/ikravets/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/softfp/libc_s.a(lib_a-signalr.o): In function `_kill_r':
signalr.c:(.text._kill_r+0xe): undefined reference to `_kill'
/Users/ikravets/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/softfp/libc_s.a(lib_a-signalr.o): In function `_getpid_r':
signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid'
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/nucleo_f401re/firmware.elf] Error 1
================================================================= [ ERROR ] Took 1.35 seconds =================================================================
```

closed 2015-03-16 by valeros

hoosierEE, 2016-05-26:
I came here searching for `_getpid` because I have a similar error, but my code is totally different:

Here's the source code:
```c++
// project/src/main.cpp
#include <Arduino.h>
#include <functional>

void method(std::function<int (int)>reader){
    reader(3);// read pin 3
}

void setup(){}
void loop(){
    method(touchRead);
}
```

Here's the platformio configuration:

```ini
# project/project.ini
[env:teensylc]
platform = teensy
framework = arduino
board = teensylc
```

Here's the error:

```
$ platformio run
[Thu May 26 13:47:49 2016] Processing teensylc (platform: teensy, board: teensylc, framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/teensylc/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m0plus -Wl,--defsym=__rtc_localtime=1
464284869 -fsingle-precision-constant --specs=nano.specs -Wl,-T"mkl26z64.ld" .pioenvs/teensylc/src/main.o -L/Users/lx/.platformio/packages/l
dscripts -L.pioenvs/teensylc -Wl,--start-group -larm_cortexM0l_math .pioenvs/teensylc/libFrameworkArduino.a -lc -lgcc -lm -Wl,--end-group  
/Users/lx/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv6-m/libc_s.a(li
b_a-signalr.o): In function `_kill_r':                                                                                                     
signalr.c:(.text._kill_r+0xe): undefined reference to `_kill'
/Users/lx/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv6-m/libc_s.a(li
b_a-signalr.o): In function `_getpid_r':                                                                                                   
signalr.c:(.text._getpid_r+0x2): undefined reference to `_getpid'
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/teensylc/firmware.elf] Error 1
======================================================= [ ERROR ] Took 0.51 seconds =======================================================
```

valeros, 2016-05-27:
Hi @hoosierEE !
Does your code compile without error in Teensyduino?

hoosierEE, 2016-05-27:
Hi @valeros no, it complains that it can't find `functional`, which is expected since it's a C++ standard library and not included in the Arduino toolchain.

valeros, 2016-05-27:
Hmm, could you please add these stubs to your code? Does it work?
```
extern "C"{
  int _getpid(){ return -1;}
  int _kill(int pid, int sig){ return -1; }
  int _write(){return -1;}
}
```

hoosierEE, 2016-06-07:
Hi @valeros it does work when I add those stubs, thanks!

commit 39f052a5 (2015-03-16): Merge pull request #123 from valeros/develop

Fix mbed framework // Resolve #122
commit 57b84789 (2015-03-16): Fix "mbed" framework // Resolve #122