platformio/platformio-core #125: Unable to install stm32 platform

garacio, 2015-03-19:
Hi!

I am trying to install an stm32 library.

On the version from pip i have an error
```
$ platformio install stm32
Installing toolchain-gccarmnoneeabi package:
Already installed
Installing tool-stlink package:
Already installed
Installing framework-spl package:
Already installed
Installing framework-cmsis package:
Already installed
Installing framework-opencm3 package:
Error: Detected unknown package 'framework-opencm3'
```

On the git version (branches master and development) i have another error

```
$ platformio install ststm32
Installing framework-libopencm3 package:
Already installed
Installing toolchain-gccarmnoneeabi package:
Already installed
Installing tool-stlink package:
Already installed
Installing framework-spl package:
Already installed
Installing framework-cmsis package:
Already installed
Installing framework-mbed package:
Downloading  [####################################]  100%
Error: The 'sha1' sum 'fd8d432765799ab2b88e643b315653b447bd0a18' of downloaded file 'framework-mbed-3.tar.gz' is not equal to remote 'f646956887ec3837b19c21e0e4b642152052892c'
```

ivankravets, 2015-03-19:
Sorry for this issue. We are going to release new PlatformIO 1.2 today, but I have problem with SourceForge storage. I can't upload "framework-mbed" :( The upload speed is crazy low ;(

I will report here when I finish uploading.

ivankravets, 2015-03-19:
Fuhh... I've just uploaded ``framework-mbed`` package. Please try again to install ``ststm32`` platform.

garacio, 2015-03-19:
A have installed ststm32 platform now

```
$ platformio install ststm32
Installing framework-libopencm3 package:
Already installed
Installing toolchain-gccarmnoneeabi package:
Already installed
Installing tool-stlink package:
Already installed
Installing framework-spl package:
Already installed
Installing framework-cmsis package:
Already installed
Installing framework-mbed package:
Already installed
Installing ldscripts package:
Already installed
The platform 'ststm32' has been successfully installed!
```

But 

```
$ platformio list
/Library/Python/2.7/site-packages/requests/packages/urllib3/util/ssl_.py:79: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
```

And trying to run

```
$ platformio run
[Fri Mar 20 01:21:00 2015] Processing base (platform: ststm32, board: disco_f429zi, framework: cmsis,spl)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
/Library/Python/2.7/site-packages/requests/packages/urllib3/util/ssl_.py:79: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
The platform 'ststm32' has not been installed yet. Would you like to install it now? [y/N]: y
Installing framework-libopencm3 package:
Already installed
Installing toolchain-gccarmnoneeabi package:
Already installed
Installing tool-stlink package:
Already installed
Installing framework-spl package:
Already installed
Installing framework-cmsis package:
Already installed
Installing framework-mbed package:
Already installed
Installing ldscripts package:
Already installed
The platform 'ststm32' has been successfully installed!
Error: The platform 'ST STM32' has not been installed yet. Use `platformio install` command
/Library/Python/2.7/site-packages/requests/packages/urllib3/util/ssl_.py:79: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
```

ivankravets, 2015-03-19:
Please ignore this warning. I fixed it in this commit https://github.com/ivankravets/platformio/commit/3fabd01e1bbe42f1efcfe27e0bf1714b515e75c9

We will release PlatformIO 1.2.0 within several hours, where this warning is suppressed.

ivankravets, 2015-03-19:
Opss. Sorry, I've not seen this error. Hm... Could I ask you to uninstall ``ststm32`` platform and install again?

```shell
> platformio uninstall ststm32
> platformio install ststm32
```

Thanks in advance!

garacio, 2015-03-19:
This warning is no matter,

```
$ platformio uninstall ststm32
Error: The platform 'ST STM32' has not been installed yet. Use `platformio install` command

$ platformio install ststm32
Installing framework-libopencm3 package:
Already installed
Installing toolchain-gccarmnoneeabi package:
Already installed
Installing tool-stlink package:
Already installed
Installing framework-spl package:
Already installed
Installing framework-cmsis package:
Already installed
Installing framework-mbed package:
Already installed
Installing ldscripts package:
Already installed
The platform 'ststm32' has been successfully installed!
```

And i have tried to delete ~/.platformio directory. Files have been downloaded and installed successfully, but the problem is still here.

ivankravets, 2015-03-19:
Could I ask you to share here ``~/.platformio/appstate.json``?

garacio, 2015-03-19:
Sure

```
$ cat ~/.platformio/appstate.json
{"installed_platforms": ["ST STM32"], "last_version": "1.1.0", "installed_packages": {"framework-libopencm3": {"version": 1, "time": 1426803313}, "toolchain-gccarmnoneeabi": {"version": 1, "time": 1426803392}, "tool-stlink": {"version": 1, "time": 1426803394}, "framework-spl": {"version": 1, "time": 1426803397}, "framework-cmsis": {"version": 2, "time": 1426803399}, "framework-mbed": {"version": 3, "time": 1426803476}, "ldscripts": {"version": 1, "time": 1426803478}}, "last_check": {"platformio_upgrade": 1426803309, "platforms_update": 1426803309, "libraries_update": 1426803309}}
```

ivankravets, 2015-03-19:
I see where is this error. Give me please output of:
```shell
platformio --version
```

ivankravets, 2015-03-19:
Looks, like invalid PlatformIO installation. Please make these steps:

1. Remove ``~/.platformio/``
2. Uninstall PlatformIO
```shell
$ pip uninstall platformio
```
3. Install latest `develop`` version:
```shell
$ pip install https://github.com/ivankravets/platformio/archive/develop.zip
```

garacio, 2015-03-20:
All works fine now, thanks!

closed 2015-03-20 by garacio

ivankravets, 2015-03-20:
Sorry for this issue. Don't forget to upgrade to stable version today! :)

garacio, 2015-03-20:
Ok, thanks!

ivankravets, 2015-03-20:
Is it yours post from login ``locofreek25``? http://www.reddit.com/r/embedded/comments/2xuec4/platformio_as_continuous_integration_tool_for/