platformio/platformio-core #132: How to configure CLion Ptoject

bkuberek, 2015-03-22:
Apologies if this is the wrong place to ask this question but I am trying get my current platformio project working on CLion.

I have a working platformio project that is getting bigger and difficult to manage and I was hoping to use CLion IDE. Any help is appreciated.

ivankravets, 2015-03-22:
You can integrate PlatformIO into any IDE. Which type of problem do you have? From CLion site:

> CLion supports the well-known CMake build system

You can replace built-in ``cmake`` system with ``platformio run`` command. Could I ask you to write an article about it and I will place it to main [PlatformIO documentation](http://docs.platformio.org/en/latest/ide.html)?

Thanks in advance!

ivankravets, 2015-03-22:
Here is interesting example how to replace built-in Eclipse builder with PlatformIO: [Building and debugging Atmel AVR (Arduino-based) project using Eclipse IDE+PlatformIO](http://www.ikravets.com/computer-life/programming/2014/06/20/building-and-debugging-atmel-avr-arduino-based-project-using-eclipse-ideplatformio)

aphelps, 2015-03-22:
I'd be interested in any results on this as well, I use IntelliJ regularly and have considered trying out CLion.  If it supports a any standard makefile system you could always put together a config that mapped to platformio commands, if CLion doesn't let you do that directly within its preferences.

ivankravets, 2015-03-22:
Friends,

Who can help PlatformIO with short manual with pictures how to configure CLion and PlatformIO? It can be any format: PDF/DOC/RTF/etc., the we will convert it to RST and put to core documentation with preserving author's rights.

We could do it, but you can see how much issues we have opened. In this situation I want to concentrate our forces to the new features for PlatformIO.

bkuberek, 2015-03-22:
This is the first time I do something with C/C++ and I am still wrapping my head around basic concepts. I would gladly contribute this write up if I could figure it out on my own. I hope someone has already gone through the trouble and could share their setup.

ivankravets, 2015-03-27:
@valeros has tried to configure CLion with PlatformIO, but didn't received any results. They don't allow to redefine own builder. I've just posted the comments to their issues tracker:

https://youtrack.jetbrains.com/issue/CPP-871#comment=27-960431

Here is explanation by CLion owners: http://blog.jetbrains.com/clion/2014/09/cmake-vs-the-others-round-1/

bkuberek, 2015-03-28:
thanks for the update

jimbojsb, 2015-07-16:
Just tried this feature out, seems like it generates a solid CMakeLists.txt but trying to build from inside clion pops a dialog asking for an executable. It seems like I can specify any old executable (i picked the platformio file). The build will then be allowed to run, and it definitely uploads the code to my board, but it then replaces the output on the console with the standard "platformio" help message. Also, when importing the project that platformio generates, every file except CMakeLists.txt is "grayed out" in the project pane, as in CLion doesn't think it's part of the project. Thoughts?

ivankravets, 2015-07-16:
@jimbojsb which PlatformIO version do you use? I've just re-tested it under Mac. See my results:

[image]

I used this example: https://github.com/platformio/platformio/tree/develop/examples/wiring-blink and this command:
```shell
platformio init --project-dir examples/wiring-blink --ide clion -b uno
```

As for "auto-complete" - see explanation in `Warning` block: http://docs.platformio.org/en/latest/ide/clion.html

voneiden, 2015-10-10:
I got auto complete working by setting `include_directories` as absolute paths and adding a line like

`add_executable(Test src/test.cpp)`

ivankravets, 2015-10-13:
Dear friends,

Please re-test full support of CLion IDE in the latest development version http://docs.platformio.org/en/latest/installation.html#development-version

Waiting for your feedbacks.

ivankravets, 2015-10-13:
@voneiden thanks a lot for the hint! :+1:

closed 2015-10-13 by ivankravets

voneiden, 2015-10-13:
@ivankravets Nice, no problem.

averri, 2016-11-09:
I have followed all the documentations, and this is what I get:

[image]

earliestearl, 2016-11-11:
I am having the problem that Clion doesn't resolve my own library in the `lib` folder, platformio created. `platformio run` works quite well without any error but I cannot use the code completion from the IDE because the editor displays me the error: _Cannot find Foo.h_ which I include in my `mainProgram.c` which is in the `src` folder.

Can someone help me, please? Thanks in advance.

ivankravets, 2016-11-11:
@earliestearl see docs http://docs.platformio.org/en/stable/ide/clion.html

> The libraries which are added, installed or used in the project after generating process wont be reflected in IDE. To fix it please run PLATFORMIO_REBUILD_PROJECT_INDEX target.

earliestearl, 2016-11-11:
@ivankravets Sorry, I should have read the instructions more carefully. But thank you so much for your fast and good reply. Now, everything's working. Thanks for your great work, PlatformIO!

ivankravets, 2016-11-11:
@earliestearl Would be thankful for a star ⭐️ https://github.com/platformio/platformio/stargazers and don't forget to follow us on https://twitter.com/PlatformIO_Org

commit d5c852d3 (2015-07-15): Project generator for CLion IDE // Resolve #132
commit de2d6818 (2015-10-13): Full support of CLion IDE including code auto-completion // Resolve #132