platformio/platformio-core #134: Add library dependencies to platformio.ini

gandy92, 2015-03-22:
The file platformio.inialready contains information on the framework required to build a project. If it is missing, platformio offers to automatically install it. However, if the same project requires a library already known to the platformio library manager, the following manual steps are necessary:
1. Build using "platformio run" and examine output for missing library
2. Perform library ID search using "platformio lib search"
3. Use "platformio lib install ID" to install library
4. Repeat from step 1

I propose adding a new statement to platformio.ini that allow to specify required libraries in the form

require-libs = ID1 [, ID2 [, ..]]

instructing platformio to check if one or more of the required libs are missing and if so, suggest to automatically install them before proceeding the build.

pelikhan, 2015-03-22:
Yes, declaring library depedencies is a great feature in npm.

ivankravets, 2015-03-23:
I don't have any objections :) How about to name this option like ``deplibs``?

gandy92, 2015-03-23:
Sounds good to me :smiley:

commit 424e87a8 (2015-05-01): Add library dependencies using "install_libs" option in platformio.ini // Resolve #134
closed 2015-05-15 by ivankravets