platformio/platformio-core #136: Arduino with #include <SD.h> gives error because Energia.h is needed

Ekran, 2015-03-24:
I use the simple blink example on an nano and add the line #include <SD.h>. The compiler gives the error:

.pioenvs\autogen_nanoatmega328\SD_ID79\utility\Sd2Card.cpp:21:21: fatal error: Energia.h: No such file or directory

Shouldn't it use something like arduino.h?

ivankravets, 2015-03-24:
This library doesn't support framework ``arduino``. In other words, this is a port of this library https://github.com/adafruit/SD

I will add support for valid Arduino SD library as soon as possible.

--------

Current solution: just create in ``src`` dir file named ``Energia.h`` with the next contents:
```c
#include <Arduino.h>
```

ivankravets, 2015-03-25:
I've just added new library which is compatible with Arduino framework. Please use it http://platformio.org/#!/lib/show/161/SD

closed 2015-03-25 by ivankravets