platformio/platformio-core #137: Regression from 1.1 to 1.2 in regards to projects with .ino files

aphelps, 2015-03-25:
I'm not sure if this is addressed in another issue, but I just updated from 1.1 to 1.2 (and then tried again with the dev branch), and it looks like something broke with the handling of .ino files.  Previously I had projects working fine in both the Arduino IDE and platformio.  The errors mostly look like this:

```
.pioenvs/nano/src/HMTL_Module.cpp:5:21: error: 'msg_hdr_t' was not declared in this scope
boolean process_msg(msg_hdr_t *msg_hdr, boolean from_rs485, boolean forwarded);
^
.pioenvs/nano/src/HMTL_Module.cpp:5:32: error: 'msg_hdr' was not declared in this scope
boolean process_msg(msg_hdr_t *msg_hdr, boolean from_rs485, boolean forwarded);
^
.pioenvs/nano/src/HMTL_Module.cpp:5:49: error: expected primary-expression before 'from_rs485'
boolean process_msg(msg_hdr_t *msg_hdr, boolean from_rs485, boolean forwarded);
```

It looks like the .cpp that is generated from my .ino is placing the generated prototypes before my #include's, and since they involve structures defined in those included headers the compiler is failing.

(For the time being I've just reverted to 1.1)

ivankravets, 2015-03-25:
Sorry, this issue is caused by https://github.com/platformio/platformio/issues/100 
I will fix it :+1:

ivankravets, 2015-03-25:
Fixed in https://github.com/platformio/platformio/commit/f800b351f36753e6f448bdfcf36b9b76e6d5bbcd

@aphelps and @gandy92,

Could I ask you to try latest ``1.3.0.dev1``? http://docs.platformio.org/en/latest/installation.html#development-version

If it works, I will release 1.3.0 as soon as possible.

Thanks in advance!

closed 2015-03-25 by ivankravets

gandy92, 2015-03-25:
Tested this with the example from #140 and it compiles without any problems. Translated source code looks great, though the name is now pioman.ccp instead of %PDENAME%.cpp (which is fine by me).

ivankravets, 2015-03-25:
I use ``pioman.cpp`` because PlatformIO can concatenate multiple *.ino/*.pde files into main CPP.

See request https://github.com/platformio/platformio/issues/130

P.S: Thanks for report! :+1: :blush:

aphelps, 2015-03-26:
Works fine again now.

ivankravets, 2015-03-27:
@aphelps  @gandy92 

Please upgrade to latest 1.3.0 https://github.com/platformio/platformio/releases/tag/v1.3.0