platformio/platformio-core #140: Wrong insertion of function prototypes converting .pde to .cpp

gandy92, 2015-03-25:
Automatic insertion of function prototypes fails in some cases, for example:

The file src/DCFBinaryStream.pde containing
```cpp
#include <DCF77.h>       //https://github.com/thijse/Arduino-Libraries/downloads
#include <Time.h>        //http://www.arduino.cc/playground/Code/Time

#define DCF_PIN 2                // Connection pin to DCF 77 device
#define DCF_INTERRUPT 0          // Interrupt number associated with pin

time_t time;

// [...]

void digitalClockDisplay(time_t _time){
  tmElements_t tm;   
  breakTime(_time, tm);
  // [...]
}
```

is converted to .pioenvs/autogen_pro8MHzatmega328/src/DCFBinaryStream.cpp
```cpp
#include <Arduino.h>
void setup();
void loop();
void digitalClockDisplay(time_t _time);
void printDigits(int digits);
#line 1 "DCFBinaryStream.pde"
#include <DCF77.h>       //https://github.com/thijse/Arduino-Libraries/downloads
#include <Time.h>        //http://www.arduino.cc/playground/Code/Time

#define DCF_PIN 2                // Connection pin to DCF 77 device
#define DCF_INTERRUPT 0          // Interrupt number associated with pin

time_t time;
[...]
```

which leads to the compiler error 
```
.pioenvs/autogen_pro8MHzatmega328/src/DCFBinaryStream.cpp:4:26: error: variable or field 'digitalClockDisplay' declared void
void digitalClockDisplay(time_t _time);
^
.pioenvs/autogen_pro8MHzatmega328/src/DCFBinaryStream.cpp:4:26: error: 'time_t' was not declared in this scope
DCFBinaryStream.pde: In function 'void loop()':
DCFBinaryStream.pde:39:30: error: 'digitalClockDisplay' was not declared in this scope
scons: *** [.pioenvs/autogen_pro8MHzatmega328/src/DCFBinaryStream.o] Error 1
```

The correct solution would be to define the prototypes at the latest possible point, most impportantly after the include directives (which however could be anywhere in the source).

Another solution would be to move the `#include` statements to the top, introducing the difficulty of correctly handling include statements within `#ifdef`/`#ifndef` blocks.

ivankravets, 2015-03-25:
Sorry, I will work on it today and fix it. I've marked your issue as duplicate of https://github.com/platformio/platformio/issues/137

gandy92, 2015-03-25:
Sorry for placing a duplicate, I completely missed #137

ivankravets, 2015-03-25:
Fixed in https://github.com/platformio/platformio/commit/f800b351f36753e6f448bdfcf36b9b76e6d5bbcd

closed 2015-03-25 by ivankravets