platformio/platformio-core #143: Fix "UnicodeDecodeError: 'ascii' codec can't decode byte"

ivankravets, 2015-03-27:
Problem occurs under Windows OS and for environment where configurable encoding is used (non UTF-8 paths). 

https://docs.python.org/2/howto/unicode.html#unicode-filenames
```python
import sys
sys.getdefaultencoding()
```

ivankravets, 2015-03-31:
We can fix this issue within PlatformIO source code, but problem still exists within SCons source code.

To avoid this issue please re-define PlatformIO [home_dir](http://docs.platformio.org/en/latest/projectconf.html#home-dir) in your project configuration file [platformio.ini](http://docs.platformio.org/en/latest/projectconf.html). PlatformIO uses user's home directory by default, which can contain "non-latin" characters. The easy way is to set ``home_dir`` with path, which doesn't contain "non-latin" chars:

**platformio.ini**
```ini
[platformio]
home_dir = C:\.platformio

[env:your_env_here]
....
```

ivankravets, 2015-03-31:
Add this issue to http://docs.platformio.org/en/latest/faq.html#windows-unicodedecodeerror-ascii-codec-can-t-decode-byte

me21, 2017-03-14:
@ivankravets This error is also triggered by storing project in path containing non-latin characters. The workaround is (obviously) to store projects elsewhere or creating symlink (directory junction) so that PlatformIO does not have to deal with non-latin paths.

ivankravets, 2019-06-05:
Please open PlatformIO IDE Terminal and type
```
pio upgrade --dev
pio update
```

Restart VSCode. Does it work now?

commit 394d2723 (2019-06-05): Fix numerous issues related to "UnicodeDecodeError" and international locales, or when project path contains non-ASCII chars // Resolve #143, Resolve #1342, Resolve #1959, Resolve #2100
closed 2019-06-05 by ivankravets