platformio/platformio-core #144: Serial does not work with panStampAVR board

Absurd-Mind, 2015-03-29:
Serial does not work when trying to compile for a panStampAVR board.

This is the test code (test.ino)
```C
void setup() {
    Serial.begin(9600);
}

void loop() {
}
```
and this is the compiler output:
```
$ platformio run
[Sun Mar 29 10:30:10 2015] Processing panStampAVR (platform: atmelavr, board: panStampAVR, framework: arduino)
-------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/panStampAVR/src/test.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=8000000L -DPANSTAMP_AVR -DARDUINO=10601 -DPLATFORMIO=010200 -I.pioenvs/panStampAVR/FrameworkArduino -I.pioenvs/panStampAVR/FrameworkArduinoVariant .pioenvs/panStampAVR/src/test.cpp
test.ino: In function 'void setup()':
test.ino:2:5: error: 'Serial' was not declared in this scope
scons: *** [.pioenvs/panStampAVR/src/test.o] Error 1
===================================== [ ERROR ] Took 0.48 seconds =====================================
```

valeros, 2015-03-29:
It is weird. "Serial" is not defined by default. You can try include "HardwareSerial.h" in your sketch before using the Serial object.

Absurd-Mind, 2015-03-29:
So this is behaviour by design. Maybe this should be added in the documentation/examples.
Thank you for the help and the awesome tool!

ivankravets, 2015-03-29:
@Absurd-Mind thanks a lot for your report :+1: 

 I've just created [PlatformIO FAQ](http://docs.platformio.org/en/latest/faq.html) page and with "Troubleshooting" section. Here is your first question: http://docs.platformio.org/en/latest/faq.html#serial-does-not-work-with-panstampavr-board

closed 2015-03-29 by ivankravets