platformio/platformio-core #145: avrdude: fuses question

dontsovcmc, 2015-03-29:
I use only platformio for upload *.ino.
I have unknown problems several times ago and reinstall, reinit platformio.

Now I see not normal fuses while uploading:

avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (H:00, E:00, L:00)

platformio.ini:
[env:scadadon_env]
platform = atmelavr
framework = arduino
board = uno

# enable auto-uploading
targets = upload
upload_port = /dev/tty.usbmodem1421

In boards.txt file of Arduino IDE we can see good fuses:
atmega328.bootloader.low_fuses=0xFF
atmega328.bootloader.high_fuses=0xDA
atmega328.bootloader.extended_fuses=0x05

Why fuses became bad?
How I can repair them using platformio?

Thanks!

ivankravets, 2015-03-29:
1. Do you have any problems with uploading? 
2. Does your ``*.ino`` work correctly?
3. ``atmega328.bootloader.*_fuses`` will be used when you will update ``bootloader``. PlatformIO doesn't allow to upgrade bootloader in your device. 

P.S: See the output from [wiring example](https://github.com/platformio/platformio/tree/develop/examples/wiring-blink) for my Arduino UNO:
```shell
(develop)➜  wiring-blink git:(develop) platformio run -e autogen_uno -t upload
[Sun Mar 29 15:24:20 2015] Processing autogen_uno (platform: atmelavr, board: uno, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/autogen_uno/firmware.hex"])
Auto-detected UPLOAD_PORT: /dev/cu.usbmodemfd1311
"/Users/ikravets/.platformio/packages/tool-avrdude/avrdude" -v -D -p atmega328p -C "/Users/ikravets/.platformio/packages/tool-avrdude/avrdude.conf" -c arduino -b 115200 -P /dev/cu.usbmodemfd1311 -U flash:w:.pioenvs/autogen_uno/firmware.hex:i

avrdude: Version 6.0.1, compiled on Dec 16 2013 at 17:26:24
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/Users/ikravets/.platformio/packages/tool-avrdude/avrdude.conf"

Using Port                    : /dev/cu.usbmodemfd1311
Using Programmer              : arduino
Overriding Baud Rate          : 115200
AVR Part                      : ATmega328P
Chip Erase delay              : 9000 us
PAGEL                         : PD7
BS2                           : PC2
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

Programmer Type : Arduino
Description     : Arduino
Hardware Version: 3
Firmware Version: 4.4
Vtarget         : 0.3 V
Varef           : 0.3 V
Oscillator      : 28.800 kHz
SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file ".pioenvs/autogen_uno/firmware.hex"
avrdude: writing flash (1030 bytes):

Writing | ################################################## | 100% 0.18s

avrdude: 1030 bytes of flash written
avrdude: verifying flash memory against .pioenvs/autogen_uno/firmware.hex:
avrdude: load data flash data from input file .pioenvs/autogen_uno/firmware.hex:
avrdude: input file .pioenvs/autogen_uno/firmware.hex contains 1030 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.15s

avrdude: verifying ...
avrdude: 1030 bytes of flash verified

avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (H:00, E:00, L:00)

avrdude done.  Thank you.
```

ivankravets, 2015-03-31:
Please reopen if you still have problems.

closed 2015-03-31 by ivankravets

dontsovcmc, 2015-04-01:
The problems were with screen and wire of another shield...

So, it's ok!
Thanks!