platformio/platformio-core #147: Add new platform "timsp432" and support for TI MSP432 LaunchPad

ivankravets, 2015-03-29:
See: http://energia.nu/releasenotes/

zlalanne, 2015-05-27:
@valeros Are you working on this currently? I can help out...

ivankravets, 2015-05-29:
@zlalanne Please check your email, @valeros has sent a message with detailed information.

Thanks in advance!

DavidKopalaCU, 2018-08-18:
Is anybody currently working on this? I would love to help out and see this get added.

GZaccaroni, 2018-09-19:
Is there a workaround to use it?

LoaderB0T, 2019-04-05:
I would really like to now if there is a workaround or an incoming fix. A few fellow students and me are currently working with the msp432 launchpad and some of us would really appreciate to use platform.io.

txf-, 2019-08-09:
Is there a summary of the issues in integrating this to platformio?

minecraft2048, 2019-10-05:
Any updates on this? I want to use platformio instead of Energia to program this board

Xinyu-Wu-0000, 2020-09-29:
Any updates on this?

Kryptoxic, 2020-11-10:
Any updates? Looking to code on vscode with my MSP432 as well.

yann-qu, 2020-11-26:
Waiting and hoping for updates....

Littleor, 2021-07-28:
Now is 2021, the MSP430 is supported, so when will the MSP432 be ok? Hoping for any updates.

txf-, 2021-08-05:
There doesn't seem that there is  much demand for it, overall. A lot of tooling seems to be very TI specific. Unfortunately it would appear that using Code Composer is the only way for now.

ivankravets, 2021-08-05:
A temporary solution is to use the Energia project ( https://energia.nu/ ) created by @robertinant. Our support for MSP430 is outdated. We don't plan to add support for TM MSP432 in nearest future. The integration process for this hardware needs some time and resources. 

If you are interested in better support for TI products in PlatformIO, please contact directly to TI support ( https://ticsc.service-now.com/ ). We need demand for it. The more people contact, the bigger chance is that TI will listen to its customers.

ivankravets, 2022-01-15:
Updates! @zceemja from our community contributed integration for MSP432 https://registry.platformio.org/platforms/zceemja/timsp432