platformio/platformio-core #148: Handle libraries which are compatible with build environment using "library.json" data

ivankravets, 2015-03-31:
Currently, PlatformIO Library Dependency Finder analyse only ``#include`` and decides which library should be used in appropriate order. However, we have situation where a few libs have the same names (or even ``#include``), but are compatible with different platforms/frameworks.

Need to analyse [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html) and use libs which correspond to "build environment" framework/platform.

gandy92, 2015-03-31:
Thank you for opening this issue. Having had time to consider the matter some more, I think the build environment should only be considered to resolve ambiguities (two libraries with the same name targeting different platforms), unless the user chooses 'strict' library handling. Otherwise we would introduce too many potential problems for the existing platforms. With much more libraries introduced for the mbed platform however, library names may become more difficult to be kept unique.

ivankravets, 2015-03-31:
>  library names may become more difficult to be kept unique.

I don't see problems here. PlatformIO Library Manager is based on ``library_id``, not ``library_name``. However, I see another problem here: if I implement this issue, you will not have ability to use "arduino" libraries for "teensy" platform. They will be ignored. 

Solution
-----------

How about [ignore_libs](http://docs.platformio.org/en/latest/projectconf.html#ignore-libs)? It allows to keep many libs with the same names, but have ability to skip some libs for the specific build environment. 

Example
-----------

Let's see into these libs:
* http://platformio.org/#!/lib/show/115/DHT22
* http://platformio.org/#!/lib/show/58/DHT22

They have the same name, but is compatible with the different platforms/frameworks. 

**Question**: How to build different envs using these libs without conflicts?
```ini
[env:arduino_board]
platform = atmelavr
framework = arduino
# skip "energia" lib from Library Dependency Finder
ignore_libs = DHT22_58

[env:energia_board]
platform = timsp430
framework = energia
# skip "arduino" lib from Library Dependency Finder
ignore_libs = DHT22_115
```

As we can see, we can have multiple libs, but using ``ignore_libs`` option from build environment we can tune our build process.

------

Should I close this issue with ``wontfix`` label?

gandy92, 2015-04-01:
I'm not sure about the 'wontfix' label, is there a 'fixlater'?

The problem I see with the ignore_libs solution is, that the platform.ini might require changes at the time another platform gets a library that happens to have the same name as one of the libraries required by the specific build environment.

So I propose a solution where the library manager uses the platform information in the library manifest if -and only if- the library name alone is ambiguous.
Additionally, it may be helpful to allow the user to explicitly state a library mapping for a specific build environment, e.g 
'''
map_lib DHT22=DHT22_58
'''

ivankravets, 2015-04-03:
> The problem I see with the ignore_libs solution is, that the platform.ini might require changes at the time another platform gets a library that happens to have the same name 

I want to emphasise, that ``ignore_libs`` can be defined within ``[env:***]`` group. This isn't global option for the all build environments. If you have problem with detecting libs in some environment, then you can exclude lib, which should not be used.

Could I ask you to describe me an example, where ``ignore_libs`` will be helpless? Thanks! :+1:

gandy92, 2015-04-05:
The problem I see with `ignore_libs` is not that it is helpless in the moment platformio is being written. Let me explain with the following example:
* I write a program that uses lib1 named 'A' and id 1. At the time of writing, there is another lib3 also named 'A' with id 3, that uses the same include file names and thus clashes with lib1 
* stating `ignore_libs = lib3` helps me avoid the problem, since at that moment only lib1 is left to match the conditions
* half a year later you clone the repository of my program and try to compile. Now, the library manager knows not only lib1 and lib3 as 'A' but also lib1002 under the same name. Incidentially that also clashes with lib1
* now you need to modify platformio.ini to add lib1002 to `ignore_libs` so that you can compile. Before you even know you have to do this you are confronted with lots of annoying compiler errors.
* a field `use_libs` with unique identifiers (like 'A_ID1' rather than 'A') defined within the `[env:***]` group could avoid this problem

IMHO this also might be an issue for library dependencies, even more so when adding more and more libraries from different sources 

What do you think?

ivankravets, 2015-04-30:
Andy, thanks! I got you now. I will implement ``use_libs`` in PlatformIO 2.0

ivankravets, 2015-05-05:
Resolved in https://github.com/platformio/platformio/commit/f001c087ec87e1a6cfc3a711e05ad0d8abd236a0

closed 2015-05-05 by ivankravets