platformio/platformio-core #151: Allow to initialise PlatformIO project for the specified IDE

ivankravets, 2015-04-02:
Aims
====
PlatformIO can be integrated to multiple IDE/editors using a few installation/customisation steps. They are described in [main documentation](http://docs.platformio.org/en/latest/ide.html). Will be useful to have additional option ``--ide`` to [platformio init](http://docs.platformio.org/en/latest/userguide/cmd_init.html) command.

How it works
==========
If user specifies supported ``--ide`` type, then PlatformIO generates fully compatible project for the specified IDE (configuration files, etc.). This project can be imported later via ``IDE -> Import Project`` and should contain preconfigured settings:

* **Include Search Path**: add to "Indexer Include Path" all paths from toolchain, framework, SDK
* **Build System**: configure build system for project to: build command - ``platformio --quiet run``, clean command - ``platformio --quiet run --target clean`` 
* **Preprocessor definitions/symbols**: Specific definitions based on board type, like ``__AVR_ATmega328__``

IDE
===

We should support these IDE/Editors from the beginning:

* ``eclipse`` Eclipse
* ``vs/visual_studio`` Visual Studio
* ``st/sublime_text`` Sublime Text
* ``uvision`` uVision :question: 

Examples
========

```shell
# Arduino compatible project for Eclipse
$ platformio init --board=uno --ide eclipse

# mbed compatible project for Microsoft Visual Studio
$ platformio init --board=lpc1768  --ide=vs
```

Questions
========

1. How about multiple ``--board`` options or environments?

If more then one board is specified, then "first-generated" environment should be used. Nevertheless, we will add additional option, like ``use_for_ide = true`` to   ``[env:xxx]`` group in ``platformio.ini``.

ivankravets, 2015-04-02:
@valeros please look into @0xc0170 https://github.com/0xc0170/project_generator 

Looks like what do we need. Nevertheless, ``project_generator `` supports only ``eclipse``, but we can extend this list via [templates](https://github.com/0xc0170/project_generator/tree/master/project_generator/templates).

matthewelse, 2015-04-02:
0xc0170/project_generator might save you a bit of work with exporting to IDEs - we currently don't support visualstudio or sublimetext, however it wouldn't take much work to get those working. The only problem is that project_generator is currently only used with ARM architecture processors, however it wouldn't take very much work to get this to work with AVR and others. I've actually been meaning to add sublimetext support to pgen since it's what I always use, so I'll open an issue over there now.
We'd definitely welcome any contribution you guys are able to make, especially towards supporting different architectures.

if you have any questions about project_generator ask me or @0xc0170

ivankravets, 2015-04-02:
@matthewelse Thanks, see my comment above :+1:  

> The only problem is that project_generator is currently only used with ARM architecture processors, however it wouldn't take very much work to get this to work with AVR and others

Do we have option to specify own ``Preprocessor definitions/symbols`` and ``Build System``? It is important for us, because PlatformIO has own code builder and uses own pre-built toolchains, frameworks and etc . In other words, it is replacement for internal IDE's  build system. 

These ``Include Search Path`` and ``Preprocessor definitions/symbols``  are need just for correct "code completion" and "Realtime Indexer".

See examples:
* http://docs.platformio.org/en/latest/ide/visualstudio.html
* http://docs.platformio.org/en/latest/ide/eclipse.html
* http://docs.platformio.org/en/latest/ide/sublimetext.html

matthewelse, 2015-04-02:
I guess that you mean doing `platformio run` instead of `make` as part of the build system?

ivankravets, 2015-04-02:
Yes. See [Get Started](http://platformio.org/#!/get-started).

0xc0170, 2015-04-02:
All project settings are predefined by a user in yaml files -> dictionaries, thus if a project dictionary is filled with proper data, it can be passed to pgen to export the project files. Is this enhancement to create a full project (all source files, include paths, macros in the IDE) ? Or just required project files with defined output binary created by your build system, be available for debugging? If you got questions, create a new issue on pgen github. We can continue the discussion there, if needed.

I would like to see more tools supported, as mentioned above, like sublime, visual studio or any other which users prefer to use.

ivankravets, 2015-04-03:
@0xc0170, I've just opened a new issue #154. We will show what do we need in the final result. 

Here are a few examples with Eclipse: https://github.com/platformio/platformio/tree/develop/examples/ide-eclipse

> Is this enhancement to create a full project (all source files, include paths, macros in the IDE) ? 

Just: include paths, macros + replacement of internal build system to PlatformIO Code Builder.

> If you got questions, create a new issue on pgen github.

I create new issue on ``pgen`` project if the issue #154  will be fixed. I need to show you a few examples with configuration files which we need and how they should be filled. 

Thanks!

commit d3a4a04a (2015-05-22): Allow to initialise PlatformIO project for the specified IDE // Resolve #151
closed 2015-05-22 by ivankravets