platformio/platformio-core #153: Upgrade packages with "GNU Tools for ARM Embedded Processors" to 4.9-2015-q1

ivankravets, 2015-04-03:
https://launchpad.net/gcc-arm-embedded/+download

- [ ] Use pre-built toolchains for Windows, Mac and Linux 64bit
- [ ] Built own toolchain for Linux 32bit
- [ ] Built own toolchain for Linux ARM Arch

Don't forget to fix symlinks for pre-built toolchains. Please use this script https://github.com/platformio/platformio/blob/develop/scripts/fixsymlink.py

closed 2015-06-10 by ivankravets

VioletGiraffe, 2018-03-09:
Why did you give up on updating the ARM toolchain?
I ran into problems compiling my template code for SAM3X (Arduino Due), it requires some c++14 features that were first supported in GCC 4.9 (the official current platform from Arduino comes with 4.8). 
I tried [this release](https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-update), just dropped it into the `toolchain-gccarmnoneeabi` folder, and it seems to work just fine.

There are much newer toolchain versions [available](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm), but that's not as important as getting full c++14 support as c++14 adds many template meta-programming improvements that are crucial for writing convenient code that's expanded at compile time and results in efficient run-time implementation. Most notably, extended `constexpr` and generic lambdas - the latter is exactly what I'm missing right now and can't compile my project without.

Even the stock AVR compiler is GCC 4.9.3 that supports `-std=c++14`. It lacks in the standard library department, but I have successfully implemented all the bits I needed which wouldn't be possible without a compliant compiler.

ivankravets, 2018-03-09:
We use the latest GCC 7 toolchain for all dev/platforms excluding Atmel SAM to be compatible with Arduino IDE.  See https://github.com/arduino/ArduinoCore-samd/blob/master/extras/package_index.json.PR.template#L30

You can file a feature request here https://github.com/arduino/ArduinoCore-samd/issues

P.S: You can manually switch between toolchains with PlatformIO. Just replace https://github.com/platformio/platform-atmelsam/blob/develop/platform.json#L38 with any version of this list https://bintray.com/platformio/dl-packages/toolchain-gccarmnoneeabi

The platform's manifest is located in ~/.platformio/platforms

VioletGiraffe, 2018-03-09:
Thanks.
Would it be possible to use a more modern toolchain without having to hack PlatformIO internal files? Can I declare my own platform based on `arduinosam` but with a newer toolchain?

ivankravets, 2018-03-09:
Sure, just fork https://github.com/platformio/platform-atmelsam, replace that line and use in `platformio.ini`
```
[env:...]
platform = https://github.com/VioletGiraffe/platform-atmelsam.git
```

Don't forget to update your fork with our changes later.

P.S: There is the right solution for this case. See https://github.com/platformio/platformio-core/issues/1367 . However, there is no ETA for it.

VioletGiraffe, 2018-03-09:
Awesome! Just what I was looking for, a one-click build without having to fiddle with the PlatformIO installation files. Thanks for the excellent support, and good job letting us pull all kinds of stuff from Github (libraries, platforms etc.). That's a very good design, if you ask me.

VioletGiraffe, 2018-03-09:
Hm. I tried following your suggestions, but this toolchain's stdlib includes are missing `<bits/c++config.h>`. Indeed, there is no such file in the toolchain archive.
However, the default (4.8.4) toolchain files also do not have this header, but other headers are also including it, and everything works. So I understand that I'm missing something, but can't find what.
Searching the report for `c++config` did not yield any results.