platformio/platformio-core #157: Add support for Intel Galileo and Intel Edison

ivankravets, 2015-04-06:
* http://www.intel.com/content/www/us/en/do-it-yourself/arduino.html
* http://arduino.cc/en/ArduinoCertified/IntelGalileo

w4ilun, 2016-02-25:
Any updates on this? Would be great to use platformIO to develop on the Intel Edison/Galileo with C/C++

D3f0, 2016-02-25:
@ivankravets How can we help?

ivankravets, 2016-02-25:
@D3f0 the interesting question... The best "helper" in this situation is Intel. We will try to contact with them.

P.S: Do you have any contacts with Intel Corp.?

D3f0, 2016-02-25:
Unfortunately I don't have any contacts. They have their own solution for IoT, which I personally don't like very much, so I don't know if they would be interested, though we can try contacting them anyway.

I've found in [their forums](https://communities.intel.com/thread/49104?start=0&tstart=0) that sketches can be compiled with:
```
[bin]/i586-poky-linux-uclibc-g++ -m32 -march=i586 --sysroot=[tools]/i586-poky-linux-uclibc -Os -Wl,--gc-sections -march=i586 -o [tmp]/sketch_feb23a.cpp.elf [tmp]\sketch_feb23a.cpp.o [tmp]/core.a -L[tmp] -lm -lpthread
```

This compilations should not be hard to Scon-ify. Previously we should be able to download the compilers, .h files and stuff, I'll take a look at the URL Arduino IDE is calling to download the support files.

w4ilun, 2016-02-25:
What would be needed to get the ball rolling? I can reach out to see what can be supported

ivankravets, 2016-02-25:
Dear @w4ilun,

We have 2 issues related to Intel hardware. In the other words, we don't have support for the Intel hardware :(

Please ask someone from @Intel to contact me@ikravets.com and I'm ready to discuss implementation process.

-------
Regards, Ivan Kravets
  - CTO, Founder, Ph.D, Researcher and Software Architect at PlatformIO.Org

D3f0, 2016-02-27:
I've taken a look at what the Arduino IDE downloads from:
  http://downloads.arduino.cc/packages/package_index.json

Apparently the Intel related parts come from this part of the file
```json
{
      "email": "support@intel.com",
      "maintainer": "Intel",
      "websiteURL": "http://maker.intel.com/",
      "name": "Intel",
      "platforms": [
        {
          "name": "Intel i586 Boards",
          "version": "1.6.2+1.0",
          "category": "Arduino Certified",
          "architecture": "i586",
          "url": "https://github.com/01org/corelibs-galileo/archive/1.6.2+1.0.tar.gz",
          "archiveFileName": "corelibs-galileo-1.6.2.tar.gz",
          "checksum": "SHA-256:e20d62b0dccf0d68dbb61d70b866d77134b770b226d6046a61c7e8d55e64e53a",
          "size": "272961",
          "boards": [
            {
              "name": "Galileo"
            }
          ],
          "toolsDependencies": [
            {
              "packager": "Intel",
              "name": "i586-poky-linux-uclibc",
              "version": "1.6.2+1.0"
            },
            {
              "packager": "Intel",
              "name": "sketchUploader",
              "version": "1.6.2+1.0"
            }
          ]
        },
        {
          "name": "Intel i686 Boards",
          "version": "1.6.2+1.0",
          "category": "Arduino Certified",
          "architecture": "i686",
          "url": "https://github.com/01org/corelibs-edison/archive/1.6.2+1.0.tar.gz",
          "archiveFileName": "corelibs-edison-1.6.2.tar.gz",
          "checksum": "SHA-256:538ab8553f832f56b04df80d44992ecc994b9c296f3fce6902832d97f99811a8",
          "size": "271420",
          "boards": [
            {
              "name": "Edison"
            }
          ],
          "toolsDependencies": [
            {
              "packager": "Intel",
              "name": "core2-32-poky-linux",
              "version": "1.6.2+1.0"
            },
            {
              "packager": "Intel",
              "name": "sketchUploader",
              "version": "1.6.2+1.0"
            }
          ]
        },
        {
          "name": "Intel i586 Boards",
          "version": "1.6.7+1.0",
          "category": "Arduino Certified",
          "architecture": "i586",
[... 230 lines cut ...]
              "size": 324645
            },
            {
              "url": "https://github.com/01org/intel-arduino-tools/archive/1.6.4+1.14-linux32.tar.gz",
              "checksum": "SHA-256:8bd55d901ef26f89a2e0e67551eb966de2b3bd90504d8204bd200510b85e1a9b",
              "host": "i686-linux-gnu",
              "archiveFileName": "1.6.4+1.14-linux32.tar.gz",
              "size": 224137
            },
            {
              "url": "https://github.com/01org/intel-arduino-tools/archive/1.6.4+1.14-linux64.tar.gz",
              "checksum": "SHA-256:2da8ac969f6e1731adfe0802d015891109b834fdedfa5ca199742d5439b4f038",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "1.6.4+1.14-linux64.tar.gz",
              "size": 216542
            }
          ]
        }
      ]
    },
```

Most contributors at [01org](https://github.com/orgs/01org/people) work at Intel. Actually @w4ilun is one of them :) Maybe they're interested in helping out here.
@ivankravets maybe we can start downloading the tools and then figure out how to build stuff with them using SCons. 
I'll take a look at [Creating a Platform](http://docs.platformio.org/en/latest/platforms/creating_platform.html) guide to start downloading the tools.

ivankravets, 2016-02-27:
Guys, you are right. You can implement this within own development platform using http://docs.platformio.org/en/latest/platforms/creating_platform.html. I propose to make separate repo for that and develop within it. I'm and @valeros will look on it and help you if need.

As for packages. For the beginning, just copy them to `~/.platformio/packages` with the valid structure (take a look at the existing packages) and add them manually to `~/.platformio/appstate.json`. For examle:
```json
{
   ....
    "installed_packages": {
        "toochain-arcelf32": {
            "version": 1, 
            "time": 1451151351
        }, 
       ...
   }
}
```

If you need any help don't hesitate to contact us.

ivankravets, 2016-02-27:
Also, I want to notice that process with adding new development platform will be significantly simplified in PlatformIO 3.0. See https://github.com/platformio/platformio/issues/479

If this issue is not critical, I propose to wait when I'll implement #479

D3f0, 2016-03-07:
@ivankravets  Should I fork platformIO and put some subfolder in platformio/boards/intel.json?

ivankravets, 2016-03-07:
@D3f0 please develop into the home directory as described in http://docs.platformio.org/en/latest/platforms/creating_platform.html

Then share here 3 files:

* intel.json
* platform-builder.py
* platform.py

We are going to start working on PlatformIO 3.0 where will not be this info like, boards, platform scripts, etc inside PlatformIO Core. See https://github.com/platformio/platformio/issues/479

ivanzhenlms, 2017-11-28:
Please add support for Intel Galileo and Edison, I tried to create a custom board with the tutorial but I didn't understand it and I have no idea. Thanks.

ngmacha, 2018-02-23:
+1 for adding support for Intel Galileo and Edison boards!

JelleRoets, 2018-03-17:
Recently I created a tutorial to build, compile and debug c/cpp projects for the Intel Edison platform, using VSCode instead of Intel's System Studio (since the latter is not working for me)

Actually I created a generic makefile that compiles the c/cpp project in a docker container, so this approach is completely platform and IDE independent, this means you can use any editor you like: Sublime, Vim, Atom, VScode, etc... 

You can find this makefile, including an example Hello World project here:  https://github.com/JelleRoets/cross-compile-edison 

Hopefully this can help. Someone familiar with creating Platformio board definitions can maybe use this tutorial to integrate Intel Edison and Galileo into Platformio.

Cabalist, 2018-05-04:
These were discontinued about a year ago.   see here: https://hackaday.com/2017/06/19/intel-discontinues-joule-galileo-and-edison-product-lines/

I think this issue should be closed.  :/

ivankravets, 2018-05-04:
It makes sense

closed 2018-05-04 by ivankravets