platformio/platformio-core #158: Refactor command line interface (especially "platform" related commands)

ivankravets, 2015-04-06:
PlatformIO 1.0 CLI
==============
```shell
$ platformio --help
Usage: platformio [OPTIONS] COMMAND [ARGS]...

Options:
  --version  Show the version and exit.
  --help     Show this message and exit.

Commands:
  boards       Pre-configured Embedded Boards
  init         Initialize new PlatformIO based project
  install      Install new platforms
  lib          Library Manager
  list         List installed platforms
  run          Process project environments
  search       Search for development platforms
  serialports  List or Monitor Serial ports
  settings     Manage PlatformIO settings
  show         Show details about installed platform
  uninstall    Uninstall platforms
  update       Update installed platforms
  upgrade      Upgrade PlatformIO to the latest version
```

PlatformIO 2.0 CLI
==============
1. Add global ``-q/--quiet`` option which will suppress user prompts // issue #152
2. Move platform related commands to ``platformio platforms`` commands group
3. Update: development platforms, packages and libs by general ``platformio update`` command.

```shell
$ platformio --help
Usage: platformio [OPTIONS] COMMAND [ARGS]...

Options:
  --version  Show the version and exit.
  --help     Show this message and exit.
  --quiet    Suppress user prompts

Commands:
  boards       Pre-configured Embedded Boards
  init         Initialize new PlatformIO based project
  lib          Library Manager
  platforms    Manage development platforms
  run          Process project environments
  serialports  List or Monitor Serial ports
  settings     Manage PlatformIO settings
  update       Update installed platforms and libs
  upgrade      Upgrade PlatformIO to the latest version

$ platformio platforms --help
Usage: platformio platforms [OPTIONS] COMMAND [ARGS]...

Options:
  --help     Show this message and exit.

Commands:
  install      Install new platforms
  list         List installed platforms
  search       Search for development platforms
  show         Show details about installed platform
  uninstall    Uninstall platforms
  update       Update installed platforms
```
--------------

Friends  @valeros, @gandy92, @pelikhan, @aphelps, @ncolomer,

* What do you think? 
* Do you like these "s" endings in the most commands? Should I replace them with ``board``, ``platform``, ``serialport``?

aphelps, 2015-04-06:
For the 's' ending, I really don't have a preference.

For the addition of the -q/--quiet option, is the intent here to just bypass places where user confirmation is required?  Unix "conventions" tend to have -q mean "use less verbose logging", whereas -f/--force or -Y/--yes is used to bypass prompts for user confirmation.

ivankravets, 2015-04-06:
@aphelps thanks! I've just changed #152 to ``-f, --force`` :+1:

commit 07022bb3 (2015-04-17): Implement PlatformIO CLI 2.0 // Resolve #158
closed 2015-05-15 by ivankravets