platformio/platformio-core #159: PlatformIO 2.0

ivankravets, 2015-04-06:
Overview
=======

Friends, thanks a lot for using PlatformIO. Our audience and community grow up extremely fast! Thank you! :+1: :blush: 

I haven't expected that PlatformIO will be transformed from "source code builder" to powerful eco system with a lot of independent instruments which allow to simplify life of each embedded developer. (See my explanation in **Atmel**'s article: [Cross-board and cross-vendor embedded development with PlatformIO](http://blog.atmel.com/2015/03/23/cross-board-and-cross-vendor-embedded-development-with-platformio/))

**What do we have today?** 

* [Cross-platform source code builder for the most popular development platforms](http://platformio.org/#!/platforms). *10+ development platforms with pre-built tool chains for famous host OS*
* [Support for the popular embedded frameworks](http://platformio.org/#!/frameworks)
* Pre-configured settings for [the most popular embedded boards](http://platformio.org/#!/boards). *100+ added embedded boards*
* [Library Manager](http://platformio.org/#!/lib). *150+ registered libraries*
* [Integration with IDE](http://docs.platformio.org/en/latest/ide.html)
* etc.

Aims
====

The key feature of PlatformIO 2.0 should be simplified new **Command Line Interface**. Currently, we have mixed set of commands which confuses PlatformIO beginner. I mean, "list/show/search/install" commands which are located on the same level like, "boards,lib,serialports,etc.".

What is more, this is annoying issue with [PlatformIO blocks command execution using user prompt](http://docs.platformio.org/en/latest/faq.html#id6). This issue #152 should fix it (global ``-f, --force`` option before each PlatformIO "command").

Lastly, we should abstract away from "PlatformIO is just command line tool". We decided to spent more time for integration into different services, IDE, etc.

Roadmap
=======

- [ ] Refactor command line interface (especially "platform" related commands) // #158
- [ ] Add global ``-f, --force`` option which will accept any confirmation prompts // #152 
- [ ] Allow to initialise PlatformIO project for the specified IDE // #151

ivankravets, 2015-04-06:
Friends @nobugs-hare, @valeros, @gandy92, @pelikhan, @aphelps, @0xc0170, @robertinant, @PaulStoffregen, @aethaniel, @xxxajk, @Lauszus, @franciscoruiz, @ncolomer, @elyulka

Which key features should be implemented in PlatformIO 2.0?

aphelps, 2015-04-07:
The first thing that comes to mind is adding an upload port picker tool, so that rather than needing to specifying the port on the command line it would just present a list from a port scan and allow the user to quickly select one for the upload.

gandy92, 2015-04-08:
Thinking of platformio usage in companies, control of the update process could be an issue. Right now, updates of the toolchains, frameworks, etc are done automatically which is fine to keep up-to-date, especially with the current pace of platformio development. 
In some environments, however, it might be necessary to read update notes, changelogs, maybe evaluate the new version in a sandbox before it is cleared for active use. In this case we would require
* a way to deactivate automatic updates
* a way of informing about changes between the currently installed version and the available update (including libraries)
* a way of updating selected libraries only

Maybe more?

ivankravets, 2015-04-08:
@aphelps 

>  The first thing that comes to mind is adding an upload port picker tool, so that rather than needing to specifying the port on the command line

Hm... but this feature is implemented. Have you tried to upload without ``--upload-port`` option?  See this part of code: https://github.com/platformio/platformio/blob/develop/platformio/builder/tools/pioupload.py#L55:L79

ivankravets, 2015-04-08:
@gandy92 

> a way to deactivate automatic updates

See ``auto_update_platforms`` in http://docs.platformio.org/en/latest/userguide/cmd_settings.html

> a way of informing about changes between the currently installed version and the available update (including libraries)

Yes, sounds good, but I'm not sure that I will implement it in SmartAnthill 2.0. Could I ask you to open separate issues: 1-st for the "changes" of platform packages and 2-nd for the changes between libs (like CVS commits history)

> a way of updating selected libraries only

Thanks, I will implement this feature in SmartAnthill 2.0 - please open an issue.

aphelps, 2015-04-09:
@ivankravets Apparently you are already ahead of me, I hadn't seen that addition since I'd last tried it.

ajford, 2015-04-09:
The ability to specify the `env` name given when running `platformio init -b foo` would be nice.

I have some overlapping boards (that confuse the auto-uploader, i.e. `diecimilaatmega328` and `sparkfun_redboard`). Thus using `-e` is required to prevent the attempt to upload using the wrong env. But `autogen_BOARDNAME` is long and sometimes confusing to remember (`diecimila` is not easy to spell right the first time). I manually edited my `platformio.ini` to change my env names to easier names (i.e. `freeduino`, `redboard`, and `pro16`), but it would be useful if there was a way to do this when `init`ing a project (or adding new boards, ala #167).

ivankravets, 2015-04-23:
@ajford Thanks, I've just opened an issue #182

ivankravets, 2015-05-23:
PlatformIO 2.0 is out! https://github.com/platformio/platformio/releases/tag/v2.0.0

closed 2015-05-23 by ivankravets