platformio/platformio-core #160: Support popular RTOS projects

beriberikix, 2015-04-07:
Is there anything preventing PlatformIO from supporting popular RTOS projects like [ChibiOS/RT](http://chibios.org/), [FreeRTOS](http://www.freertos.org/) & [NuttX](http://nuttx.org/)?* A quick search on the web UI returned 0 results.

* there are many, many more RTOS projects out there.

ivankravets, 2015-04-08:
> Is there anything preventing PlatformIO from supporting popular RTOS projects

Yes, shortage of free time and human resource :weary:

PlatformIO is open source project, I would be thankful if someone contribute with pull requests. However, we will add support for the popular RTOS projects as soon as possible.

beriberikix, 2015-05-05:
What would be involved in adding one? I'm [playing around](https://github.com/ChibiOS/ChibiOS/blob/master/demos/STM32/RT-STM32F407-DISCOVERY/main.c) with @ChibiOS and it has 2 main parts - the RTOS and the HAL. Is the HAL considered a Framework while the RTOS is a library? Or is the RTOS a Platform?

ivankravets, 2015-05-05:
We can implement @ChibiOS within PlatformIO like a `framework`. @ChibiOS supports different development platforms which is supported by PlatformIO too.

1. We should fix all issues for the new [1.5.0](https://github.com/platformio/platformio/milestones/1.5.0) release. 
2. Need to finish work on [PlatformIO 2.0](https://github.com/platformio/platformio/issues/159).

@valeros How much efforts you need to add support for @ChibiOS? Will it be 1.6.0 or 2.1.0 release?

valeros, 2015-05-05:
I'm going to add ChibiOS when I have finished PlatformIO 2.0 features.

beriberikix, 2015-06-02:
Is the work for 2.0 complete? Would love to migrate my current chibi project to use PlatfromIO :)

ivankravets, 2015-06-02:
@beriberikix yes, 2.0 is released, but we are working on polishing (improvements/bug fixes) :blush: 

See our milestones https://github.com/platformio/platformio/milestones.

beriberikix, 2015-10-11:
Any chance of including this in an upcoming milestone?

ivankravets, 2015-10-13:
Dear @beriberikix,

We are working on our separate project @SmartAnthill and @valeros don't have time for @PlatformIO at the moment.

I propose you to make similar issue on the RTOS project forums etc:

* http://forum.chibios.org/phpbb/index.php
* http://www.freertos.org/FreeRTOS_Support_Forum_Archive/freertos_support_forum_archive_index.html

We can provide counselling help totally for free. The implementation of RTOS can be achieved via framework's script. See examples: https://github.com/platformio/platformio/tree/develop/platformio/builder/scripts/frameworks

Sorry for that :cry:

beriberikix, 2015-10-14:
Fair enough. Thanks for all the support thus far.

TA1DB, 2020-11-01:
After a long hiatus of decades, I have started working on embedded systems again for the last 5 years. Two of the greatest projects I've seen during this time are Platformio and Nuttx. Unfortunately when I start a nuttx work, I move away from platformio and partially forget some of its features. Besides, when I import a Pio project into Eclipse, for example, I can easily see and compile all options of Makefile, I can easily debug; this would be a great feature and value for Nuttx !. It would be really great when Nuttx will be supported on pio! Thanks in advance Ivan 🤓

me21, 2021-03-09:
Regarding ChibiOS: there is a library for PlatformIO which enables using ChibiOS/RT alongside with Arduino: https://platformio.org/lib/show/2986/ChRt