platformio/platformio-core #162: teensy31 stops execution when using float

gandy92, 2015-04-08:
porting my avr arduino firmware to teensy 3.1 using arm4, I encountered a strange problem. 

The code compiles and uploads perfectly, but execution stops at the first point a float variable is used. Assigning a value to a float variable seems not to be the problem (maybe the code is removed during optimization?) but a multiplication already makes the arm stop dead.

has anyone seen this before and know how to fix this? I can do integer operations for now but eventually I need to use libraries that make internal use of floating point arithmetics.

thanks, Andy.

ivankravets, 2015-04-08:
@gandy92 I've just remove hard FPU. Please [upgrade PlatformIO](http://docs.platformio.org/en/latest/installation.html#development-version) to the latest ``develop`` and update all platforms via ``> platformio update``.

Did it help you?

closed 2015-04-08 by ivankravets

gandy92, 2015-04-08:
That was fast :smile: 

Thank you for fixing this, everything works like a charme, now.

ivankravets, 2015-04-08:
The issues from @gandy92 should be fixed as soon as possible :blush:

commit 44c78525 (2015-04-08): Remove hard FPU for Cortex M4 // Resolve #162