platformio/platformio-core #164: Add support for Quirkbot

paulobarcelos, 2015-04-09:
Hey! First of all, congrats for such an amazing project, it's a real life saver!

I was wondering if there is a policy for adding new boards...

I wanted to include the specs for the Quirkbot (https://www.kickstarter.com/projects/1687812426/quirkbot-make-your-own-robots-with-drinking-straws), a new board we are releasing in August. (I am the lead developer on the project)

Is there a preferred way to do it? Should I just make a pull request with the addition to the ````/boards/misc.json````?

Thanks!

ivankravets, 2015-04-09:
Thanks you for the feedback! :blush: 

I see that Quirkbot is based on ``atmega32u4``. You can use the settings from ``leonardo`` board correct a few lines:
```json
{
    "quirkbot": {
        "build": {
            "core": "arduino",
            "extra_flags": "-DARDUINO_ARCH_AVR -DARDUINO_AVR_LEONARDO",
            "f_cpu": "16000000L",
            "mcu": "atmega32u4",
            "pid": "0x8036",
            "usb_product": "Quirkbot ",
            "variant": "leonardo",
            "vid": "0x2341"
        },
        "frameworks": ["arduino"],
        "name": "Quirkbot",
        "platform": "atmelavr",
        "upload": {
            "disable_flushing": true,
            "maximum_ram_size": 2560,
            "maximum_size": 28672,
            "protocol": "avr109",
            "require_upload_port" : true,
            "speed": 57600,
            "use_1200bps_touch": true,
            "wait_for_upload_port": true
        },
        "url": "http://quirkbot.com",
        "vendor": "Quirkbot"
    }
```

The one field which should be specified is ``variant``. Do you have specific ``pin_map`` for this board?

Finally, you can test this board with the latest version of PlatformIO:
1. Install PlatformIO
2.Create ``boards`` folder in ``~/.platformio`` folder and place here your own ``*.json`` files. Like, ``testquirkbot.json`` with the contents described above.
3. Check that this board is in ``platformio boards quirk``
4. Initialise new project ``platformio init --bord quirkbot``
5. Place your code to ``src`` and dependent libraries to ``lib``.

If this works for you, then you can contribute this board to PlatformIO core ``/boards/misc.json``.

P.S: I see you are developing own Web-IDE and Cloud Compiler. Don't forget that you can use PlatformIO as cross-platform code builder for these tasks.

paulobarcelos, 2015-04-09:
Thanks for the thorough feedback!

Our setup is very similar to the ArduinoLillyPad USB, so I will use that as a base.

But yeah, we do have our own variant (and also bootloader - but I don't think it matters on this scope, or?). Where should we place the ````pins_arduino.h````?

And regarding the cloud compiler, we are evaluating a few different options, but Plataformio is definitely on the top of the list - so thank you for that!

ivankravets, 2015-04-09:
> But yeah, we do have our own variant (and also bootloader - but I don't think it matters on this scope, or?). Where should we place the pins_arduino.h?

Please put it temporary to ``~/.platformio/packages/framework-arduinoavr/variants/quirkbot``. If all these changes will work on your PC, please share here ``pins_arduino.h`` and ``misc.json`` (@valeros will add it to our packages repository).

P.S: I'm going to release 1.4.0 tomorrow. If you give us this information, I'll add Quirkbot to this release (because I don't know when will be next release). We are going to work on PlatformIO 2.0 #159, which doesn't intend adding of new boards/platforms, just user CLI and integration with IDE and CI services.

paulobarcelos, 2015-04-09:
Regarding vid and pid. We don't have them. At least not yet...
Is that a requirement for you to be able to add the board to this release? Could we go in with 0x0000 ids at this point?

ivankravets, 2015-04-10:
> Regarding vid and pid.

Just skip these fields. Don't use "usb_product, pid, vid".

paulobarcelos, 2015-04-10:
Ok, overall it works great, but I got some problems with ````vid````, ````pid````, and ````usb_product````.

If I just omit the 3, I get these errors:

`````
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.cpp:83:29: error: 'USB_VID' was not declared in this scope
D_DEVICE(0x00,0x00,0x00,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,0,1);
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.h:282:55: note: in definition of macro 'D_DEVICE'
{ 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.cpp:83:37: error: 'USB_PID' was not declared in this scope
D_DEVICE(0x00,0x00,0x00,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,0,1);
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.h:282:60: note: in definition of macro 'D_DEVICE'
{ 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.cpp:86:37: error: 'USB_VID' was not declared in this scope
D_DEVICE(DEVICE_CLASS,0x00,0x00,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,0,1);
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.h:282:55: note: in definition of macro 'D_DEVICE'
{ 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.cpp:86:45: error: 'USB_PID' was not declared in this scope
D_DEVICE(DEVICE_CLASS,0x00,0x00,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,0,1);
^
`````
If I just set them to blank (````"vid": "", "pid":"":, "usb_product":""````) then I get:

`````
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.cpp:57:13: error: operator '==' has no left operand
#if USB_VID == 0x2341
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.cpp:62:15: error: operator '==' has no left operand
#elif USB_VID == 0x1b4f
^
In file included from .pioenvs/autogen_quirkbot/FrameworkArduino/USBAPI.h:38:0,
from .pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.cpp:19:
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.h:282:59: error: expected primary-expression before ',' token
{ 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.cpp:83:2: note: in expansion of macro 'D_DEVICE'
D_DEVICE(0x00,0x00,0x00,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,0,1);
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.h:282:64: error: expected primary-expression before ',' token
{ 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.cpp:83:2: note: in expansion of macro 'D_DEVICE'
D_DEVICE(0x00,0x00,0x00,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,0,1);
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.h:282:59: error: expected primary-expression before ',' token
{ 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.cpp:86:2: note: in expansion of macro 'D_DEVICE'
D_DEVICE(DEVICE_CLASS,0x00,0x00,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,0,1);
^
.pioenvs/autogen_quirkbot/FrameworkArduino/USBCore.h:282:64: error: expected primary-expression before ',' token
{ 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
^

`````

So for I went with a squatted registers:
`````
"vid": "0xf055",
"pid": "0xf055",
"usb_product": "Quirkbot" 
`````

And everything works fine.

If you are OK with it, here are the file contents https://gist.github.com/paulobarcelos/4bf8a684008152578fb3

ivankravets, 2015-04-10:
@valeros Could I ask you to test it?

ivankravets, 2015-04-13:
@paulobarcelos Your  board has been added in the latest 1.4.0 release. Does it work properly for you? 

P.S: Please re-open if have any problems.

closed 2015-04-13 by ivankravets