platformio/platformio-core #170: teensy build broken in 1.3.0

hamaker, 2015-04-09:
Hi there, thanks for building platform IO, looks really cool and just the kind of thing I was looking for! Building arduino and teensy stuff without having to use a clumsy IDE! YAY!

When building my first teensy project though, I got an error message that I couldn't understand immediately (I'm not much of a C programmer, sadly):
```
arm-none-eabi-g++ -o .pioenvs/autogen_teensy/firmware.elf -Os -Wl,--gc-sections -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -T /Users/niels/.platformio/packages/ldscripts/mk20dx128.ld .pioenvs/autogen_teensy/src/main.o -L.pioenvs/autogen_teensy -Wl,--start-group -lc -lgcc -lm .pioenvs/autogen_teensy/libFrameworkArduino.a -Wl,--end-group
.pioenvs/autogen_teensy/libFrameworkArduino.a(mk20dx128.o): In function `ResetHandler':
/Users/niels/code/personal/teensy/test/.pioenvs/autogen_teensy/FrameworkArduino/mk20dx128.c:763: undefined reference to `__rtc_localtime'
```
It turns out that this issue has actually already been fixed, in 960eccad8c132b00a0b59b303f85151a99dd2e6f. So this is not so much a report for the platformio devs, as for other users who have the same issue I had. It was solved by adding these lines to the teensy builder script (in my case found in `/usr/local/lib/python2.7/site-packages/platformio/builder/scripts/`, see commit for exact place to add them):
```
env.Append(
    LINKFLAGS=["-Wl,--defsym=__rtc_localtime=$UNIX_TIME"]
)
```
Or probably you could just update to the development version...
Cheers,
Niels

ivankravets, 2015-04-09:
@hamaker thanks for the feedback!

Yes, I know about it. I'm going to release 1.4.0 tomorrow. However, here is more easy solution with adding build flags to your env:
```ini
[env:your_teensy]
...
build_flags = -Wl,--defsym=__rtc_localtime=$UNIX_TIME
```

Does it work for you?

hamaker, 2015-04-09:
Hi @ivankravets, thanks for the quick response! Yes, that works, and indeed is a much cleaner solution, compared to my editing packaged files :anguished:

ivankravets, 2015-04-09:
Yep, PlatformIO is flexible and powerful instrument. However, I don't know hot to explain about it each developer :disappointed: Maybe, documentation isn't good. I'll work on it! :smile:

ivankravets, 2015-04-09:
Sorry, I reopen this issue and will close when 1.4.0 will be released!

closed 2015-04-09 by ivankravets