platformio/platformio-core #171: Auto update of platforms rendered project unbuildable until reinstalling from source

iterati, 2015-04-10:
I did not realize that the default settings for platformio were to auto-update libraries and platforms (this should be turned off, IMO, as an unbuildable project is worse than having to manually update a dependency) and after a number of successful builds, an update happened and I had to spend quite some time troubleshooting before I was able to fix the issue. The fix involved removing the platformio distribution I downloaded from the site, grabbing the source from github, and building/installing from the source.

I believe the problem was that updates to the teensy platform libraries were not compatible with the 1.3.0 build. Even after multiple attempts, platformio upgrade would return that it was at the most up to date version.

Here's the error my build would throw on 1.3.0:
```.pioenvs/autogen_teensy31/libFrameworkArduino.a(mk20dx128.o): In function `ResetHandler':
/Users/jmil/dev/unicrn/.pioenvs/autogen_teensy31/FrameworkArduino/mk20dx128.c:764: undefined reference to `__rtc_localtime'```

The project again builds successfully on 1.4.0.dev2. I recommend turning off the platform and library auto-updates by default. I can't imagine any development environment where that would be desirable. You're pinning platformio's requirements to a version in its requirements.txt for a reason.

ivankravets, 2015-04-17:
Thanks a lot! Autoupdates will be disabled in PlatformIO 2.0. See my latest commit b52ff11.

iterati, 2015-04-17:
Awesome. Thanks for your attention on this and your work on this project. It's really streamlined my development workflow.

commit b52ff11d (2015-04-17): Disabled automatic updates by default for platforms, packages and libraries // Resolve #171
closed 2015-05-15 by ivankravets