platformio/platformio-core #172: Support installing through ST package control

beriberikix, 2015-04-11:
https://packagecontrol.io/docs/submitting_a_package

Sublime Text has a nice ecosystem package manager. Publishing to Package Control would allow set up with one command across Operating Systems.

ivankravets, 2015-04-20:
What do you want to install using ST editor? PlatformIO or "PlatformIO Build System" for ST?

beriberikix, 2015-04-20:
Ideally both. The installed could detect if PlatformIO is already installed and if not start the process. It could also update or at least notify the user to update.

ivankravets, 2015-04-20:
> could detect if PlatformIO is already installed and if not start the process

Here will be an issue with the system permissions. ST starts without administrative permissions in 99%, when ``pip`` sometimes requires ``sudo`` permission. Yes, we could install PlatformIO within ST plugins, but ``platformio`` will not be available in this situation from cmd/shell.

The installation problem of PlatformIO is the main pain for me now. We lose ~50% users on the "Installation" step. To my mind, the cross-platform "PlatformIO Installer Application" should fix this problem. 

> Ideally both. 

I've added "enhancement" label for this issue. If we don't have issues with more higher priorities, we will work on it. 

Nevertheless, I hope that someone will help us and implement it :blush:

ivankravets, 2015-09-05:
@PlatformIO 2.3.0 has been released with simplified installation procedure (just single command `pip install platformio`) and enhanced Project Generator for multiple IDE. Please re-init your Sublime Text project. See docs http://docs.platformio.org/en/latest/ide/sublimetext.html#project-generator

ivankravets, 2016-04-05:
Implemented in separate plugin named [Deviot](https://github.com/gepd/Deviot) :+1:

closed 2016-04-05 by ivankravets

beriberikix, 2016-04-05:
:rocket: