platformio/platformio-core #173: Super-quick installation failed on OSX 10.10.3

beriberikix, 2015-04-13:
$ python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"

==> Installing Python Package Manager ...
Requirement already up-to-date: pip in /usr/local/lib/python2.7/site-packages
[SUCCESS]

==> Installing PlatformIO and dependencies ...
Requirement already up-to-date: setuptools in /usr/local/lib/python2.7/site-packages
Requirement already up-to-date: platformio in /usr/local/lib/python2.7/site-packages
Requirement already up-to-date: click>=3.0 in /usr/local/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: pyserial in /usr/local/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: requests>=2.4.0 in /usr/local/lib/python2.7/site-packages (from platformio)
    Command "/usr/local/opt/python/bin/python2.7 -c "import setuptools, tokenize;__file__='/var/folders/00/144nr000h01000cxqpysvccm004hkq/T/pip-Fdnyzc-build/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/00/144nr000h01000cxqpysvccm004hkq/T/pip-K4mvNa-record/install-record.txt --compile" failed with error code 1 in /var/folders/00/144nr000h01000cxqpysvccm004hkq/T/pip-Fdnyzc-build

[FAILURE]
The installation process has been FAILED!
Please report about this problem here
< https://github.com/platformio/platformio/issues >

ivankravets, 2015-04-13:
Did you try with ``sudo`` permission?

beriberikix, 2015-04-13:
Yup, but it doesn't seem to add it to my Path (tried restarting.)

me$ sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
Password:

==> Installing Python Package Manager ...
Requirement already up-to-date: pip in /usr/local/lib/python2.7/site-packages
[SUCCESS]

==> Installing PlatformIO and dependencies ...
Requirement already up-to-date: setuptools in /usr/local/lib/python2.7/site-packages
Requirement already up-to-date: platformio in /usr/local/lib/python2.7/site-packages
Requirement already up-to-date: click>=3.0 in /usr/local/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: pyserial in /usr/local/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: requests>=2.4.0 in /usr/local/lib/python2.7/site-packages (from platformio)
Collecting http://sourceforge.net/projects/scons/files/latest/download
  Downloading http://sourceforge.net/projects/scons/files/latest/download (621kB)
Installing collected packages: scons
  Running setup.py install for scons
Successfully installed scons
[SUCCESS]

 ==> Installation process has been successfully FINISHED! <==

Usage: __main__.py [OPTIONS] COMMAND [ARGS]...

Options:
  --version  Show the version and exit.
  --help     Show this message and exit.

Commands:
  boards       Pre-configured Embedded Boards
  init         Initialize new PlatformIO based project
  install      Install new platforms
  lib          Library Manager
  list         List installed platforms
  run          Process project environments
  search       Search for development platforms
  serialports  List or Monitor Serial ports
  settings     Manage PlatformIO settings
  show         Show details about installed platform
  uninstall    Uninstall platforms
  update       Update installed platforms
  upgrade      Upgrade PlatformIO to the latest version

 Please RESTART your Terminal Application and run `platformio --help` command.
me$ platformio
-bash: platformio: command not found

ivankravets, 2015-04-13:
Give me please an output of this command:
```shell
echo $PATH
```

beriberikix, 2015-04-13:
/Users/me/bin/google-cloud-sdk/bin:/usr/local/git/current/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/bin/g4bin

ivankravets, 2015-04-13:
Hm...

Does this command work for you?
```shell
pip --version
```

beriberikix, 2015-04-13:
yup! pip 6.1.1 from /usr/local/lib/python2.7/site-packages (python 2.7)

ivankravets, 2015-04-13:
Hm... Could you try to uninstall/install PlatformIO via "pip":

> pip uninstall platformio
> pip install platformio

Looks like you have a problem with "setuptools" package. Could I ask you to show the output of "pip list"command?

Also, please verify that latest version of setuptools is installed.

With best regards, Ivan Kravets
- "Creative comes from talent, and never from knowledge" (c)
_________________________
http://www.ikravets.com

beriberikix, 2015-04-13:
Hurray! I wasn't running the latest setuptools. After upgrading, uninstall than install worked :smile:

closed 2015-04-14 by beriberikix

ivankravets, 2015-04-16:
I've just added this information to FAQ: https://github.com/platformio/platformio/commit/bf0f4c8de43f92ad95155a98cc3448b7efdf14ec

Sorry for the issue.