platformio/platformio-core #175: Use alternative protocol to upload to STM32

beriberikix, 2015-04-14:
I recently got the @armstrap, which is a [`STM32F407VET6`](http://docs.armstrap.org/en/latest/hardware-overview.html) with a built-in JTAG powered by a @blacksphere Black Magic. To test it out quickly with platformio, I used the `disco_f407vg` since it is also a f407. However, the configuration looks for ST-Link and fails. I'm not too familiar with the Black Magic, so I'm not positive what it needs. 

That said, how do I specify another means to upload? I found the [`upload_protocol`](http://docs.platformio.org/en/latest/projectconf.html#upload-protocol) field, but the possible values aren't documented.

Thanks!

armstrap, 2015-04-15:
The way I upload my code (via the CLI)

 > arm-none-eabi-gdb testprogram.elf
target extended-remote /dev/ttyACM0
monitor swdp_scan
attach 1
load   <----- Flashes your STM32F4 chip with binary testprogram.elf
run

Note:  `/dev/ttyACM0` is Ubuntu specific

beriberikix, 2015-04-18:
Got it. @ivankravets any idea how to specify gdb?

ivankravets, 2015-04-20:
@valeros, please show @beriberikix how to create own platform using our ``stm32`` platform as "base frame" and extend functionality. More details are in http://docs.platformio.org/en/latest/platforms/creating_platform.html

valeros, 2015-04-21:
Hello @beriberikix! I'll try to help you.
First step is create new platform:
You need to create new folder `platforms` in platformio home dir `~/.platformio` or `%HOMEPATH%\.platformio` (on Windows) and copy there two files (`ststm32gdb.py, ststm32gdb-builder.py`) from this gist: https://gist.github.com/valeros/ef7b632c5bcc4541c670
Now, you can install new platform with command `platformio install ststm32gdb`.

Second step is add new board:
You need to create new folder `boards` in platformio home dir `~/.platformio` or `%HOMEPATH%\.platformio` (on Windows) and create there new file `armstrap.json` with board specification content:
```json
{
    "armstrap_eagle": {
        "build": {
            "core": "stm32",
            "extra_flags": "-DSTM32F40_41xxx",
            "f_cpu": "168000000L",
            "ldscript": "armstrap_stm32_flash.ld",
            "cpu": "cortex-m4",
            "mcu": "stm32f407vet6",
            "variant": "stm32f4"
        },
        "frameworks": ["cmsis", "spl", "libopencm3", "mbed"],
        "name": "Armstrap Eagle",
        "platform": "ststm32gdb",
        "upload": {
            "maximum_ram_size": 196608,
            "maximum_size": 524288
        },
        "url": "http://docs.armstrap.org/en/latest/hardware-overview.html",
        "vendor": "Armstrap"
    }
}
```

Third step is add a new ldscript for armstrap board to this package `~/.platformio/packages/ldscripts`. You can get it from this gist:
https://gist.github.com/valeros/88c89c3573e78df70579

After these steps, you can try to upload your firmware with next command:
`platformio run -t upload --upload-port  /dev/ttyACM0`

Unfortunately, I don't have board to check this aproach, but in theory it should work.
That's all, please tell us about your results.

beriberikix, 2015-04-22:
Thanks for the detailed steps! When I tried installing the platform, I get an error (see below.) It appears as if the directory is not being discovered on my Mac. Do I need to add anything in addition to my $PATH?

```
$ platformio --version
PlatformIO, version 1.4.0
$ ls ~/.platformio/platforms/
ststm32gdb-builder.py	ststm32gdb.py
$ platformio search
atmelavr (available packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa)
--------
Atmel AVR 8- and 32-bit MCUs deliver a unique combination of performance, power efficiency and design flexibility. Optimized to speed time to market-and easily adapt to new ones-they are based on the industrys most code-efficient architecture for C and assembly programming.

atmelsam (available packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa)
--------
Atmel | SMART offers Flash- based ARM products based on the ARM Cortex-M0+, Cortex-M3 and Cortex-M4 architectures, ranging from 8KB to 2MB of Flash including a rich peripheral and feature mix.

espressif (available packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa)
---------
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.

freescalekinetis (available packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa)
----------------
Freescale Kinetis Microcontrollers is family of multiple hardware- and software-compatible ARM Cortex-M0+, Cortex-M4 and Cortex-M7-based MCU series. Kinetis MCUs offer exceptional low-power performance, scalability and feature integration.

nordicnrf51 (available packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa)
-----------
The Nordic nRF51 Series is a family of highly flexible, multi-protocol, system-on-chip (SoC) devices for ultra-low power wireless applications. nRF51 Series devices support a range of protocol stacks including Bluetooth Smart (previously called Bluetooth low energy), ANT and proprietary 2.4GHz protocols such as Gazell.

nxplpc (available packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa)
------
The NXP LPC is a family of 32-bit microcontroller integrated circuits by NXP Semiconductors. The LPC chips are grouped into related series that are based around the same 32-bit ARM processor core, such as the Cortex-M4F, Cortex-M3, Cortex-M0+, or Cortex-M0. Internally, each microcontroller consists of the processor core, static RAM memory, flash memory, debugging interface, and various peripherals.

ststm32 (available packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa)
-------
The STM32 family of 32-bit Flash MCUs based on the ARM Cortex-M processor is designed to offer new degrees of freedom to MCU users. It offers a 32-bit product range that combines very high performance, real-time capabilities, digital signal processing, and low-power, low-voltage operation, while maintaining full integration and ease of development.

teensy (available packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa)
------
Teensy is a complete USB-based microcontroller development system, in a very small footprint, capable of implementing many types of projects. All programming is done via the USB port. No special programmer is needed, only a standard "Mini-B" USB cable and a PC or Macintosh with a USB port.

timsp430 (available packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa)
--------
MSP430 microcontrollers (MCUs) from Texas Instruments (TI) are 16-bit, RISC-based, mixed-signal processors designed for ultra-low power. These MCUs offer the lowest power consumption and the perfect mix of integrated peripherals for thousands of applications.

titiva (available packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa)
------
Texas Instruments TM4C12x MCUs offer the industrys most popular ARM Cortex-M4 core with scalable memory and package options, unparalleled connectivity peripherals, advanced application functions, industry-leading analog integration, and extensive software solutions.

$ platformio install ststm32gdb
Error: Unknown platform 'ststm32gdb'
```

valeros, 2015-04-22:
This bug fixed in platformio2_0 branch.
In stable 1.4.0, you need to modify one single line in the file `ststm32gdb.py`:
Please. replace line, `class Ststm32gdbPlatform(Ststm32Platform):` to
`class Ststm32GdbPlatform(Ststm32Platform):`

beriberikix, 2015-04-22:
It installs! I'm so close, but I have 1 issue left (might be a general PlatformIO question.)

Using the [Blinky sample](https://s3.amazonaws.com/armstrap-public/examples/armstrap_blinkyexamples_1.0.0.zip) from Armstrap, I get a compile error that it cannot find the periperhal import, whether I place it in the libs folder or directly in the src (see below.) Suggestions? I didn't see an include section mentioned in the [platform.ini docs](http://docs.platformio.org/en/latest/projectconf.html). Also, maybe this is a n00b question but should installing a platform also download the peripherals?

$ platformio run -t upload --upload-port /dev/cu.usbmodem7B4097C1
[Wed Apr 22 06:26:32 2015] Processing autogen_armstrap_eagle (targets: upload, platform: ststm32gdb, board: armstrap_eagle, framework: cmsis)
--------------------------------------------------------------------------------
arm-none-eabi-gcc -o .pioenvs/autogen_armstrap_eagle/src/main.o -c -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -MMD -DF_CPU=168000000L -DSTM32F4 -DSTM32F40_41xxx -DPLATFORMIO=010400 -I.pioenvs/autogen_armstrap_eagle/CMSIS -I.pioenvs/autogen_armstrap_eagle/STM32F4xx -I.pioenvs/autogen_armstrap_eagle/FrameworkCMSIS -I.pioenvs/autogen_armstrap_eagle/FrameworkCMSISVariant .pioenvs/autogen_armstrap_eagle/src/main.c
In file included from .pioenvs/autogen_armstrap_eagle/src/main.c:1:0:
.pioenvs/autogen_armstrap_eagle/src/main.h:2:28: fatal error: stm32f4xx_gpio.h: No such file or directory
#include "stm32f4xx_gpio.h"
^
compilation terminated.
scons: *** [.pioenvs/autogen_armstrap_eagle/src/main.o] Error 1
========================= [ ERROR ] Took 0.78 seconds =========================

valeros, 2015-04-22:
Try this `platformio.ini`:
```
platform = ststm32gdb
framework = cmsis,spl
board = armstrap_eagle
```

beriberikix, 2015-04-22:
Ah, makes sense! Got the success message. I will try uploading it tonight when I get home from work.

Thanks again.

beriberikix, 2015-04-23:
Is this an SCons error?

```
$ platformio run -t upload --upload-port /dev/cu.usbmodem7B4097C1
[Wed Apr 22 21:03:25 2015] Processing autogen_armstrap_eagle (platform: ststm32gdb, board: armstrap_eagle, framework: cmsis,spl)
--------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/autogen_armstrap_eagle/firmware.elf"])
scons: *** [upload] TypeError : execv() arg 2 must contain only strings
Traceback (most recent call last):
File "/usr/local/lib/scons-2.3.4/SCons/Action.py", line 1065, in execute
result = self.execfunction(target=target, source=rsources, env=env)
File "/Users/jberi/.platformio/platforms/ststm32gdb-builder.py", line 19, in BeforeUpload
stdout=PIPE, stdin=PIPE, stderr=STDOUT
File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 710, in __init__
errread, errwrite)
File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1335, in _execute_child
raise child_exception
TypeError: execv() arg 2 must contain only strings
========================= [ ERROR ] Took 0.95 seconds =========================
```

valeros, 2015-04-23:
Please, try updated version ststm32dgb-builder.py:
https://gist.github.com/valeros/28d84a7a8f78825e6956

beriberikix, 2015-04-23:
That did the trick @valeros! I know have BLINKENLICHTEN :low_brightness: :high_brightness: :low_brightness: 

As an aside, the executable hangs until I forcibly quit  (see ^C in output below) and the firmware doesn't re-run when I power cycle the unit. I was having the same problem when I was using my Makefile, so I can take that issue back to the @armstrap Forum.

```
$ platformio run -t upload --upload-port /dev/cu.usbmodem7B4097C1
[Thu Apr 23 06:14:37 2015] Processing autogen_armstrap_eagle (platform: ststm32gdb, board: armstrap_eagle, framework: cmsis,spl)
--------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/autogen_armstrap_eagle/firmware.elf"])

^CGNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140228-cvs
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf...done.
(gdb) Remote debugging using /dev/cu.usbmodem7B4097C1
(gdb) Target voltage: 4.9V
scons: *** [upload] Build interrupted.Available Targets:
No. Att Driver
1      STM32F4xx

(gdb) Attaching to program: /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf, Remote target
0x08000218 in SystemInit ()
at .pioenvs/autogen_armstrap_eagle/src/system_stm32f4xx.c:226
226	  RCC->PLLCFGR = 0x24003010;
(gdb) Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x264 lma 0x8000188
Start address 0x80002e8, load size 1004
Transfer rate: 2 KB/sec, 502 bytes/write.
(gdb) The program being debugged has been started already.
Start it from the beginning? (y or n) [answered Y; input not from terminal]
Starting program: /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf

Program received signal SIGINT, Interrupt.
0x080001c6 in delay (ms=550905, ms@entry=250)
at .pioenvs/autogen_armstrap_eagle/src/main.c:48
48	    while(ms--) {
(gdb) quit
A debugging session is active.

Inferior 1 [Remote target] will be killed.

Quit anyway? (y or n) [answered Y; input not from terminal]

Error: Aborted by user
```

valeros, 2015-04-23:
Nice!
I made a small correction in ststm32dgb-builder.py:
`https://gist.github.com/valeros/28d84a7a8f78825e6956`
Can you test it?

beriberikix, 2015-04-23:
Ah, that makes sense. @ work but will try again tonight!

beriberikix, 2015-04-24:
No luck.

```
$ cat ~/.platformio/platforms/ststm32gdb-builder.py
# Copyright (C) Ivan Kravets <me@ikravets.com>
# See LICENSE for details.

"""
    Builder for ST STM32 Series ARM microcontrollers with GDB upload.
"""

import subprocess
from subprocess import PIPE, STDOUT
from os.path import join

from SCons.Script import (COMMAND_LINE_TARGETS, AlwaysBuild, Default,
                          DefaultEnvironment, Exit, SConscript)

def BeforeUpload(target, source, env):  # pylint: disable=W0613,W0621
    p = subprocess.Popen(
        [env.subst("$UPLOADER"), env.subst(source)[0]],
        stdout=PIPE, stdin=PIPE, stderr=STDOUT
    )
    result = p.communicate(
        input=b'''target extended-remote %s\nmonitor swdp_scan\nattach 1\nload\nrun\nkill\nquit''' % env.subst("$UPLOAD_PORT"))[0]
    print(result.decode())
    p.kill()
    if "error" in result.decode().lower():
        Exit("Error: Can't upload firmware with GDB")

env = DefaultEnvironment()

SConscript(env.subst(join("$PIOBUILDER_DIR", "scripts", "basearm.py")))

env.Replace(
    UPLOADER=join(
        "$PIOPACKAGES_DIR", "toolchain-gccarmnoneeabi",
        "bin", "arm-none-eabi-gdb"
    )
)

env.Append(
    CPPDEFINES=[
        "${BOARD_OPTIONS['build']['variant'].upper()}"
    ],

    LINKFLAGS=[
        "-nostartfiles",
        "-nostdlib"
    ]
)

#
# Target: Build executable and linkable firmware
#

target_elf = env.BuildFirmware()

#
# Target: Build the .bin file
#

if "uploadlazy" in COMMAND_LINE_TARGETS:
    target_firm = join("$BUILD_DIR", "firmware.bin")
else:
    target_firm = env.ElfToBin(join("$BUILD_DIR", "firmware"), target_elf)

#
# Target: Print binary size
#

target_size = env.Alias("size", target_elf, "$SIZEPRINTCMD")
AlwaysBuild(target_size)

#
# Target: Upload by default .bin file
#

upload = env.Alias(
    ["upload", "uploadlazy"], target_elf, BeforeUpload)
AlwaysBuild(upload)

#
# Target: Define targets
#

Default([target_firm, target_size])
$ platformio run -t upload --upload-port /dev/cu.usbmodem7B4097C1
[Thu Apr 23 19:58:12 2015] Processing autogen_armstrap_eagle (platform: ststm32gdb, board: armstrap_eagle, framework: cmsis,spl)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/autogen_armstrap_eagle/firmware.elf"])
^CGNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140228-cvs
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
For bug reporting instructions, please see:scons: *** [upload] Build interrupted.

<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf...done.
(gdb) Remote debugging using /dev/cu.usbmodem7B4097C1
(gdb) Target voltage: 4.9V
Available Targets:
No. Att Driver
1      STM32F4xx
(gdb) Attaching to program: /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf, Remote target
delay (ms=7521484, ms@entry=2500) at .pioenvs/autogen_armstrap_eagle/src/main.c:49
49	        __NOP();
(gdb) Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x264 lma 0x8000188
Start address 0x80002e8, load size 1004
Transfer rate: 2 KB/sec, 502 bytes/write.
(gdb) The program being debugged has been started already.
Start it from the beginning? (y or n) [answered Y; input not from terminal]
Starting program: /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf

Program received signal SIGINT, Interrupt.
delay (ms=4995101, ms@entry=2500) at .pioenvs/autogen_armstrap_eagle/src/main.c:49
49	        __NOP();
(gdb) Kill the program being debugged? (y or n) [answered Y; input not from terminal]
(gdb)
Error: Aborted by user
```

valeros, 2015-04-24:
About firmware re-run:
Can you try change the order of commands kill and run (at first kill debug session and than run program)?
```input=b"target extended-remote %s\nmonitor swdp_scan\nattach 1\nload\nkill\nrun\nquit" % env.subst("$UPLOAD_PORT"))[0]```

beriberikix, 2015-04-24:
Just replace that in ststm32gdb-builder.py, right? Question - to update this file all I'm doing is deleting the *.pyc and re-runinng `platformio install ststm32gdb`. Is that right?

```
$ cat ~/.platformio/platforms/ststm32gdb-builder.py
# Copyright (C) Ivan Kravets <me@ikravets.com>
# See LICENSE for details.

"""
    Builder for ST STM32 Series ARM microcontrollers with GDB upload.
"""

import subprocess
from subprocess import PIPE, STDOUT
from os.path import join

from SCons.Script import (COMMAND_LINE_TARGETS, AlwaysBuild, Default,
                          DefaultEnvironment, Exit, SConscript)

def BeforeUpload(target, source, env):  # pylint: disable=W0613,W0621
    p = subprocess.Popen(
        [env.subst("$UPLOADER"), env.subst(source)[0]],
        stdout=PIPE, stdin=PIPE, stderr=STDOUT
    )
    result = p.communicate(
        input=b"target extended-remote %s\nmonitor swdp_scan\nattach 1\nload\nkill\nrun\nquit" % env.subst("$UPLOAD_PORT"))[0]
    print(result.decode())
    p.kill()
    if "error" in result.decode().lower():
        Exit("Error: Can't upload firmware with GDB")

env = DefaultEnvironment()

SConscript(env.subst(join("$PIOBUILDER_DIR", "scripts", "basearm.py")))

env.Replace(
    UPLOADER=join(
        "$PIOPACKAGES_DIR", "toolchain-gccarmnoneeabi",
        "bin", "arm-none-eabi-gdb"
    )
)

env.Append(
    CPPDEFINES=[
        "${BOARD_OPTIONS['build']['variant'].upper()}"
    ],

    LINKFLAGS=[
        "-nostartfiles",
        "-nostdlib"
    ]
)

#
# Target: Build executable and linkable firmware
#

target_elf = env.BuildFirmware()

#
# Target: Build the .bin file
#

if "uploadlazy" in COMMAND_LINE_TARGETS:
    target_firm = join("$BUILD_DIR", "firmware.bin")
else:
    target_firm = env.ElfToBin(join("$BUILD_DIR", "firmware"), target_elf)

#
# Target: Print binary size
#

target_size = env.Alias("size", target_elf, "$SIZEPRINTCMD")
AlwaysBuild(target_size)

#
# Target: Upload by default .bin file
#

upload = env.Alias(
    ["upload", "uploadlazy"], target_elf, BeforeUpload)
AlwaysBuild(upload)

#
# Target: Define targets
#

Default([target_firm, target_size])
```

valeros, 2015-04-24:
Yes, just replace that line in ststm32gdb-builder.py
There is no need to delete .pyc files and reinstall platform.

beriberikix, 2015-04-24:
Same results as before.

```
$ platformio run -t upload --upload-port /dev/cu.usbmodem7B4097C1
[Fri Apr 24 04:32:00 2015] Processing autogen_armstrap_eagle (platform: ststm32gdb, board: armstrap_eagle, framework: cmsis,spl)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/autogen_armstrap_eagle/firmware.elf"])
^CGNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140228-cvs
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
scons: *** [upload] Build interrupted.Reading symbols from /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf...done.

(gdb) Remote debugging using /dev/cu.usbmodem7B4097C1
(gdb) Target voltage: 4.9V
Available Targets:
No. Att Driver
1      STM32F4xx
(gdb) Attaching to program: /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf, Remote target
delay (ms=8127453, ms@entry=2500) at .pioenvs/autogen_armstrap_eagle/src/main.c:48
48	    while(ms--) {
(gdb) Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x264 lma 0x8000188
Start address 0x80002e8, load size 1004
Transfer rate: 2 KB/sec, 502 bytes/write.
(gdb) Kill the program being debugged? (y or n) [answered Y; input not from terminal]
(gdb) Starting program: /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf

Program received signal SIGINT, Interrupt.
delay (ms=4700663, ms@entry=2500) at .pioenvs/autogen_armstrap_eagle/src/main.c:48
48	    while(ms--) {
(gdb) A debugging session is active.

Inferior 1 [Remote target] will be killed.

Quit anyway? (y or n) [answered Y; input not from terminal]

Error: Aborted by user
```

valeros, 2015-04-24:
Let's try different approach.
Try this `ststm32gdb-builder.py`:
https://gist.github.com/valeros/28d84a7a8f78825e6956
Put the new file `upload.gdb` in your project folder (near platformio.ini) with this content:
```
target extended-remote /dev/cu.usbmodem7B4097C1
monitor swdp_scan
attach 1
load
detach
kill
quit
```

beriberikix, 2015-04-25:
Errors, though not very descriptive.

```
$ platformio run
[Fri Apr 24 20:27:46 2015] Processing autogen_armstrap_eagle (targets: upload, platform: ststm32gdb, board: armstrap_eagle, framework: cmsis,spl)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
/Users/me/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gdb /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf -batch -x /Users/me/Downloads/blinky-pio-armstrap/upload.gdb
Target voltage: 4.9V
Available Targets:
No. Att Driver
1      STM32F4xx
0x080001c6 in delay (ms=1145011, ms@entry=2500) at .pioenvs/autogen_armstrap_eagle/src/main.c:48
48	    while(ms--) {
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x264 lma 0x8000188
Start address 0x80002e8, load size 1004
Transfer rate: 2 KB/sec, 502 bytes/write.
/Users/me/Downloads/blinky-pio-armstrap/upload.gdb:6: Error in sourced command file:
The program is not being run.
======================================================================================= [ ERROR ] Took 1.70 seconds =======================================================================================
```

armstrap, 2015-04-25:
When I run the `arm-none-eabi-gdb` from the command line,  `arm-none-eabi-gdb` asks me after the upload whether I want to "Start it from the beginning?".  Do you think this prompt is what it's hanging on?  See output below...

> arm-none-eabi-gdb armstrap_eagle_testprogram.elf
   GNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20131129-cvs
   Copyright (C) 2013 Free Software Foundation, Inc.
   License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
   This is free software: you are free to change and redistribute it.
   There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
   and "show warranty" for details.
   This GDB was configured as "--host=i686-linux-gnu --target=arm-none-eabi".
   For bug reporting instructions, please see:
   <http://www.gnu.org/software/gdb/bugs/>...
   Reading symbols from armstrap_eagle_testprogram.elf...done.
   (gdb) target extended-remote /dev/ttyACM0
   Remote debugging using /dev/ttyACM0
   (gdb) monitor swdp_scan
   Target voltage: 4.9V
   Available Targets:
   No. Att Driver
    1      STM32F4xx
   (gdb) attach 1
   Attaching to program: armstrap_eagle_testprogram.elf, Remote target
   0x080013ae in main () at ../source/main.cpp:51
   51		};
   (gdb) load
   Loading section .isr_vector, size 0x188 lma 0x8000000
   Loading section .text, size 0x1ccc lma 0x8000188
   Loading section .data, size 0x864 lma 0x8001e54
   Start address 0x8001334, load size 9912
   Transfer rate: 17 KB/sec, 826 bytes/write.
   (gdb) run
   The program being debugged has been started already.
   Start it from the beginning? (y or n) y   <----------------  USER PROMPT
   Starting program: armstrap_eagle_testprogram.elf

valeros, 2015-04-25:
Let's try upload with minimum set of commands:
```
target extended-remote /dev/cu.usbmodem7B4097C1
monitor swdp_scan
attach 1
load
quit
```

beriberikix, 2015-04-25:
No errors this time, but no blinking. @armstrap I'm using the [standard sample](https://gist.github.com/beriberikix/07580e3dd98def9be4d2) - shouldn't the USER LED be toggling? After I flash it the LED just [stays solid](https://youtu.be/sc1EAVex2uc), no matter what value I pass into `delay()`.

```
$ platformio run
[Sat Apr 25 11:49:04 2015] Processing autogen_armstrap_eagle (targets: upload, platform: ststm32gdb, board: armstrap_eagle, framework: cmsis,spl)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
/Users/me/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gdb /Users/me/Downloads/blinky-pio-armstrap/.pioenvs/autogen_armstrap_eagle/firmware.elf -batch -x /Users/me/Downloads/blinky-pio-armstrap/upload.gdb
Target voltage: 4.9V
Available Targets:
No. Att Driver
1      STM32F4xx
delay (ms=133394, ms@entry=250) at .pioenvs/autogen_armstrap_eagle/src/main.c:48
48	    while(ms--) {
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x264 lma 0x8000188
Start address 0x80002e8, load size 1004
Transfer rate: 2 KB/sec, 502 bytes/write.
A debugging session is active.

Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) [answered Y; input not from terminal]
======================================================================================= [SUCCESS] Took 1.46 seconds =======================================================================================
```

valeros, 2015-04-25:
The board should be blinking.
I tried last `upload.gdb` with my discovery board and it works fine.
@armstrap
We run GDB in batch mode (without prompts), but it still hangs on `run` command execution.

armstrap, 2015-04-26:
@valeros Wouldn't the upload with minimum set of commands also include the confirmation to disconnect from the target?

> target extended-remote /dev/ttyACM0
monitor swdp_scan
attach 1
load
quit
yes

Here is my output:
> $ arm-none-eabi-gdb armstrap_eagle_testprogram.elf
GNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20131129-cvs
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-linux-gnu --target=arm-none-eabi".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from armstrap_eagle_testprogram.elf...done.
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor swdp_scan
Target voltage: 4.9V
Available Targets:
No. Att Driver
 1      STM32F4xx
(gdb) attach 1
Attaching to program: armstrap_eagle_testprogram.elf, Remote target
0x0800161a in delay (ms=121411) at ../source/main.cpp:71
71	    while(ms--) {
(gdb) load
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x1ccc lma 0x8000188
Loading section .data, size 0x864 lma 0x8001e54
Start address 0x8001334, load size 9912
Transfer rate: 17 KB/sec, 826 bytes/write.
(gdb) quit
A debugging session is active.
.
	Inferior 1 [Remote target] will be detached.
.
Quit anyway? (y or n) y
Detaching from program: armstrap_eagle_testprogram.elf, Remote target
$

@beriberikix Where did you find the link to that example hosted on my s3 account?  It's really old and buggy.  Please consider replacing it with the one in the [github repo](https://github.com/armstrap/armstrap-eclipse-workspace/tree/master/Task-01-Blink).   Or at a minimum, change 

```c++
void loop()
{
    uint32_t counter = 0;
    while (1) {
        counter++;
        GPIO_ResetBits(GPIOC, LEDS);
        GPIO_SetBits(GPIOC, LED[counter % 2]); // ERROR 'LED' array out of bounds referencing
        delay(250);
    };
}
```

to

``` c++
void loop()
{
    uint32_t counter = 0;
    while (1) {
        counter++;
        GPIO_ToggleBits(GPIOC, GPIO_Pin_1);
        delay(250);
    };
}
```
to avoid the out-of-range error you are probably seeing.

valeros, 2015-04-26:
GDB does not require confirmations in batch mode.
Here is my log:
```
Reading symbols from disco_l152rb\firmware.elf...done.
Reset_Handler () at .pioenvs\disco_l152rb\FrameworkCMSISVariant/startup_stm32l1xx_md.S:73
73        movs r1, #0
Loading section .isr_vector, size 0x10c lma 0x8000000
Loading section .text, size 0x288 lma 0x800010c
Start address 0x8000230, load size 916
Transfer rate: 4 KB/sec, 458 bytes/write.
A debugging session is active.
        Inferior 1 [Remote target] will be killed.
Quit anyway? (y or n) [answered Y; input not from terminal]
```

armstrap, 2015-04-26:
@valeros You are correct!  Sorry about that.  I had to remove the `kill` command but running from the cli seems to upload (from an Ubuntu machine) on my end.  Do you have one of the armstrap boards to test?
```
>$ cat upload.gdb 
target extended-remote /dev/ttyACM0
monitor swdp_scan
attach 1
load
detach
quit

>$ arm-none-eabi-gdb armstrap_eagle_testprogram.elf --batch -x upload.gdb 
Target voltage: 4.9V
Available Targets:
No. Att Driver
 1      STM32F4xx
0x0800160a in delay (ms=388154) at ../source/main.cpp:71
71	    while(ms--) {
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x1ccc lma 0x8000188
Loading section .data, size 0x864 lma 0x8001e54
Start address 0x8001334, load size 9912
Transfer rate: 17 KB/sec, 826 bytes/write.
>$ 
```

Here is the output when I have the gdb `kill` command.  The upload was not success in this case (see below).

```
>$ cat upload.gdb 
target extended-remote /dev/ttyACM0
monitor swdp_scan
attach 1
load
detach
kill
quit

>$ arm-none-eabi-gdb armstrap_eagle_testprogram.elf --batch -x upload.gdb 
Target voltage: 4.9V
Available Targets:
No. Att Driver
 1      STM32F4xx
0x08001614 in delay (ms=1332313) at ../source/main.cpp:71
71	    while(ms--) {
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x1ccc lma 0x8000188
Loading section .data, size 0x864 lma 0x8001e54
Start address 0x8001334, load size 9912
Transfer rate: 17 KB/sec, 826 bytes/write.
upload.gdb:6: Error in sourced command file:
The program is not being run.
>$ 
```

valeros, 2015-04-26:
@armstrap Unfortunately, I don't have any armstrap board, maybe @beriberikix can test it.

beriberikix, 2015-04-26:
@armstrap using togglebits fixed it! I originally downloaded it from the [Getting Started guide](http://docs.armstrap.org/en/latest/getting-started-eclipse-development-tools.html#download-and-installation.)

So to recap:
1. I'm using the `ststm32gdb.py`, `armstrap.json`& `armstrap_stm32_flash.ld` from [94838565](https://github.com/platformio/platformio/issues/175#issuecomment-94838565)
2. `ststm32gdb-builder.py` from [95949898](https://github.com/platformio/platformio/issues/175#issuecomment-95949898)
3. `upload.gdb` from [96258195](https://github.com/platformio/platformio/issues/175#issuecomment-96258195)
4.  Modified version of [Armstrap blinky examples](https://s3.amazonaws.com/armstrap-public/examples/armstrap_blinkyexamples_1.0.0.zip) with the `loop()` from [96307906](https://github.com/platformio/platformio/issues/175#issuecomment-96307906)

Thanks for the help! Is it possible to roll the gdb commands back into the builder script for future users?

commit 5a162b63 (2015-04-27): Merge pull request #187 from valeros/develop

Add GDB as alternative uploader to ststm32 platform // Resolve #175
commit 60dc0268 (2015-04-27): Add GDB as alternative uploader to ststm32 platform // Resolve #175
closed 2015-04-27 by valeros