platformio/platformio-core #178: PlatformIO breaks serial interface for Arduino Leonardo (but only for PlatformIO)

WillemWillemWillem, 2015-04-16:
When I flash my Leonardo with your software, it will start disconnecting from serial immediately after having been connected. Your software will say the serial device doesn't exist, or give 

Luckily the Arduino IDE can still flash it (I have open the "Tools" menu in the 500ms after I've reset the Arduino, select the right port, then press the reset button again while uploading). After the Arduino IDE has flashed it it stays connected via serial again, and PlatformIO can flash it again.

As you can imagine this is very tiresome and is making me look for alternatives.

Here's some outputs:

    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio run --upload-port "/dev/ttyACM0"
    [Thu Apr 16 02:25:48 2015] Processing leonardo (targets: upload, platform: atmelavr, board: leonardo, framework: Arduino)
    -----------------------------------------------------------------------------------
    BeforeUpload(["upload"], [".pioenvs/leonardo/firmware.hex"])
    scons: *** [upload] /dev/ttyACM0: No such file or directory
    =========================== [ ERROR ] Took 2.39 seconds ===========================
    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio run --upload-port "/dev/ttyACM0"
    [Thu Apr 16 02:25:53 2015] Processing leonardo (targets: upload, platform: atmelavr, board: leonardo, framework: Arduino)
    -----------------------------------------------------------------------------------
    BeforeUpload(["upload"], [".pioenvs/leonardo/firmware.hex"])
    scons: *** [upload] Broken pipe
    =========================== [ ERROR ] Took 2.40 seconds ===========================

When I have just flashed the Leonardo with the Arduino IDE, I still have to press the reset button when it's started uploading. Here's the output that goes along with that:

    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio run --upload-port "/dev/ttyACM0"
    [Thu Apr 16 02:25:31 2015] Processing leonardo (targets: upload, platform: atmelavr, board: leonardo, framework: Arduino)
    -----------------------------------------------------------------------------------
    BeforeUpload(["upload"], [".pioenvs/leonardo/firmware.hex"])
    "/home/ford/.platformio/packages/tool-avrdude/avrdude" -v -D -p atmega32u4 -C "/home/ford/.platformio/packages/tool-avrdude/avrdude.conf" -c avr109 -b 57600 -P /dev/ttyACM0 -U flash:w:.pioenvs/leonardo/firmware.hex:i

    avrdude: Version 6.0.1, compiled on Apr  3 2014 at 21:52:43
    Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
    Copyright (c) 2007-2009 Joerg Wunsch

    System wide configuration file is "/home/ford/.platformio/packages/tool-avrdude/avrdude.conf"

    Using Port                    : /dev/ttyACM0
    Using Programmer              : avr109
    Overriding Baud Rate          : 57600
    AVR Part                      : ATmega32U4
    Chip Erase delay              : 9000 us
    PAGEL                         : PD7
    BS2                           : PA0
    RESET disposition             : dedicated
    RETRY pulse                   : SCK
    serial program mode           : yes
    parallel program mode         : yes
    Timeout                       : 200
    StabDelay                     : 100
    CmdexeDelay                   : 25
    SyncLoops                     : 32
    ByteDelay                     : 0
    PollIndex                     : 3
    PollValue                     : 0x53
    Memory Detail                 :

    Block Poll               Page                       Polled
    Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
    ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
    eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
    flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
    lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
    signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

    Programmer Type : butterfly
    Description     : Atmel AppNote AVR109 Boot Loader

    Connecting to programmer: .
    Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
    Programmer supports auto addr increment.
    Programmer supports buffered memory access with buffersize=128 bytes.

    Programmer supports the following devices:
    Device code: 0x44

    avrdude: devcode selected: 0x44
    avrdude: AVR device initialized and ready to accept instructions

    Reading | ################################################## | 100% 0.00s

    avrdude: Device signature = 0x1e9587
    avrdude: safemode: lfuse reads as FF
    avrdude: safemode: hfuse reads as D8
    avrdude: safemode: efuse reads as CB
    avrdude: reading input file ".pioenvs/leonardo/firmware.hex"
    avrdude: writing flash (240 bytes):

    Writing | ################################################## | 100% 0.02s

    avrdude: 240 bytes of flash written
    avrdude: verifying flash memory against .pioenvs/leonardo/firmware.hex:
    avrdude: load data flash data from input file .pioenvs/leonardo/firmware.hex:
    avrdude: input file .pioenvs/leonardo/firmware.hex contains 240 bytes
    avrdude: reading on-chip flash data:

    Reading | ################################################## | 100% 0.00s

    avrdude: verifying ...
    avrdude: 240 bytes of flash verified

    avrdude: safemode: lfuse reads as FF
    avrdude: safemode: hfuse reads as D8
    avrdude: safemode: efuse reads as CB
    avrdude: safemode: Fuses OK (H:CB, E:D8, L:FF)

    avrdude done.  Thank you.

    =========================== [SUCCESS] Took 5.53 seconds ===========================

I hope you can do something about this. To me this seems like you use a bugged/outdated firmware version. Feel free to contact me.

ivankravets, 2015-04-22:
Could ask you to try upload firmware without ``--upload-port``? Like:
```shell
platformio run --target upload
```

PlatformIO will automatically detect upload port.

WillemWillemWillem, 2015-04-22:
I have. Here's the command run twice without `--upload-port`:

    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio run
    [Wed Apr 22 21:37:39 2015] Processing leonardo (targets: upload, platform: atmelavr, board: leonardo, framework: Arduino)
    --------------------------------------------------------------------------------
    BeforeUpload(["upload"], [".pioenvs/leonardo/firmware.hex"])
    Auto-detected UPLOAD_PORT: /dev/ttyACM0
    "/home/ford/.platformio/packages/tool-avrdude/avrdude" -v -D -p atmega32u4 -C "/home/ford/.platformio/packages/tool-avrdude/avrdude.conf" -c avr109 -b 57600 -P /dev/ttyACM0 -U flash:w:.pioenvs/leonardo/firmware.hex:i

    avrdude: Version 6.0.1, compiled on Apr  3 2014 at 21:52:43
    Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
    Copyright (c) 2007-2009 Joerg Wunsch

    System wide configuration file is "/home/ford/.platformio/packages/tool-avrdude/avrdude.conf"

    Using Port                    : /dev/ttyACM0
    Using Programmer              : avr109
    Overriding Baud Rate          : 57600
    AVR Part                      : ATmega32U4
    Chip Erase delay              : 9000 us
    PAGEL                         : PD7
    BS2                           : PA0
    RESET disposition             : dedicated
    RETRY pulse                   : SCK
    serial program mode           : yes
    parallel program mode         : yes
    Timeout                       : 200
    StabDelay                     : 100
    CmdexeDelay                   : 25
    SyncLoops                     : 32
    ByteDelay                     : 0
    PollIndex                     : 3
    PollValue                     : 0x53
    Memory Detail                 :

    Block Poll               Page                       Polled
    Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
    ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
    eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
    flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
    lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
    signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

    Programmer Type : butterfly
    Description     : Atmel AppNote AVR109 Boot Loader

    Connecting to programmer: .
    Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
    Programmer supports auto addr increment.
    Programmer supports buffered memory access with buffersize=128 bytes.

    Programmer supports the following devices:
    Device code: 0x44

    avrdude: devcode selected: 0x44
    avrdude: AVR device initialized and ready to accept instructions

    Reading | ################################################## | 100% 0.00s

    avrdude: Device signature = 0x1e9587
    avrdude: safemode: lfuse reads as FF
    avrdude: safemode: hfuse reads as D8
    avrdude: safemode: efuse reads as CB
    avrdude: reading input file ".pioenvs/leonardo/firmware.hex"
    avrdude: writing flash (240 bytes):

    Writing | ################################################## | 100% 0.02s

    avrdude: 240 bytes of flash written
    avrdude: verifying flash memory against .pioenvs/leonardo/firmware.hex:
    avrdude: load data flash data from input file .pioenvs/leonardo/firmware.hex:
    avrdude: input file .pioenvs/leonardo/firmware.hex contains 240 bytes
    avrdude: reading on-chip flash data:

    Reading | ################################################## | 100% 0.00s

    avrdude: verifying ...
    avrdude: 240 bytes of flash verified

    avrdude: safemode: lfuse reads as FF
    avrdude: safemode: hfuse reads as D8
    avrdude: safemode: efuse reads as CB
    avrdude: safemode: Fuses OK (H:CB, E:D8, L:FF)

    avrdude done.  Thank you.

    ========================= [SUCCESS] Took 5.62 seconds =========================

That was the first time. As you can see it was succesful. However, when I run it again, it does this:

    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio run
    [Wed Apr 22 21:37:56 2015] Processing leonardo (targets: upload, platform: atmelavr, board: leonardo, framework: Arduino)
    ------------------------------------------------------------------------------------------------------------------------------------------------------------------------
    BeforeUpload(["upload"], [".pioenvs/leonardo/firmware.hex"])
    scons: *** [upload] Explicit exit, status Error: Please specify `upload_port` for environment or use global `--upload-port` option.
    For the some development platforms it can be USB flash drive

    Error: Please specify `upload_port` for environment or use global `--upload-port` option.
    For the some development platforms it can be USB flash drive

    ===================================================================== [ ERROR ] Took 2.33 seconds =====================================================================

ivankravets, 2015-04-22:
It means that problem occurs from the 2-nd attempt to upload firmware?

WillemWillemWillem, 2015-04-22:
Correct. After the (Arduino) Leonardo has been written to by PlatformIO it not longer lists as a serial device at /dev/ for longer than half a second after connecting via USB. The Arduino IDE still manages to write to it in this short timeframe or somehow prevents the device from disconnecting from serial. PlatformIO doesn't detect the device or when timed right (plug in or reset the device a bit after `BeforeUpload['upload']` appears) it says this:

    scons: *** [upload] Broken pipe

On a side note: thanks for the quick response. PlatformIO still seems a lot more comfortable in use than the official Arduino tooling IMO, because it's more flexible (doesn't dictate the use of one specific project structure).

ivankravets, 2015-04-22:
Thanks! I suppose where this issue maybe hidden. I've just created new branch-feature where we will try to resolve this issue. Please try this first build:
```shell
pip uninstall platformio 
pip install https://github.com/platformio/platformio/archive/feature/issue-178.zip
```

WillemWillemWillem, 2015-04-22:
No, sorry, this hasn't changed anything as far as I can see. To me it would seem that the problem is in the behavior of the Leonardo after having been flashed with PlatformIO, as it really does behave differently from the 'flashed by Arduino IDE"-way. Does PlatformIO upload firmware for the device aside from the code you give as an input? I'd suspect that to be the problem, but I'm pretty new to this.

# Run 1 - forgot to press reset button

    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio run
    [Thu Apr 23 01:49:17 2015] Processing leonardo (targets: upload, platform: atmelavr, board: leonardo, framework: Arduino)
    --------------------------------------------------------------------------------
    BeforeUpload(["upload"], [".pioenvs/leonardo/firmware.hex"])
    Auto-detected UPLOAD_PORT: /dev/ttyACM0
    scons: *** [upload] Explicit exit, status Error: Couldn't find a board on the selected port. Check that you have the correct port selected. If it is correct, try pressing the board's reset button after initiating the upload.
    Error: Couldn't find a board on the selected port. Check that you have the correct port selected. If it is correct, try pressing the board's reset button after initiating the upload.
    ========================= [ ERROR ] Took 14.46 seconds =========================

# Run 2 - success!

    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio run
    [Thu Apr 23 01:49:42 2015] Processing leonardo (targets: upload, platform: atmelavr, board: leonardo, framework: Arduino)
    --------------------------------------------------------------------------------
    BeforeUpload(["upload"], [".pioenvs/leonardo/firmware.hex"])
    Auto-detected UPLOAD_PORT: /dev/ttyACM0
    "/home/ford/.platformio/packages/tool-avrdude/avrdude" -v -D -p atmega32u4 -C "/home/ford/.platformio/packages/tool-avrdude/avrdude.conf" -c avr109 -b 57600 -P /dev/ttyACM0 -U flash:w:.pioenvs/leonardo/firmware.hex:i

    avrdude: Version 6.0.1, compiled on Apr  3 2014 at 21:52:43
    Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
    Copyright (c) 2007-2009 Joerg Wunsch

    System wide configuration file is "/home/ford/.platformio/packages/tool-avrdude/avrdude.conf"

    Using Port                    : /dev/ttyACM0
    Using Programmer              : avr109
    Overriding Baud Rate          : 57600
    AVR Part                      : ATmega32U4
    Chip Erase delay              : 9000 us
    PAGEL                         : PD7
    BS2                           : PA0
    RESET disposition             : dedicated
    RETRY pulse                   : SCK
    serial program mode           : yes
    parallel program mode         : yes
    Timeout                       : 200
    StabDelay                     : 100
    CmdexeDelay                   : 25
    SyncLoops                     : 32
    ByteDelay                     : 0
    PollIndex                     : 3
    PollValue                     : 0x53
    Memory Detail                 :

    Block Poll               Page                       Polled
    Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
    ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
    eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
    flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
    lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
    calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
    signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

    Programmer Type : butterfly
    Description     : Atmel AppNote AVR109 Boot Loader

    Connecting to programmer: .
    Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
    Programmer supports auto addr increment.
    Programmer supports buffered memory access with buffersize=128 bytes.

    Programmer supports the following devices:
[... 25 lines cut ...]
    avrdude: 240 bytes of flash verified

    avrdude: safemode: lfuse reads as FF
    avrdude: safemode: hfuse reads as D8
    avrdude: safemode: efuse reads as CB
    avrdude: safemode: Fuses OK (H:CB, E:D8, L:FF)

    avrdude done.  Thank you.

    ========================= [SUCCESS] Took 4.94 seconds =========================

# Run 3 - same problem :/

    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio run
    [Thu Apr 23 01:49:50 2015] Processing leonardo (targets: upload, platform: atmelavr, board: leonardo, framework: Arduino)
    --------------------------------------------------------------------------------
    BeforeUpload(["upload"], [".pioenvs/leonardo/firmware.hex"])
    scons: *** [upload] Explicit exit, status Error: Please specify `upload_port` for environment or use global `--upload-port` option.
    For the some development platforms it can be USB flash drive

    Error: Please specify `upload_port` for environment or use global `--upload-port` option.
    For the some development platforms it can be USB flash drive

    ========================= [ ERROR ] Took 2.31 seconds =========================
    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio run

ivankravets, 2015-04-23:
Let's rephrase what do we have:

1. PlatformIO can upload firmware to Leonardo board if it was previously flashed by Arduino IDE?
2. See [Arduino IDE upload code](https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/packages/uploaders/SerialUploader.java#L107:L149) / [PlatformIO upload code](https://github.com/platformio/platformio/blob/develop/platformio/builder/tools/pioupload.py) and [atmelavr](https://github.com/platformio/platformio/blob/develop/platformio/builder/scripts/atmelavr.py#L17:L54) platform.

Could you provide TeamViewer access to your Mac with connected Leonardo board? Because we don't have this board and we can't reproduce this issue.

ivankravets, 2015-04-23:
Could I ask to run this command when you connect Leonardo to USB:
```shell
platformio serialports list
```

PlatformIO uses this command to auot-detect available upload ports.

WillemWillemWillem, 2015-04-23:
That seems correct, yes. However, I do not have a Mac, I run Lubuntu 14.04. I'll have a look at the sources, and yes, I've just installed TeamViewer so give me a sign when you have the time.

# Output after having been flashed with PlatformIO:
    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio serialports list
    /dev/ttyS31
    ----------
    Hardware ID: n/a
    Description: ttyS31

    /dev/ttyS30
    ----------
    Hardware ID: n/a
    Description: ttyS30

    /dev/ttyS29
    ----------
    Hardware ID: n/a
    Description: ttyS29

    /dev/ttyS28
    ----------
    Hardware ID: n/a
    Description: ttyS28

    /dev/ttyS27
    ----------
    Hardware ID: n/a
    Description: ttyS27

    /dev/ttyS26
    ----------
    Hardware ID: n/a
    Description: ttyS26

    /dev/ttyS25
    ----------
    Hardware ID: n/a
    Description: ttyS25

    /dev/ttyS24
    ----------
    Hardware ID: n/a
    Description: ttyS24

    /dev/ttyS23
    ----------
    Hardware ID: n/a
    Description: ttyS23

    /dev/ttyS22
    ----------
    Hardware ID: n/a
    Description: ttyS22

    /dev/ttyS21
    ----------
    Hardware ID: n/a
    Description: ttyS21

    /dev/ttyS20
    ----------
    Hardware ID: n/a
    Description: ttyS20

    /dev/ttyS19
    ----------
    Hardware ID: n/a
    Description: ttyS19

    /dev/ttyS18
    ----------
    Hardware ID: n/a
    Description: ttyS18

    /dev/ttyS17
    ----------
    Hardware ID: n/a
    Description: ttyS17

    /dev/ttyS16
    ----------
    Hardware ID: n/a
    Description: ttyS16

    /dev/ttyS15
    ----------
    Hardware ID: n/a
    Description: ttyS15

    /dev/ttyS14
    ----------
    Hardware ID: n/a
    Description: ttyS14

    /dev/ttyS13
    ----------
    Hardware ID: n/a
    Description: ttyS13

    /dev/ttyS12
    ----------
    Hardware ID: n/a
    Description: ttyS12

    /dev/ttyS11
    ----------
    Hardware ID: n/a
    Description: ttyS11

    /dev/ttyS10
    ----------
    Hardware ID: n/a
    Description: ttyS10

    /dev/ttyS9
    ----------
    Hardware ID: n/a
    Description: ttyS9

    /dev/ttyS8
    ----------
    Hardware ID: n/a
    Description: ttyS8

    /dev/ttyS7
    ----------
    Hardware ID: n/a
    Description: ttyS7

    /dev/ttyS6
    ----------
    Hardware ID: n/a
    Description: ttyS6

    /dev/ttyS5
    ----------
    Hardware ID: n/a
    Description: ttyS5

    /dev/ttyS4
    ----------
    Hardware ID: n/a
    Description: ttyS4

    /dev/ttyS3
    ----------
    Hardware ID: n/a
    Description: ttyS3

    /dev/ttyS2
    ----------
    Hardware ID: n/a
    Description: ttyS2

    /dev/ttyS1
    ----------
    Hardware ID: n/a
    Description: ttyS1

    /dev/ttyS0
    ----------
    Hardware ID: n/a
    Description: ttyS0

# Output after having been flashed with Arduino IDE:
    ford@FordPrefect:~/Documents/work/42tech/cbgb/arduino$ platformio serialports list
    /dev/ttyS31
    ----------
    Hardware ID: n/a
    Description: ttyS31

    /dev/ttyS30
    ----------
    Hardware ID: n/a
    Description: ttyS30

    /dev/ttyS29
    ----------
    Hardware ID: n/a
    Description: ttyS29

    /dev/ttyS28
    ----------
    Hardware ID: n/a
    Description: ttyS28

    /dev/ttyS27
    ----------
    Hardware ID: n/a
    Description: ttyS27

    /dev/ttyS26
    ----------
    Hardware ID: n/a
    Description: ttyS26

    /dev/ttyS25
    ----------
    Hardware ID: n/a
    Description: ttyS25

    /dev/ttyS24
    ----------
    Hardware ID: n/a
    Description: ttyS24

    /dev/ttyS23
    ----------
    Hardware ID: n/a
    Description: ttyS23

    /dev/ttyS22
    ----------
    Hardware ID: n/a
    Description: ttyS22

    /dev/ttyS21
    ----------
    Hardware ID: n/a
    Description: ttyS21

    /dev/ttyS20
    ----------
    Hardware ID: n/a
    Description: ttyS20

    /dev/ttyS19
    ----------
    Hardware ID: n/a
    Description: ttyS19

    /dev/ttyS18
    ----------
    Hardware ID: n/a
    Description: ttyS18

    /dev/ttyS17
    ----------
    Hardware ID: n/a
    Description: ttyS17

    /dev/ttyS16
    ----------
    Hardware ID: n/a
    Description: ttyS16

    /dev/ttyS15
    ----------
    Hardware ID: n/a
    Description: ttyS15

    /dev/ttyS14
    ----------
    Hardware ID: n/a
    Description: ttyS14

    /dev/ttyS13
    ----------
    Hardware ID: n/a
    Description: ttyS13

    /dev/ttyS12
    ----------
    Hardware ID: n/a
    Description: ttyS12

    /dev/ttyS11
    ----------
    Hardware ID: n/a
    Description: ttyS11

    /dev/ttyS10
    ----------
    Hardware ID: n/a
    Description: ttyS10

    /dev/ttyS9
    ----------
    Hardware ID: n/a
    Description: ttyS9

    /dev/ttyS8
    ----------
    Hardware ID: n/a
    Description: ttyS8

    /dev/ttyS7
    ----------
    Hardware ID: n/a
    Description: ttyS7

    /dev/ttyS6
    ----------
    Hardware ID: n/a
    Description: ttyS6

    /dev/ttyS5
    ----------
    Hardware ID: n/a
    Description: ttyS5

    /dev/ttyS4
    ----------
    Hardware ID: n/a
    Description: ttyS4

    /dev/ttyS3
    ----------
    Hardware ID: n/a
    Description: ttyS3

    /dev/ttyS2
    ----------
    Hardware ID: n/a
    Description: ttyS2

    /dev/ttyS1
    ----------
    Hardware ID: n/a
    Description: ttyS1

    /dev/ttyS0
    ----------
    Hardware ID: n/a
    Description: ttyS0

    /dev/ttyACM0
    ----------
    Hardware ID: USB VID:PID=2341:8036
    Description: ttyACM0

ivankravets, 2015-04-23:
@valeros pelase contact with @WillemWillemWillem and try to debug the Leonardo behaviour.

P.S: @WillemWillemWillem please email to valeros08@gmail.com.

WillemMali, 2015-05-14:
The issue's solved, uploads work without resets now. There were two problems:

* Not using Arduino.h in my project, which lists the Arduino Leonardo as a serial device. Without Arduino.h the Arduino Leonardo will still register as a serial device, but will disconnect less than a second after connecting/resetting.
* A bug in PlatformIO making a reset of the Arduino Leonardo required during upload, fixed in this version:
    https://github.com/platformio/platformio/archive/feature/issue-178.zip
Which can be installed using `pip install [the URL above]`.
The changes in this version will be merged into PlatformIO 1.5.0 soon by @valeros.

closed 2015-05-14 by ivankravets

andreas271828, 2015-12-01:
Hi @WillemMali,

Can you please provide more details regarding Arduino.h. I'm trying to avoid including Arduino specific files or only want to include what's absolutely necessary. Can you explain what is happening and what code in Arduino.h (or files that are included indirectly) is preventing that from happening.

Also, I have two Leonardos and now broke both of them this way. Unfortunately, I just can't get anything running with the Reset button - with both Leonardos. I tried to use the Reset button to upload software without the 1200bps trick (that the Arduino IDE and PlatformIO do automatically) and I tried to use the button to get the microcontroller back into a programmable state as described above by you. I can't seem to ever see the device again with platformio serialports list or in Arduino IDE, even shortly after pressing the button or plugging the USB cable back in.

I use Linux Mint 17.

Cheers,
Andreas

WillemMali, 2015-12-02:
Hi Andreas,

I personally am under the impression that Arduino.h puts the Arduino in a
permanent 'ready to flash software' state, which makes uploading a lot
smoother (no reset needed like you normally would). There's no easy way (I
have found) to extract the specific part of Arduino.h which does this, and
as it's a HUGE include it'll be hard to find the right part.

Could you maybe share (a stripped down version of) the code you've used to
brick your Arduino? If you want, I can load it on my Leonardo and try to
reproduce your bricking. Then I'll have some more motivation to find a way
to fix this, and it'll be easier to play around with (if it is possible to
undo, which I'd expect).

As an aside, have you tried uploading the blink example using the Arduino
IDE, and fiddling with the timing of pressing the reset button? Getting
your Arduino's running again in some way will make debugging this a lot
easier. It worked for me, by only releasing the reset button something like
3 seconds after pressing the upload button. Try to time it with the outputs
you get from the console and fiddle.

That's all I have for now, good luck!
Willem

andreas271828, 2015-12-02:
Thanks Willem!

It's a huge challenge to press and release the Reset button at the right time and then also click the Upload button at the right time or even select a different port within the short time given, because sometimes the port just keeps changing! But .... I got back access to both Leonardos :)

You mentioned one of the main reasons why I actually want to get rid of Arduino.h and the rest of the framework: it's HUGE ;) The second reason is that it's less portable.
My code's working fine without it (just using core AVR includes), but it's breaking the serial interface every time, even when I include Arduino.h. If you use sketches it executes some extra code that's not executed otherwise (see main.cpp in the relevant framework code). I'll see if I can clean that up and use it directly or if I have to go with the extra load for now.

Thanks again,
Andreas

WillemMali, 2015-12-02:
Hi Andreas,
No problem at all! I'm glad you mannaged to unbrick them :)
If you do find out what piece of code in Arduino.h does this I'd be very interested; I personally couldn't find it and went with Arduino.h (and the longer compile times), but it'd be a relief to be rid of it.
Regards,Willem-------- Original message --------From: andreas271828 <notifications@github.com> Date: 12/02/2015  08:11  (GMT+01:00) To: platformio/platformio <platformio@noreply.github.com> Cc: WillemMali <willem.mali@gmail.com> Subject: Re: [platformio] PlatformIO breaks serial interface for Arduino
  Leonardo (but only for PlatformIO) (#178) 
Thanks Willem!

It's a huge challenge to press and release the Reset button at the right time and then also click the Upload button at the right time or even select a different port within the short time given, because sometimes the port just keeps changing! But .... I got back access to both Leonardos :)

You mentioned one of the main reasons why I actually want to get rid of Arduino.h and the rest of the framework: it's HUGE ;) The second reason is that it's less portable.

My code's working fine without it (just using core AVR includes), but it's breaking the serial interface every time, even when I include Arduino.h. If you use sketches it executes some extra code that's not executed otherwise (see main.cpp in the relevant framework code). I'll see if I can clean that up and use it directly or if I have to go with the extra load for now.

Thanks again,

Andreas

commit 96b06db5 (2015-05-14): Fix firmware uploading under Linux OS for Arduino Leonardo board // Resolve #178