platformio/platformio-core #179: deepSleep with espressif

TimJay, 2015-04-19:
esp8266/Arduino added support for deepSleep(), please update the platform.

ivankravets, 2015-04-21:
@TimJay this feature with ``deepSleep()`` has not been released official yet. It's located in develop branch. We can't update ``arduino-espressif`` framework to non-stable branch for the all our users. 

@valeros please prepare new ``arduino-espressif`` package and explain @TimJay how he can replace manually within ``~/.platformio/packages`` folder.

TimJay, 2015-04-21:
Ah, ok. Missed that. Thanks for looking into the package anyhow @valeros :+1: 
I tried replacing the content of the extracted local packages with the new version from esp8266/Arduino, but ran into compiling issues.

ivankravets, 2015-04-21:
> but ran into compiling issues

@valeros knows about it. He will explain you how to use this ``develop`` snapshot.

TimJay, 2015-04-21:
Nice! Great work btw, thanks.

valeros, 2015-04-21:
Hello @TimJay. You can download the new arduino framework for ESP from this link:
https://drive.google.com/file/d/0B3M423-5AXSDblF6aDZfZUxYWEU
Now, you can delete the folder with old version of `arduino-espressif` in `~/.platformio/packages` and unzip there new version.

TimJay, 2015-04-21:
Hei @valeros
after the replacement, I get a bunch of linker errors:
```
xtensa-lx106-elf-g++ -o .pioenvs/autogen_esp01/firmware.elf -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -T /home/___/.platformio/packages/ldscripts/esp8266.ld .pioenvs/autogen_esp01/src/main.o -L/home/___/.platformio/packages/framework-arduinoespressif/sdk/lib -L.pioenvs/autogen_esp01 -Wl,--start-group .pioenvs/autogen_esp01/libFrameworkArduino.a .pioenvs/autogen_esp01/libFrameworkArduinoVariant.a -lgcc -lc -lpp -lmain -lwpa -llwip -lnet80211 -lphy -lhal -Wl,--end-group
.pioenvs/autogen_esp01/libFrameworkArduino.a(libc_replacements.o): In function `strtod':
libc_replacements.c:(.irom0.text+0x344): undefined reference to `sinh'
libc_replacements.c:(.irom0.text+0x34e): undefined reference to `sinh'
libc_replacements.c:(.irom0.text+0x35c): undefined reference to `hypot'
libc_replacements.c:(.irom0.text+0x366): undefined reference to `hypot'
libc_replacements.c:(.irom0.text+0x374): undefined reference to `hypotf'
.pioenvs/autogen_esp01/libFrameworkArduino.a(libc_replacements.o): In function `__errno':
libc_replacements.c:(.irom0.text+0x37e): undefined reference to `hypotf'
libc_replacements.c:(.irom0.text+0x38c): undefined reference to `logf'
libc_replacements.c:(.irom0.text+0x396): undefined reference to `logf'
```
Let me know if you need to full log...

valeros, 2015-04-21:
Sorry, it's my mistake.
Try this platformio.ini
```
[env:esp01_8266]
platform = espressif
framework = arduino
board = esp01
build_flags = -lm
```

TimJay, 2015-04-21:
```
xtensa-lx106-elf-g++ -o .pioenvs/autogen_esp01/firmware.elf -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -T /home/___/.platformio/packages/ldscripts/esp8266.ld .pioenvs/autogen_esp01/src/main.o -L/home/___/.platformio/packages/framework-arduinoespressif/sdk/lib -L.pioenvs/autogen_esp01 -Wl,--start-group .pioenvs/autogen_esp01/libFrameworkArduino.a .pioenvs/autogen_esp01/libFrameworkArduinoVariant.a -lm -lgcc -lc -lpp -lmain -lwpa -llwip -lnet80211 -lphy -lhal -Wl,--end-group
/home/___/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/autogen_esp01/firmware.elf section `.text' will not fit in region `iram1_0_seg'
```

valeros, 2015-04-21:
Strange, can you show me your main file source code?

TimJay, 2015-04-21:
Just a very basic deepSleep test:
```
#include "main.h"

void setup() {
	pinMode(BUILTIN_LED, OUTPUT);
	int i;
	for (i = 0; i < 10; i++) {
		digitalWrite(BUILTIN_LED, HIGH);
		delay(50);
		digitalWrite(BUILTIN_LED, LOW);
		delay(50);
	}
	ESP.deepSleep(5000000, WAKE_RF_DISABLED);
}

void loop() {
}

```

TimJay, 2015-04-21:
main.h
```
#ifndef _main_h_
#define _main_h_
#include "Arduino.h"

#ifdef __cplusplus
extern "C" {
#endif /* __cplusplus */
void loop();
void setup();
#ifdef __cplusplus
} // extern "C"
#endif /* __cplusplus */

#endif /* _main_h_ */
```

valeros, 2015-04-21:
Strange, I compiled your code without errors.
Could you show me full log, please?

TimJay, 2015-04-21:
```
platformio run 
[Tue Apr 21 21:17:10 2015] Processing autogen_esp01 (platform: espressif, build_flags: -lm, board: esp01, framework: arduino)
--------------------------------------------------------------------------------
xtensa-lx106-elf-g++ -o .pioenvs/autogen_esp01/src/main.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO=10601 -DPLATFORMIO=010400 -I/home/___/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/autogen_esp01/FrameworkArduino -I.pioenvs/autogen_esp01/FrameworkArduinoVariant .pioenvs/autogen_esp01/src/main.cpp
xtensa-lx106-elf-gcc -o .pioenvs/autogen_esp01/FrameworkArduino/cont_util.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO=10601 -I/home/___/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/autogen_esp01/FrameworkArduino -I.pioenvs/autogen_esp01/FrameworkArduinoVariant .pioenvs/autogen_esp01/FrameworkArduino/cont_util.c
xtensa-lx106-elf-gcc -o .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_noniso.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO=10601 -I/home/___/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/autogen_esp01/FrameworkArduino -I.pioenvs/autogen_esp01/FrameworkArduinoVariant .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_noniso.c
xtensa-lx106-elf-gcc -o .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_wiring.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO=10601 -I/home/___/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/autogen_esp01/FrameworkArduino -I.pioenvs/autogen_esp01/FrameworkArduinoVariant .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_wiring.c
xtensa-lx106-elf-gcc -o .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_wiring_analog.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO=10601 -I/home/___/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/autogen_esp01/FrameworkArduino -I.pioenvs/autogen_esp01/FrameworkArduinoVariant .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_wiring_analog.c
xtensa-lx106-elf-gcc -o .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_wiring_digital.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO=10601 -I/home/___/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/autogen_esp01/FrameworkArduino -I.pioenvs/autogen_esp01/FrameworkArduinoVariant .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_wiring_digital.c
xtensa-lx106-elf-gcc -o .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_wiring_pulse.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO=10601 -I/home/___/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/autogen_esp01/FrameworkArduino -I.pioenvs/autogen_esp01/FrameworkArduinoVariant .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_wiring_pulse.c
xtensa-lx106-elf-gcc -o .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_wiring_shift.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO=10601 -I/home/___/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/autogen_esp01/FrameworkArduino -I.pioenvs/autogen_esp01/FrameworkArduinoVariant .pioenvs/autogen_esp01/FrameworkArduino/core_esp8266_wiring_shift.c
xtensa-lx106-elf-gcc -o .pioenvs/autogen_esp01/FrameworkArduino/libc_replacements.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO=10601 -I/home/___/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/autogen_esp01/FrameworkArduino -I.pioenvs/autogen_esp01/FrameworkArduinoVariant .pioenvs/autogen_esp01/FrameworkArduino/libc_replacements.c
xtensa-lx106-elf-g++ -o .pioenvs/autogen_esp01/FrameworkArduino/Esp.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDU [... cut ...] bFrameworkArduinoVariant.a -lm -lgcc -lc -lpp -lmain -lwpa -llwip -lnet80211 -lphy -lhal -Wl,--end-group
/home/___/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/autogen_esp01/firmware.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/autogen_esp01/firmware.elf] Error 1
========================= [ ERROR ] Took 2.93 seconds =========================
```

valeros, 2015-04-21:
OK, I have reproduced this error on Ubuntu x64.
I will try to fix it tomorrow.

TimJay, 2015-04-21:
Great, thanks!

valeros, 2015-04-22:
To avoid breaking the stable release I have created individual branch for your case. Here are two required steps to use it:
1. Delete your previous version of PlatformIO:
`pip uninstall platformio`
2. Install new fixed version:
`pip install https://github.com/valeros/platformio/archive/feature/esp_newframework.zip`

TimJay, 2015-04-22:
Works nicely with the updated `~/.platformio/packages/arduino-espressif` installed and the default `platformio.ini`.
[DeepSleep hacked ESP-01](http://tim.jagenberg.info/2015/01/18/low-power-esp8266/) module now sleeps at 33microAmps :+1:

valeros, 2015-04-22:
Glade to hear that! 
We will try to add full support for new framework as soon as possible.

ivankravets, 2015-05-22:
Resolved in PlatformIO 2.0.0

closed 2015-05-22 by ivankravets