platformio/platformio-core #184: Unregister Libraries

thijse, 2015-04-24:
I'm splitting up a Github repository with multiple Arduino libraries in several repositories, each containing one library. I've registered these libraries (awaiting moderation), and I would like to unregister the old ones. Is there a way to do that?

ivankravets, 2015-04-25:
Thanks! I've just pushed a few pull-requests to your repos. Could I ask you to merge them?

thijse, 2015-04-25:
Hi Ivan,

Thanks for your modifications! I've pulled in the documentation changes,
but I wonder: Want is the rationale behind excluding the *extras*
directories? I would really like to keep them: They contain the library
documentation. In the case of  CmdMessenger, the extra's folder also
contains the PC side code needed to have Arduino-to-PC communication.
Without this, the library would be almost useless.

Thijs

ivankravets, 2015-04-25:
> Want is the rationale behind excluding the *extras* directories? 
PlatformIO Library Crawler stores library source code in the own storage. In this case, I prefer to keep just functional source code. Using ``exclude`` field allowed to re-pack original libraries to 20-50Kb for 90% cases. 

If developer wants to read additional information about library, he can use  on-line Library Registry and navigate by library URL. However, I will add additional information for each lib, where will be "direct" url where library is located.

So, if you want to keep ``extras`` in the library installation folder, then please ignore my pull-requests. Please don't forget to remove ``include`` field in the one of lib, because it means that library source code is located in this ``include`` directory.

thijse, 2015-04-27:
Hi Ivan,

Thanks! I will keep the extra's for now. If there comes an option to add a
documentation URL, I will leave the documentation on Github.

Furthermore: Great work on platformIO! package management will help to get
Arduino (and it's cousins) software development to the next level!

If I might make a few suggestions on where to go next:

   - Add a 1-click install version for windows, and add a simple GUI:
   Arduino users in particular are often beginners in software development and
   many will get frightened by the steps they have to take now and it's CLI.
   - Add versioning to the dependencies. Major version package updates may
   well break compatibility with packages that the depend on them. Getting a
   consistent set of packages is not an easy problem, but it is interesting
   (from a computer science perspective) I've attached a few articles on the
   topic that I came across. In particular, the OPIUM article explained the
   connection between package management and pseudo-boolean problem solving in
   a way that I understood it.

Thanks for your help!

Thijs

ivankravets, 2015-04-27:
> If there comes an option to add a documentation URL, I will leave the documentation on Github.

``library.json`` has [url](http://docs.platformio.org/en/latest/librarymanager/config.html#url) field. Please use it.

> Add a 1-click install version for windows, and add a simple GUI:
   Arduino users in particular are often beginners in software development and
   many will get frightened by the steps they have to take now and it's CLI.

Could I ask you to open separate issue for it? I agree with you, we lost 50% potential users in the installation steps :( 

> Add versioning to the dependencies. Major version package updates may
   well break compatibility with packages that the depend on them.

What do you mean under package? Because we have in PlatformO [packages](http://docs.platformio.org/en/latest/platforms/creating_platform.html#packages) and [libraries](http://docs.platformio.org/en/latest/librarymanager/index.html).

ivankravets, 2015-05-05:
If you have any questions don't hesitate to reopen this issue.

closed 2015-05-05 by ivankravets