platformio/platformio-core #185: Nucleo F411RE not running code correctly

dave93cab, 2015-04-27:
I have a nucleo F411RE, I have tried compiling/downloading files such as blink and my own code but the code does not appear to run correctly. Printf's only transmit first character and while(1) loop does not appear to be executed. Same programs compile and run fine with mbed online compiler.

Repeated on multiple computers with same results. compile/download to freescale freedom board works correctly on both computers.

Can submit .bin generated by mbed and platformio if required, to compare?

Thanks

valeros, 2015-04-27:
Looks like mbed fixed this issue in new version. Can you test this firmware.bin with your board?
http://drp.io/f/Q11RtNlT

dave93cab, 2015-04-27:
Hi, 

I can try this later. I have the M10 firmware which is the latest from the mbed developer site. Do you mind posting the link to the fixed issue?

dave93cab, 2015-04-27:
Just loaded the .bin file. If it was the blink example it didn't work, LED1 is steady. I tried it on 2 different boards with the same result. Let me know if you need anything to help diagnose the issue.

valeros, 2015-04-28:
Yes, it's blink example. Could you please test this version:
http://drp.io/f/XyVcTlGp

dave93cab, 2015-04-28:
Hi valeros, I just tried that one and same result I'm afraid: the LED doesn't blink.

Dave

valeros, 2015-05-04:
Hi @dave93cab !
Could you try this version, please?
http://bit.ly/1E8TxDA

dave93cab, 2015-05-04:
Valeros, that one worked great! Did you find the issue?

Thanks
Dave

valeros, 2015-05-04:
I almost got it.
Could you try firmwares from this archive, please?
http://bit.ly/1bsqLGQ
It will help me clarify the problem. Thanks!

dave93cab, 2015-05-04:
None of those worked

ivankravets, 2015-05-14:
Fixed in https://github.com/platformio/platformio/commit/4ce09f1d4330f9da482c8111ce6451d14f2b081a

closed 2015-05-14 by ivankravets

dave93cab, 2015-05-14:
Great, I will give this another go later :)

ivankravets, 2015-05-14:
I'm going to release 1.5.0 within 1 hour. Please upgrade your current version of PlatformIO to stable and retest. 

Thanks in advance!